espressif/esp-matter #1288: Error installing Matter SDK in branch 'release/v1.2'

lunDreame, 2025-02-22:
Hello, I installed it with the command below, but I get an error.
    ```shell
    $ cd ${sdk_path}
    $ git clone --branch release/v1.2 --depth 1 https://github.com/espressif/esp-matter.git esp-matter
    $ cd esp-matter
    $ git submodule update --init --depth 1
    $ cd ./connectedhomeip/connectedhomeip
    $ ./scripts/checkout_submodules.py --platform esp32 linux --shallow
    $ cd ${sdk_path}/esp-matter
    $ bash ./install.sh
    ```

shubhamdp, 2025-02-25:
@lunDreame Problem is due to incompatibility of Python v3.13 with cffi v1.15.1

This was fixed on master in https://github.com/project-chip/connectedhomeip/pull/36549/files

Can you check if bumping the cffi to v1.17.1 would fix the problem?

```
cd esp-matter/connectedhomeip/connectedhomeip
curl https://github.com/shubhamdp/connectedhomeip/commit/a15a6245f5141ca3d4c0b66e794679a1f5a6a1ad.patch | git am

# remove previous env and re-run the install.sh
rm -rf .environment
cd ../../
source /path/to/esp-idf/export.sh
./install.sh
```

lunDreame, 2025-02-25:
Thank you I will checked for this issue

shubhamdp, 2025-03-07:
@lunDreame were you able to run `install.sh` successfully? If yes, let me know so that I can raise those changes and we can update the submodule.

lunDreame, 2025-03-08:
Yes, is succesfully worked thank you for fix the issue

closed 2025-03-09 by dhrishi