espressif/esp-matter #1289: when thermostat has a AUTO feature, the HEAT and COOL modes stop triggering the app attribute callback

albrownwood, 2025-02-24:
Testing on iOS ecosystem
this seem to be working in the past
not matter log is visible in console when changing temp in iOS
only AUTO mode calls the callback on UI action

here is the code that creates the device:

```
node::config_t node_config;
      node_t *node = node::create(&node_config, app_attribute_update_cb_thermostat, app_identification_cb);

      thermostat::config_t thermostat_config;
      endpoint_t *endpoint = thermostat::create(node, &thermostat_config, ENDPOINT_FLAG_NONE, NULL);
      cluster_t *thermostat_cluster = cluster::get(endpoint, Thermostat::Id);

      //auto
      cluster::thermostat::feature::auto_mode::config_t auto_config;
      auto_config.min_setpoint_dead_band = 5;
      cluster::thermostat::feature::auto_mode::add(thermostat_cluster, &auto_config);

      device_ctx->thermostat_endpoint_id = endpoint::get_id(endpoint);
      ESP_LOGI(TAG, "thermostat endpoint created with endpoint_id %d", device_ctx->thermostat_endpoint_id);
```

commenting the auto section makes it work normally again for HEAT and COOL modes.
Not sure this behavior is expected.

Jerry-ESP, 2025-02-25:
I test on my side.:
1. Create a thermostat with auto mode, and commissioning into the Apple Home.
2. Change the mode to Heat or Cool, then change the heat or cool setpoint.
3. The device not receive any messages at this time.

But when I test through chip-tool, the device can receive message under the same conditions.

So I think the Apple Home App not send any commands to the device at this time.

Jerry-ESP, 2025-02-27:
> Hi [@Jerry-ESP](https://github.com/Jerry-ESP), thanks for confirming you observe the same issue. Is that an expected beahvior or do you think there is some incompatbility between esp-matter and Apple Home regarding thermostats? What would be the next step to resolve this?

This appears to be an unexpected behavior related to the Apple Home integration with the thermostat. If you have access to a feedback channel, we recommend reporting this issue to Apple. Additionally, we will also escalate this matter to Apple for further investigation.

albrownwood, 2025-02-28:
Thanks @Jerry-ESP I'll keep this open for now until you get some answer from Apple.