espressif/esp-matter #1291: Stuck with matter customization

ravencarcass, 2025-02-25:
Hello, 
I'm stuck  with a project for school where I need to make my own  Matter device. 

This device need to be able to control 2 steppermotors and 10 different display modes. 
This all need to be controlled with a google home app. The teachers decided to use Matter. Because they had a few students that tried Matter before but it never worked. I would love to be proven wrong. 

But this is where the shoe pinches. Namely, I thought to start from a matter example and build on that. I need the following interfaces: I2C, SPI, PWM, Matter and GPIO. 

So I have everything finished separately, but I don't know where to start. Personally, I tought to start at the user guide for matter SDK, which is what I have done. The problem, though, is that I hardly understand anything they say. 

The only thing I think I understood is the data model. I created one for my project.

[image]

Can somebody help me with implementing this? Does have somebody experiences with matter SDK and customize them for specific purposes?

Jerry-ESP, 2025-02-26:
I think you can setup the esp-matter sdk first and build the light example, after that, you can add other device types base on the light example. Please refer to this document: https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html

ravencarcass, 2025-02-26:
I've build the light example earlier and it worked. The only things indeed is that I then need to add those things. But I don't know how.

ravencarcass, 2025-02-26:
Ok, I've this filled in from the matter SDK manual. 

`  drift = endpoint::create(node(), ENDPOINT_FLAG_NONE); // Endpoint 1 (drift)
  tilt = endpoint::create(node(), ENDPOINT_FLAG_NONE); // Endpoint 2 (tilt)
  displayContent = endpoint::create(node(), ENDPOINT_FLAG_NONE); // Endpoint 3 (displayContent)

  uint32_t custom_cluster_id = 0x131bfc00; 
  drift = cluster::create(drift, custom_cluster_id, CLUSTER_FLAG_SERVER);
  tilt = cluster::create(tilt, custom_cluster_id, CLUSTER_FLAG_SERVER);
  displayContent = cluster::create(displayContent, custom_cluster_id, CLUSTER_FLAG_SERVER);

  uint32_t position_drift_attribute_id = 0x0001;
  uint16_t default_drift_position = 135;
  driftLevel = attribute::create(drift, position_drift_attribute_id, ATTRIBUTE_FLAG_NONE, esp_matter_uint16(default_drift_position));

  static esp_err_t move_drift_level_callback(const CreateCommandPath &command_path, TLVReader &tlv_data, void *opaque_ptr) {
    ESP_LOGI(TAG, "MoveDriftLevel command received");
    // add your code here for functionality
    return ESP_OK;
  }

  uint32_t position_tilt_attribute_id = 0x0002;
  uint16_t default_tilt_position = 135;
  tiltLevel = attribute::create(tilt, position_tilt_attribute_id, ATTRIBUTE_FLAG_NONE, esp_matter_uint16(default_tilt_position));

  static esp_err_t move_tilt_level_callback(const CreateCommandPath &command_path, TLVReader &tlv_data, void *opaque_ptr) {
    ESP_LOGI(TAG, "MoveTiltLevel command received");
    // add your code here for functionality
    return ESP_OK;
  }

  uint32_t display_content_attribute_id = 0x0003;
  uint16_t default_display_content = 0;
  Content = attribute::create(diplayContent, display_content_attribute_id, ATTRIBUTE_FLAG_NONE, esp_matter_uint16(default_display_content));

  static esp_err_t display_content_callback(const CreateCommandPath &command_path, TLVReader &tlv_data, void *opaque_ptr) {
    ESP_LOGI(TAG, "displayContent command received");
    // add your code here for functionality
    return ESP_OK;
  }`

Jerry-ESP, 2025-02-26:
Please note: For the custom endpoint, cluster, attribute,  you can't handle them through Goole Home App. The Google ecosystem can only control standard matter device which defined in the matter spec.

ravencarcass, 2025-02-26:
ok, is their then a matter system that I can use on my phone, so that can try and handle it?
I'm BTW working on the XML file. This is what I get.

```
<?xml version="1.0"?>
<configurator>
    <domain name="CHIP"/>

    <cluster>
        <domain>General</domain>
        <name>Sample ESP</name>
        <code>0x131BFC00</code>
        <description>The Sample ESP cluster showcases a manufacturer custom cluster</description>

        <endpoint id="0" reserved="true"/>

        <endpoint id="1" type="drift">
            <cluster side="server" name="LevelControl">
                <attribute code="0x0000" name="positionHorizontal" type="int16u" writable="true"/>
                <command source="client" code="0x00" name="MoveHorizontalLevel"/>
            </cluster>
        </endpoint>

        <endpoint id="2" type="tilt">
            <cluster side="server" name="LevelControl">
                <attribute code="0x0001" name="positionTilt" type="int16u" writable="true"/>
                <command source="client" code="0x01" name="MoveTiltLevel"/>
            </cluster>
        </endpoint>

        <endpoint id="3" type="displayContent">
            <cluster side="server" name="LevelControl">
                <attribute code="0x0002" name="displayMode" type="int8u" writable="true"/>
                <command source="client" code="0x02" name="displayImage"/>
            </cluster>
        </endpoint>
    </cluster>
</configurator>
```

ravencarcass, 2025-02-26:
I read part 2.5 now 10 times even more. It is for real verry frustrating that I'm not able to fix it. I begin to read and everything I read is relative easy but it's just the problem to apply the theory in the code.  This is what I get if I follow the color_temperture_lightbulb. 

app_main.cpp
 ```
/*

This example code is in the Public Domain (or CC0 licensed, at your option.)

Unless required by applicable law or agreed to in writing, this

software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR

CONDITIONS OF ANY KIND, either express or implied.

*/

#include <esp_err.h>

#include <esp_log.h>

#include <nvs_flash.h>

#include <esp_matter.h>

#include <esp_matter_console.h>

#include <esp_matter_ota.h>

#include <common_macros.h>

#include <app_priv.h>

#include <app_reset.h>

#if CHIP_DEVICE_CONFIG_ENABLE_THREAD

#include <platform/ESP32/OpenthreadLauncher.h>

#endif

#include <app/server/CommissioningWindowManager.h>

#include <app/server/Server.h>

#ifdef CONFIG_ENABLE_SET_CERT_DECLARATION_API

#include <esp_matter_providers.h>

#include <lib/support/Span.h>

#ifdef CONFIG_SEC_CERT_DAC_PROVIDER

#include <platform/ESP32/ESP32SecureCertDACProvider.h>

#elif defined(CONFIG_FACTORY_PARTITION_DAC_PROVIDER)

#include <platform/ESP32/ESP32FactoryDataProvider.h>

#endif

using namespace chip::DeviceLayer;

#endif

static const char *TAG = "app_main";

uint16_t light_endpoint_id = 0;

using namespace esp_matter;

[... 760 lines cut ...]

ESP_LOGI(TAG, "Initialising driver");

/*init btm*/

button_config_t button_config = button_driver_get_config();

button_handle_t handle = iot_button_create (&button_config);

iot_button_register_cb(handle, BUTTON_PRESS_DOWN, app_driver_button_toggle_cb);

app_reset_button_register(handle);

/*init led*/

led_driver_config_t led_config = led_driver_get_config();

led_driver_init(&led_config);

app_driver_attribute_set_defaults();

return ESP_OK;

}
```

jonsmirl, 2025-02-28:
Don't make custom clusters, they take too long and you will never finish in time.

Just map the existing light bulb app to control the motors and displays. Make 12 light bulb devices if you need to. That will work fine in GoogleHome. Start from the light bulb example. Make 12 bulbs instead of just one. Give them names so that you know what they do. Then modify the LED driver code to control your stepper or display.  You will be able to finish that in a reasonable of time for a class.

jonsmirl, 2025-02-28:
Make 12 endpoints here instead of just one

https://github.com/espressif/esp-matter/blob/main/examples/light/main/app_main.cpp#L183

ravencarcass, 2025-02-28:
I have currently done this which might work if I also use a slider for changing the display content. Which makes the application rate higher.
 ```
    motor_horizontal::config_t light_config;
    light_config.level_control.current_level = DEFAULT_BRIGHTNESS;
    light_config.level_control.on_level = DEFAULT_BRIGHTNESS;
    light_config.level_control.lighting.start_up_current_level = DEFAULT_BRIGHTNESS;
    
    motor_vertical::config_t light_config;
    light_config.level_control.current_level = DEFAULT_BRIGHTNESS;
    light_config.level_control.on_level = DEFAULT_BRIGHTNESS;
    light_config.level_control.lighting.start_up_current_level = DEFAULT_BRIGHTNESS;
    
    display_content::config_t light_config;
    light_config.level_control.current_level = DEFAULT_BRIGHTNESS;
    light_config.level_control.on_level = DEFAULT_BRIGHTNESS;
    light_config.level_control.lighting.start_up_current_level = DEFAULT_BRIGHTNESS;

//endpoints
    endpoint_t *mot_hor = motor_horizontal::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
    ABORT_APP_ON_FAILURE(mot_hor != nullptr, ESP_LOGE(TAG, "Failed to create horizontal motor endpoint"));
    
    mot_hor_endpoint_id = endpoint::get_id(mot_hor);
    ESP_LOGI(TAG, "motor horizontal created with mot_hor_endpoint_id %d", mot_hor_endpoint_id );

    endpoint_t *mot_ver = motor_vertical::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
    ABORT_APP_ON_FAILURE(mot_ver != nullptr, ESP_LOGE(TAG, "Failed to create vertical motor endpoint"));
    
    mot_ver_endpoint_id = endpoint::get_id(mot_ver);
    ESP_LOGI(TAG, "motor vertical created with mot_ver_endpoint_id %d", mot_ver_endpoint_id );

    endpoint_t *disp = display_content::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
    ABORT_APP_ON_FAILURE(disp != nullptr, ESP_LOGE(TAG, "Failed to create display content endpoint"));

    disp_endpoint_id = endpoint::get_id(disp);
    ESP_LOGI(TAG, "disp created with disp_endpoint_id %d", disp_endpoint_id );
```

Jerry-ESP, 2025-03-03:
I think you need to understand the basic definition of the esp-matter data model, please refer to this document: https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#developing-your-product.   

If you want to define 12 dimmable light, you need call the same API for 12 times:
dimmable_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
dimmable_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
dimmable_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
dimmable_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
......

The light_config is the default config of the light, is you want to define different configuration for every endpoint, you need to use different  light_config.

The driver should be called in this function: 

static esp_err_t app_attribute_update_cb(attribute::callback_type_t type, uint16_t endpoint_id, uint32_t cluster_id,
                                         uint32_t attribute_id, esp_matter_attr_val_t *val, void *priv_data)
{
    esp_err_t err = ESP_OK;

    if (type == PRE_UPDATE) {
        /* Driver update */
        //Call driver here
    }

    return err;
}

ravencarcass, 2025-03-03:
well
While the weekend did I've find an example with the dimmer.
From their did I made the next code.
The code involve 12 dimmable modules:

```
    dimmable_light::config_t light_config;
    light_config.on_off.on_off = DEFAULT_POWER;
    light_config.level_control.current_level = DEFAULT_BRIGHTNESS;
    light_config.level_control.on_level = DEFAULT_BRIGHTNESS;
    light_config.level_control.lighting.start_up_current_level = DEFAULT_BRIGHTNESS;

    // endpoint handles can be used to add/modify clusters.
    endpoint_t *endpoint = dimmable_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
    ABORT_APP_ON_FAILURE(endpoint != nullptr, ESP_LOGE(TAG, "Failed to create extended color light endpoint"));

    light_endpoint_id = endpoint::get_id(endpoint);
    ESP_LOGI(TAG, "Light created with endpoint_id %d", light_endpoint_id);

///////////////////////////////////////////////////////////////////////////

    endpoint_t *endpoint_1 = dimmable_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
    ABORT_APP_ON_FAILURE(endpoint != nullptr, ESP_LOGE(TAG, "Failed to create extended color light endpoint"));

    light_endpoint_id = endpoint::get_id(endpoint_1);
    ESP_LOGI(TAG, "Light created with endpoint_id %d", light_endpoint_id_1);

//////////////////////////////////////////////////////////////////////////

    endpoint_t *endpoint_2 = dimmable_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
    ABORT_APP_ON_FAILURE(endpoint != nullptr, ESP_LOGE(TAG, "Failed to create extended color light endpoint"));

    light_endpoint_id = endpoint::get_id(endpoint_2);
    ESP_LOGI(TAG, "Light created with endpoint_id %d", light_endpoint_id_2);

//////////////////////////////////////////////////////////////////////////

    endpoint_t *endpoint_3 = dimmable_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
    ABORT_APP_ON_FAILURE(endpoint != nullptr, ESP_LOGE(TAG, "Failed to create extended color light endpoint"));

    light_endpoint_id = endpoint::get_id(endpoint_3);
    ESP_LOGI(TAG, "Light created with endpoint_id %d", light_endpoint_id_3);

//////////////////////////////////////////////////////////////////////////

    endpoint_t *endpoint_4 = dimmable_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
    ABORT_APP_ON_FAILURE(endpoint != nullptr, ESP_LOGE(TAG, "Failed to create extended color light endpoint"));

    light_endpoint_id = endpoint::get_id(endpoint_4);
    ESP_LOGI(TAG, "Light created with endpoint_id %d", light_endpoint_id_4);

//////////////////////////////////////////////////////////////////////////

    endpoint_t *endpoint_5 = dimmable_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
    ABORT_APP_ON_FAILURE(endpoint != nullptr, ESP_LOGE(TAG, "Failed to create extended color light endpoint"));

    light_endpoint_id = endpoint::get_id(endpoint_5);
    ESP_LOGI(TAG, "Light created with endpoint_id %d", light_endpoint_id_5);

//////////////////////////////////////////////////////////////////////////

    endpoint_t *endpoint_6 = dimmable_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
    ABORT_APP_ON_FAILURE(endpoint != nullptr, ESP_LOGE(TAG, "Failed to create extended color light endpoint"));

    light_endpoint_id = endpoint::get_id(endpoint_6);
    ESP_LOGI(TAG, "Light created with endpoint_id %d", light_endpoint_id_6);

//////////////////////////////////////////////////////////////////////////

    endpoint_t *endpoint_7 = dimmable_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
    ABORT_APP_ON_FAILURE(endpoint != nullptr, ESP_LOGE(TAG, "Failed to create extended color light endpoint"));

    light_endpoint_id = endpoint::get_id(endpoint_7);
    ESP_LOGI(TAG, "Light created with endpoint_id %d", light_endpoint_id_7);

//////////////////////////////////////////////////////////////////////////

    endpoint_t *endpoint_8 = dimmable_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
    ABORT_APP_ON_FAILURE(endpoint != nullptr, ESP_LOGE(TAG, "Failed to create extended color light endpoint"));

    light_endpoint_id = endpoint::get_id(endpoint_8);
    ESP_LOGI(TAG, "Light created with endpoint_id %d", light_endpoint_id_8);

//////////////////////////////////////////////////////////////////////////

endpoint_t *endpoint_9 = dimmable_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
ABORT_APP_ON_FAILURE(endpoint != nullptr, ESP_LOGE(TAG, "Failed to create extended color light endpoint"));

light_endpoint_id = endpoint::get_id(endpoint_9);
ESP_LOGI(TAG, "Light created with endpoint_id %d", light_endpoint_id_9);

//////////////////////////////////////////////////////////////////////////

endpoint_t *endpoint_10 = dimmable_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
ABORT_APP_ON_FAILURE(endpoint != nullptr, ESP_LOGE(TAG, "Failed to create extended color light endpoint"));

light_endpoint_id = endpoint::get_id(endpoint_10);
ESP_LOGI(TAG, "Light created with endpoint_id %d", light_endpoint_id_10);

//////////////////////////////////////////////////////////////////////////

endpoint_t *endpoint_11 = dimmable_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
ABORT_APP_ON_FAILURE(endpoint != nullptr, ESP_LOGE(TAG, "Failed to create extended color light endpoint"));

light_endpoint_id = endpoint::get_id(endpoint_11);
ESP_LOGI(TAG, "Light created with endpoint_id %d", light_endpoint_id_11);

//////////////////////////////////////////////////////////////////////////
```

ravencarcass, 2025-03-07:
well since wednesday did it works and could I add some more function in name of I2C. But their are a few things that I find weird. I would like to add 3 GPIO's and 2 ledc pins. But if I add the initialization(timer and channel) of ledc does it report the next: out-of-order initializers are nonstandard in C++. If I test this apart from the matter application works it seamless, what is weird that it don't works in the light example. I escaped the init for the ledc and it worked but once I add the ledc to it gave it me a problem.
```
void periferal_init(void){
    i2c_config_t confi2c = {
        .mode = I2C_MODE_MASTER, 
        .sda_io_num = I2C_MASTER_SDA_IO,
        .scl_io_num = I2C_MASTER_SCL_IO,
        .sda_pullup_en = GPIO_PULLUP_ENABLE,
        .scl_pullup_en = GPIO_PULLUP_ENABLE,
        .master = { .clk_speed = I2C_MASTER_FREQ_HZ } // Correcte initialisatie
    };

    i2c_param_config(I2C_NUM_0, &confi2c);
    i2c_driver_install(I2C_NUM_0,confi2c.mode,0,0,0);

    ledc_timer_config_t timer = {
        .speed_mode         = LEDC_LOW_SPEED_MODE,
        .timer_num          = LEDC_TIMER_1,
        .duty_resolution    = LEDC_TIMER_13_BIT,
        .freq_hz            = 500,
        .clk_cfg            = LEDC_AUTO_CLK
    };
    ledc_timer_config(&timer);

    ledc_channel_config_t channel = {
        .gpio_num       = 10,
        .speed_mode     = LEDC_LOW_SPEED_MODE,
        .channel        = LEDC_CHANNEL_1,
        .intr_type      = LEDC_INTR_DISABLE,
        .timer_sel      = LEDC_TIMER_1,
        .duty           = DUTY,
        .hpoint         = 0
    };
    ledc_channel_config(&channel);
}

```

ravencarcass, 2025-04-22:
I pulled it appart in side codes and solved each issue apart and it works now.  Somethings went wrong in the order to define every thing. So thanks for your supprt.

closed 2025-04-22 by ravencarcass