espressif/esp-matter #1292: Issue while Building light application from esp-matter directory in wsl esp-idf (IDFGH-13898)

TGG-3, 2025-02-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

build completed successfully

### What is the actual behavior?

 error: 'CONFIG_ESP_WIFI_STATIC_RX_BUFFER_NUM' was not declared in this scope; did you mean 'WIFI_STATIC_TX_BUFFER_NUM'?
  346 |     cfg = WIFI_INIT_CONFIG_DEFAULT();
      |           ^~~~~~~~~~~~~~~~~~~~~~~~

### Steps to reproduce.

1. cd esp-matter/examples/light
2. idf.py set-target esp32h2
3. idf.py build

### Build or installation Logs.

```plain
g3@Gayathri:~$ cd esp-matter/examples/light
g3@Gayathri:~/esp-matter/examples/light$ idf.py build
Executing action: all (aliases: build)
Running cmake in directory /home/g3/esp-matter/examples/light/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/g3/.espressif/python_env/idf5.3_py3.10_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/g3/esp-matter/examples/light"...
IDF_VERSION: 5.3.0; idf version number: 5.3.0
Sequence of SDKCONFIG files: sdkconfig.defaults
-- Component directory /home/g3/esp-matter/examples/common/blemesh_platform does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/g3/esp-matter/examples/common/cmake_common does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/g3/esp-matter/examples/common/external_platform does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/g3/esp-matter/examples/common/relinker does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/g3/esp-matter/examples/common/secondary_ble_adv does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/g3/esp-matter/examples/common/utils does not contain a CMakeLists.txt file. No component will be added
-- Building ESP-IDF components for target esp32h2
NOTICE: Skipping optional dependency: espressif/cmake_utilities
NOTICE: Skipping optional dependency: espressif/mdns
NOTICE: Skipping optional dependency: espressif/esp_insights
NOTICE: Skipping optional dependency: espressif/cmake_utilities
NOTICE: Skipping optional dependency: espressif/mdns
NOTICE: Skipping optional dependency: espressif/esp_insights
NOTICE: Skipping optional dependency: espressif/mdns
NOTICE: Skipping optional dependency: espressif/esp_insights
NOTICE: Skipping optional dependency: espressif/cmake_utilities
NOTICE: Skipping optional dependency: espressif/mdns
NOTICE: Skipping optional dependency: espressif/esp_insights
NOTICE:
Following dependencies have new versions available:
Dependency "espressif/button": "3.3.1" -> "3.3.2"
Consider running "idf.py update-dependencies" to update your lock file.
[... 92 lines cut ...]
/home/g3/platform/ESP32_custom/ESP32Utils.cpp:346:11: note: maximum limit of 1000 namespaces searched for 'CONFIG_ESP_WIFI_TX_BUFFER_TYPE'
/home/g3/platform/ESP32_custom/ESP32Utils.cpp:346:11: error: 'CONFIG_ESP_WIFI_DYNAMIC_RX_MGMT_BUF' was not declared in this scope
  346 |     cfg = WIFI_INIT_CONFIG_DEFAULT();
      |           ^~~~~~~~~~~~~~~~~~~~~~~~
/home/g3/platform/ESP32_custom/ESP32Utils.cpp:346:11: note: maximum limit of 1000 namespaces searched for 'CONFIG_ESP_WIFI_DYNAMIC_RX_MGMT_BUF'
/home/g3/platform/ESP32_custom/ESP32Utils.cpp:346:11: error: 'CONFIG_ESP_WIFI_ESPNOW_MAX_ENCRYPT_NUM' was not declared in this scope
  346 |     cfg = WIFI_INIT_CONFIG_DEFAULT();
      |           ^~~~~~~~~~~~~~~~~~~~~~~~
/home/g3/platform/ESP32_custom/ESP32Utils.cpp:346:11: note: maximum limit of 1000 namespaces searched for 'CONFIG_ESP_WIFI_ESPNOW_MAX_ENCRYPT_NUM'
/home/g3/platform/ESP32_custom/ESP32Utils.cpp:346:11: error: no match for 'operator=' (operand types are 'wifi_init_config_t' and '<brace-enclosed initializer list>')
  346 |     cfg = WIFI_INIT_CONFIG_DEFAULT();
      |           ^~~~~~~~~~~~~~~~~~~~~~~~
/home/g3/esp-idf/components/esp_wifi/include/esp_wifi.h:124:3: note: candidate: 'constexpr wifi_init_config_t& wifi_init_config_t::operator(const wifi_init_config_t&)'
  124 | } wifi_init_config_t;
      |   ^~~~~~~~~~~~~~~~~~
/home/g3/esp-idf/components/esp_wifi/include/esp_wifi.h:124:3: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to 'const wifi_init_config_t&'
/home/g3/esp-idf/components/esp_wifi/include/esp_wifi.h:124:3: note: candidate: 'constexpr wifi_init_config_t& wifi_init_config_t::operator(wifi_init_config_t&&)'
/home/g3/esp-idf/components/esp_wifi/include/esp_wifi.h:124:3: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to 'wifi_init_config_t&&'
At global scope:
cc1plus: note: unrecognized command-line option '-Wno-unknown-warning-option' may have been intended to silence earlier diagnostics
ninja: build stopped: subcommand failed.
[886/1807] Building C object esp-...ble/nimble/host/src/ble_gap.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/g3/esp-matter/examples/light/build/log/idf_py_stderr_output_31012 and /home/g3/esp-matter/examples/light/build/log/idf_py_stdout_output_31012
```

shubhamdp, 2025-02-26:
@TGG-3 can you please share the below details

- ESP-Matter Commit Id:
- ESP-IDF Commit Id:
- SoC (eg: ESP32 or ESP32-C3):
- Logs (Please attach the log file):
- Host Machine OS:
- Host Machine Python version:
- CMake version:

shubhamdp, 2025-03-18:
@TGG-3 any update on this?