espressif/esp-matter #1298: Matter Bridge distinguishes the connected devices

MaplestoryAlen, 2025-02-27:
I currently have two devices connected through a bridge. Device A is EP3 (thermostat) and EP4 (fan), while device B is EP5 (thermostat) and EP5 (fan). At this point, I don't know how to distinguish that EP3 and EP4 are physical devices, while EP5 and EP6 are another physical device.

esp_err_t create_bridge_devices(esp_matter::endpoint_t *ep, uint32_t device_type_id, void *priv_data)
{
    esp_err_t err = ESP_OK;

    ESP_LOGI("bridge", "Endpoint ID: %d", *ep);
    ESP_LOGI("bridge", "Device Type ID: %d", device_type_id);
    switch (device_type_id) {
    case ESP_MATTER_ON_OFF_LIGHT_DEVICE_TYPE_ID: {
        on_off_light::config_t on_off_light_conf;
        err = on_off_light::add(ep, &on_off_light_conf);
        break;
    }
    case ESP_MATTER_DIMMABLE_LIGHT_DEVICE_TYPE_ID: {
        dimmable_light::config_t dimmable_light_conf;
        err = dimmable_light::add(ep, &dimmable_light_conf);
        break;
    }
    case ESP_MATTER_COLOR_TEMPERATURE_LIGHT_DEVICE_TYPE_ID: {
        color_temperature_light::config_t color_temperature_light_conf;
        err = color_temperature_light::add(ep, &color_temperature_light_conf);
        break;
    }
    case ESP_MATTER_EXTENDED_COLOR_LIGHT_DEVICE_TYPE_ID: {
        extended_color_light::config_t extended_color_light_conf;
        err = extended_color_light::add(ep, &extended_color_light_conf);
        break;
    }
    case ESP_MATTER_ON_OFF_SWITCH_DEVICE_TYPE_ID: {
        on_off_switch::config_t switch_config;
        err = on_off_switch::add(ep, &switch_config);
        break;
    }
    case ESP_MATTER_THERMOSTAT_DEVICE_TYPE_ID:
    	{
          thermostat::config_t thermostat;
          err = thermostat::add(ep, &thermostat);
          break;  	  
    	}
	case ESP_MATTER_FAN_DEVICE_TYPE_ID:
    	{
          fan::config_t fan;
          err = fan::add(ep, &fan);
          break;  	  
    	}
	case ESP_MATTER_HUMIDITY_SENSOR_DEVICE_TYPE_ID: 
		{
		   humidity_sensor::config_t humidity_sensor;
           err = humidity_sensor::add(ep, &humidity_sensor);
           break;  	
		}
	case ESP_MATTER_ROOM_AIR_CONDITIONER_DEVICE_TYPE_ID:
		{
		   room_air_conditioner::config_t room_air_conditioner;
           err = room_air_conditioner::add(ep, &room_air_conditioner);
           break;  	
		}	   
    default: {
        ESP_LOGE(TAG, "Unsupported bridged matter device type");
        return ESP_ERR_INVALID_ARG;
             }
    }
    return err;
}

void UpdateEspnowDevice(uint8_t addr,uint32_t get_device_type)
{

	espnow_ctrl_bind_info_t bind_info = 
	{
	  .mac = {0x1a,0x2b,0x3c,0x4a,0x5b,addr},	
	  .initiator_attribute = ESPNOW_ATTRIBUTE_BASE,
	};
    if(get_device_type == THERMOSTAT_DEVICE_TYPE)
    {
	   matter_device_type_id = ESP_MATTER_THERMOSTAT_DEVICE_TYPE_ID;
    }
	else if(get_device_type == FAN_DEVICE_TYPE)
	{
	   matter_device_type_id = ESP_MATTER_FAN_DEVICE_TYPE_ID;
	}
	espnow_bridge_match_bridged_switch(bind_info.mac, bind_info.initiator_attribute,matter_device_type_id); 
}

Please take a look at the picture below. How should I display these three devices at this time? These devices are not BLE mesh and Zigbee, they are just ordinary devices

[image]

jonsmirl, 2025-02-28:
You need to use the TAGLIST feature on the Descriptor cluster to differentiate them.

MaplestoryAlen, 2025-03-05:
> You need to use the TAGLIST feature on the Descriptor cluster to differentiate them. You can add serial number or whatever to the taglist.

Thank you very much

closed 2025-03-05 by MaplestoryAlen