espressif/esp-matter #1300: How to detect online Thread end devices from a Matter controller application

pavel808, 2025-02-27:
I have a Matter controller application based on the provided Matter controller example application running on an ESP Thread Border Router device.  

Thread end devices get commissioned via this application via the Thread Border Router and matter commands can be sent successfully. 

However, how can I detect which end devices are actually powered on / online? Is there a way to ping the devices via some API. I can't seem to find this solution. Thanks.

jonsmirl, 2025-02-28:
Use MDNS to look for their entries. You need to know their compressed fabric ID and node ID.  You can do this same browse from the ESP MDNS API.

~~~
jonsmirl@ares:~$ avahi-browse --all | grep matter
+ enp3s0f1 IPv6 950545243CFE2D15-00000000FE0D396C             _matter._tcp         local
+ enp3s0f1 IPv6 9CC0373667C4F408-204B50A898A38F65             _matter._tcp         local
+ enp3s0f1 IPv4 C7944E937AA07DD5-008462EE7296AD71             _matter._tcp         local
+ enp3s0f1 IPv4 C7944E937AA07DD5-0083161AE5399201             _matter._tcp         local
+ enp3s0f1 IPv4 C7944E937AA07DD5-00867AF748BEBA01             _matter._tcp         local
+ enp3s0f1 IPv4 950545243CFE2D15-00000000FE0D396C             _matter._tcp         local
+ enp3s0f1 IPv4 9CC0373667C4F408-204B50A898A38F65             _matter._tcp         local
+ enp3s0f1 IPv6 C7944E937AA07DD5-0083161AE5399201             _matter._tcp         local
+ enp3s0f1 IPv6 C7944E937AA07DD5-00867AF748BEBA01             _matter._tcp         local
+ enp3s0f1 IPv6 C7944E937AA07DD5-008462EE7296AD71             _matter._tcp         local
~~~

pavel808, 2025-03-02:
@jonsmirl Thanks for your reply. The end  devices are commissioned and communicate via Thread though. Only the Thread border Router gets assigned an IP address and mDNS entry.

jonsmirl, 2025-03-02:
Try the Linux avahi-browse command and you should see them.
jonsmirl@ares:~$ avahi-browse --all | grep matter
There is a similar command for MacOS

You can also get MDNS free browser apps for your phone which can see them

pavel808, 2025-03-03:
@jonsmirl Ah yes. Indeed I can see the devices using `avahi-browse --all | grep matter`. Thanks.

I am trying to figure how to use the the ESP mDNS Service to lookup the Matter Node IDs from within my Matter controller application.

jonsmirl, 2025-03-03:
use mdns_query_ptr()

wqx6, 2025-03-12:
Normally the controller will send subscription command to the end-device to check whether it is online/offline. The device will report to the controller periodically, if the controller doesn't receive the report for a longer time than on period, the device can considered to be offline.

jonsmirl, 2025-03-12:
Yes, there are two different scenarios. 1) you know about the device and just want to monitor it's online state. In that case use the subscription.  2) there are multiple commissioners so the second controller is unaware of newly commissioned devices. In that cases you need to use MDNS.

MDNS does work as a very low overhead way of determining if a known device is online, but it will take a few minutes to notice the device is offline whereas a subscription will know immediately.