espressif/esp-matter #1301: OperationalSessionSetup[3:00000000E226355C]: operational discovery failed: 32

bilalahmaddev, 2025-02-28:
Why my commissioned device received these errors when Matter app is not opened: 

[image]

bilalahmaddev, 2025-02-28:
Where can I enable this to make session permanent?

jonsmirl, 2025-02-28:
auto_resubscribe on esp
~~~
    subscribe_command(uint64_t node_id, ScopedMemoryBufferWithSize<AttributePathParams> &&attr_paths,
                      ScopedMemoryBufferWithSize<EventPathParams> &&event_paths, uint16_t min_interval,
                      uint16_t max_interval, bool auto_resubscribe = true, attribute_report_cb_t attribute_cb = nullptr,
                      event_report_cb_t event_cb = nullptr, subscribe_done_cb_t done_cb = nullptr,
                      subscribe_failure_cb_t connect_failure_cb = nullptr)
        : m_node_id(node_id)
        , m_min_interval(min_interval)
        , m_max_interval(max_interval)
        , m_auto_resubscribe(auto_resubscribe)
        , m_buffered_read_cb(*this)
        , m_attr_paths(std::move(attr_paths))
        , m_event_paths(std::move(event_paths))
        , on_device_connected_cb(on_device_connected_fcn, this)
        , on_device_connection_failure_cb(on_device_connection_failure_fcn, this)
        , attribute_data_cb(attribute_cb)
        , event_data_cb(event_cb)
        , subscribe_done_cb(done_cb)
        , subscribe_failure_cb(connect_failure_cb)
    {
~~~

wqx6, 2025-03-12:
Hi, there is a feature named subscription resumption on the end-devices which helps the end-devices establishing CASE session with the controller/commissioner and reporting their statuses to the controller/commissioner. This could decrease the rate of the no-response error.

Looks like you are using a special Matter app to commission the end-device. And that Matter-app will send subscription request to the end-device after commissioning. The end-device will store that subscription and try to resume it on reboot. It will discover the Matter app over mDNS. If your Matter app is not open, it will not advertise mDNS service, so the end-device fails to discover the Matter app and resume subscription.

dhrishi, 2025-06-06:
@bilalahmaddev Is there any update you can share? Please close this issue if resolved

dhrishi, 2025-09-13:
Closing as the questions are answered

closed 2025-09-13 by dhrishi