espressif/esp-matter #1302: Installation of esp-matter failure

emesamwong, 2025-02-28:
**Describe the bug**
A clear and concise description of what the bug is.
When I try to install the esp-matter, there is some abnormal message during Python setup stage. please refer to the esp-matter error log.txt

I checked the esp-matter & connectedhomeip status are clean
[image]

How could I do to solve this problem?

**Environment**
- ESP-Matter Commit Id: release/v1.1 (#548ef58)  
- Submodule connectedhomeip commit Id: #83f4163
- ESP-IDF Commit Id:N/a
- SoC (eg: ESP32 or ESP32-C3):N/a
- Device Logs (Please attach the log file):

[esp-matter error log.txt](https://github.com/user-attachments/files/19030951/esp-matter.error.log.txt)

- Host Machine OS: linux
- Host Machine Python version: 3.10

- Commissioner app and versions if present: N/a
- Home hub app and versions if present:N/a
- Commissioner's logs if present:N/a

**Any additional details**
...

emesamwong, 2025-03-01:
For more information, I upload the clone log, and cmd that I executed, please help to review any steps is abnormal. 

[check hash log.txt](https://github.com/user-attachments/files/19035529/check.hash.log.txt)
[clone esp-matter log.txt](https://github.com/user-attachments/files/19035530/clone.esp-matter.log.txt)
[install log.txt](https://github.com/user-attachments/files/19035528/install.log.txt)

the linux environment is below:

[image]

emesamwong, 2025-03-03:
@shubhamdp Thank you for your reminder, but the problems have not been fixed. 
Here is the Log:

[esp-matter install fail log (installed esp-idf).txt](https://github.com/user-attachments/files/19050840/esp-matter.install.fail.log.installed.esp-idf.txt)

[image]

I checked that the /.environment folder has been removed before running ./install.sh

[image]
The prerequisites on Linux cmd were also executed, but nothing was upgraded. 

For more information, I tried to run ./install.sh in esp-matter release/v1.2, which was successful to install the environment, and switch back to release/v1.1 ((#548ef58 for esp-matter &  #83f4163 for Submodule connectedhomeip) but failed to build the project with the below error:

[image]

emesamwong, 2025-03-03:
Sorry for the missing info of esp-idf for the "esp-matter install fail log (installed esp-idf).txt"
The esp-idf commit : v5.1 (# cbce221e88d52665523093b2b6dd0ebe3f1243f1)

andy31415, 2025-03-06:
From my checks on the connectedhomeip sdk side:
  - this is using a very old release of the SDK (we recommend at least 1.3 due to many critical bug fixes, this is on 1.1)
  - historically there have been issues with gdbgui requiring gevent which would not compile, so the workaround can be to remove the gdbgui requirements from both espressif and the sdk (gdbgui is not required to compile applications)

shubhamdp, 2025-03-12:
@emesamwong when running the bootstrap on the fresh clone of connectedhomeip's v1.1-branch I had to do following tweaks and it went through, see if this patch can help you get through this, copy this patch into a file and do a git apply.

```
diff --git a/scripts/setup/constraints.txt b/scripts/setup/constraints.txt
index c722ba1b..531dd3d9 100644
--- a/scripts/setup/constraints.txt
+++ b/scripts/setup/constraints.txt
@@ -44,7 +44,8 @@ cbor2==5.4.6
     #   imgtool
 certifi==2022.12.7
     # via requests
-cffi==1.15.1
+cffi==1.15.1; python_version < "3.13"
+cffi==1.17.1; python_version >= "3.13"
     # via cryptography
 charset-normalizer==3.0.1
     # via requests
@@ -65,7 +66,7 @@ colorama==0.4.6
     #   west
 coloredlogs==15.0.1
     # via -r requirements.txt
-construct==2.10.54
+construct==2.10.70
     # via -r requirements.esp32.txt
 contextlib2==21.6.0
     # via
@@ -195,7 +196,7 @@ mypy==0.971
     # via -r requirements.txt
 mypy-extensions==1.0.0
     # via mypy
-mypy-protobuf==3.2.0
+mypy-protobuf==3.5.0
     # via -r requirements.txt
 numpy==1.24.2
     # via pandas
@@ -236,7 +237,7 @@ portpicker==1.5.2
 prettytable==2.5.0
 prompt-toolkit==3.0.43
     # via ipython
-protobuf==3.20.1
+protobuf==4.24.4
     # via
     #   -r requirements.txt
     #   mypy-protobuf
@@ -315,7 +316,7 @@ pyudev==0.24.0 ; platform_machine != "aarch64" and sys_platform == "linux"
     # via
     #   -r requirements.mbed.txt
     #   mbed-tools
-pyyaml==6.0
+pyyaml==6.0.1
     # via
     #   idf-component-manager
     #   mobly
@@ -383,7 +384,7 @@ traitlets==5.9.0
     # via
     #   ipython
     #   matplotlib-inline
-types-protobuf==3.19.22
+types-protobuf==4.24.0.2
     # via
     #   -r requirements.txt
     #   mypy-protobuf
diff --git a/scripts/setup/requirements.esp32.txt b/scripts/setup/requirements.esp32.txt
index 7c4eac1c..08542016 100644
--- a/scripts/setup/requirements.esp32.txt
+++ b/scripts/setup/requirements.esp32.txt
@@ -7,7 +7,7 @@ reedsolo>=1.5.3,<=1.5.4
 bitstring>=3.1.6
 ecdsa>=0.16.0
 kconfiglib==13.7.1
-construct==2.10.54
+construct==2.10.70
 python-socketio<5
 itsdangerous<2.1 ; python_version < "3.11"
 #
diff --git a/scripts/setup/requirements.txt b/scripts/setup/requirements.txt
index 53d0037d..e5ab3d31 100644
--- a/scripts/setup/requirements.txt
+++ b/scripts/setup/requirements.txt
@@ -56,9 +56,9 @@ coloredlogs
 watchdog
 build==0.8.0
 mypy==0.971
-mypy-protobuf==3.2.0
-protobuf==3.20.1
-types-protobuf==3.19.22
+mypy-protobuf==3.5.0
+protobuf==4.24.4
+types-protobuf==4.24.0.2
 
 # scripts/tools/memory
 anytree
```

```
cd esp-matter/connectedhomeip/connectedhomeip
rm -rf .environment
git apply /tmp/setup.patch
cd -

# since we moved to higher protobuf we need this so that pure python parsing is used, but this is slower (toll for using older version)
export PROTOCOL_BUFFERS_PYTHON_IMPLEMENTATION=python

./install.sh
```

Let me know if it works for you or not. But as suggested my @andy31415  you should use the newer release.

emesamwong, 2025-03-16:
@shubhamdp Thank you for your support. I tried to apply the patch however it contains error when execute "git apply setup.patch"

I would like to show my steps:
1. create a new text file, copy the patch content into the text file and save it as setup.patch
2. execute the "git apply [path]/setup.patch" after removed .environment inside the connectedhomeip

Here is the log of error:
[image]

And the setup.patch location:
[image]

shubhamdp, 2025-03-18:
Can you please check the connectedhomeip revision that you used? I used the latest v1.1-branch(9bceea9fc468ae92b3fe64e665f82a78799390de) to generate the patch.

If I click on copy to clipboard button on that text box, paste it inside the file and run git apply, it just works.

dhrishi, 2025-04-17:
@emesamwong Please close the issue if resolved

closed 2025-06-06 by dhrishi