espressif/esp-matter #1303: linking CXX executable failure light.elf

ravencarcass, 2025-03-02:
Hello, 
I am having some problems building a program.  I want to change an example of Matter: light. I am running this on an Ubuntu OS and have changed the program to VSCode. I first built the light example with no problems and then I made some changes and then gave me the following things. I've already searched on many fora. But did find some who had the same issue but did not work for my failure.

VSCode reports the following in the terminal: 
```
[1390/1393] Linking CXX executable light.elf
FAILED: light.elf 
: && /home/robbe/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/riscv32-esp-elf-g++ -march=rv32imac_zicsr_zifencei -nostartfiles -march=rv32imac_zicsr_zifencei --specs=nosys.specs    -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32C6=0 -Wl,--Map=/home/robbe/esp/light/build/light.map -Wl,--no-warn-rwx-segments -Wl,--orphan-handling=warn -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T rom.api.ld -T esp32c6.peripherals.ld -T esp32c6.rom.ld -T esp32c6.rom.api.ld -T esp32c6.rom.rvfp.ld -T esp32c6.rom.wdt.ld -T esp32c6.rom.systimer.ld -T esp32c6.rom.version.ld -T esp32c6.rom.phy.ld -T esp32c6.rom.coexist.ld -T esp32c6.rom.net80211.ld -T esp32c6.rom.pp.ld -T esp32c6.rom.newlib.ld -T esp32c6.rom.newlib-normal.ld -T esp32c6.rom.heap.ld -T memory.ld -T sections.ld CMakeFiles/light.elf.dir/project_elf_src_esp32c6.c.obj -o light.elf -L/home/robbe/esp/esp-idf/components/riscv/ld   -L/home/robbe/esp/esp-idf/components/soc/esp32c6/ld   -L/home/robbe/esp/esp-idf/components/esp_rom/esp32c6/ld   -L/home/robbe/esp/light/build/esp-idf/esp_system/ld   -L/home/robbe/esp/esp-idf/components/esp_phy/lib/esp32c6   -L/home/robbe/esp/esp-idf/components/esp_coex/lib/esp32c6   -L/home/robbe/esp/esp-idf/components/esp_wifi/lib/esp32c6 esp-idf/riscv/libriscv.a  esp-idf/esp_driver_gpio/libesp_driver_gpio.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/esp_bootloader_format/libesp_bootloader_format.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/efuse/libefuse.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_security/libesp_security.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/pthread/libpthread.a  esp-idf/cxx/libcxx.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/esp_driver_gptimer/libesp_driver_gptimer.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/esp_driver_uart/libesp_driver_uart.a  esp-idf/app_trace/libapp_trace.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_driver_pcnt/libesp_driver_pcnt.a  esp-idf/esp_driver_spi/libesp_driver_spi.a  esp-idf/esp_driver_mcpwm/libesp_driver_mcpwm.a  esp-idf/esp_driver_i2s/libesp_driver_i2s.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_driver_sdspi/libesp_driver_sdspi.a  esp-idf/esp_driver_sdio/libesp_driver_sdio.a  esp-idf/esp_driver_rmt/libesp_driver_rmt.a  esp-idf/esp_driver_tsens/libesp_driver_tsens.a  esp-idf/esp_driver_sdm/libesp_driver_sdm.a  esp-idf/esp_driver_i2c/libesp_driver_i2c.a  esp-idf/esp_driver_ledc/libesp_driver_ledc.a  esp-idf/esp_driver_parlio/libesp_driver_parlio.a  esp-idf/esp_driver_usb_serial_jtag/libesp_driver_usb_serial_jtag.a  esp-idf/driver/libdriver.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/esp_vfs_console/libesp_vfs_console.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_coex/libesp_coex.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/bt/libbt.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/console/libconsole.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_driver_cam/libesp_driver_cam.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/esp_https_server/libesp_https_server.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/fatfs/libfatfs.a  esp-idf/ieee802154/libieee802154.a  esp-idf/json/libjson.a  esp-idf/mqtt/libmqtt.a  esp-idf/nvs_sec_provider/libnvs_sec_provider.a  esp-idf/rt/librt.a  esp-idf/spiffs/libspiffs.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/espressif__button/libespressif__button.a  esp-idf/espressif__led_strip/libespressif__led_strip.a  esp-idf/espressif__led_indicator/libespressif__led_indicator.a  esp-idf/espressif__esp_bsp_devkit/libespressif__esp_bsp_devkit.a  esp-idf/main/libmain.a  esp-idf/espressif__esp_delta_ota/libespressif__esp_delta_ota.a  esp-idf/espressif__esp_encrypted_img/libespressif__esp_encrypted_img.a  esp-idf/espressif__rmaker_common/libespressif__rmaker_common.a  esp-idf/espressif__esp_diagnostics/libespressif__esp_diagnostics.a  esp-idf/espressif__cbor/libespressif__cbor.a  esp-idf/espressif__esp_diag_data_store/libespressif__esp_diag_data_store.a  esp-idf/espressif__esp_insights/libespressif__esp_insights.a  esp-idf/espressif__mdns/libespressif__mdns.a  esp-idf/espressif__esp-serial-flasher/libespressif__esp-serial-flasher.a  esp-idf/espressif__esp_rcp_update/libespressif__esp_rcp_update.a  esp-idf/espressif__esp_secure_cert_mgr/libespressif__esp_secure_cert_mgr.a  esp-idf/chip/libchip.a  esp-idf/esp_matter_console/libesp_matter_console.a  esp-idf/esp_matter/libesp_matter.a  esp-idf/esp_matter_bridge/libesp_matter_bridge.a  esp-idf/app_bridge/libapp_bridge.a  esp-idf/app_reset/libapp_reset.a  esp-idf/espressif__json_generator/libespressif__json_generator.a  esp-idf/espressif__json_parser/libespressif__json_parser.a  esp-idf/app_trace/libapp_trace.a  esp-idf/app_trace/libapp_trace.a  esp-idf/cmock/libcmock.a  esp-idf/unity/libunity.a  esp-idf/esp_driver_cam/libesp_driver_cam.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/esp_https_server/libesp_https_server.a  esp-idf/ieee802154/libieee802154.a  esp-idf/nvs_sec_provider/libnvs_sec_provider.a  -u nvs_sec_provider_include_impl  esp-idf/rt/librt.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/protocomm/libprotocomm.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/espressif__esp_bsp_devkit/libespressif__esp_bsp_devkit.a  esp-idf/espressif__led_indicator/libespressif__led_indicator.a  esp-idf/espressif__led_strip/libespressif__led_strip.a  esp-idf/esp_matter_bridge/libesp_matter_bridge.a  esp-idf/espressif__button/libespressif__button.a  esp-idf/spiffs/libspiffs.a  esp-idf/espressif__json_parser/libespressif__json_parser.a  esp-idf/esp_matter_console/libesp_matter_console.a  esp-idf/esp_matter/libesp_matter.a  esp-idf/esp_matter_console/libesp_matter_console.a  esp-idf/esp_matter/libesp_matter.a  esp-idf/json/libjson.a  esp-idf/chip/libchip.a  esp-idf/fatfs/libfatfs.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/espressif__esp_delta_ota/libespressif__esp_delta_ota.a  esp-idf/espressif__esp_encrypted_img/libespressif__esp_encrypted_img.a  esp-idf/espressif__esp_insights/libespressif__esp_insights.a  esp-idf/espcoredump/libespcoredump.a  -u esp_system_include_coredump_init  esp-idf/espressif__esp_diagnostics/libespressif__esp_diagnostics.a  esp-idf/espressif__rmaker_common/libespressif__rmaker_common.a  esp-idf/mqtt/libmqtt.a  esp-idf/espressif__cbor/libespressif__cbor.a  esp-idf/espressif__esp_diag_data_store/libespressif__esp_diag_data_store.a  esp-idf/espressif__esp_rcp_update/libespressif__esp_rcp_update.a  esp-idf/espressif__esp-serial-flasher/libespressif__esp-serial-flasher.a  esp-idf/espressif__esp_secure_cert_mgr/libespressif__esp_secure_cert_mgr.a  -Wl,--start-group  esp-idf/chip/lib/libCHIP.a  esp-idf/chip/lib/libCHIP.a  esp-idf/bt/libbt.a  /home/robbe/esp/esp-idf/components/bt/controller/lib_esp32c6/esp32c6-bt-lib/esp32c6/libble_app.a  esp-idf/espressif__mdns/libespressif__mdns.a  esp-idf/console/libconsole.a  esp-idf/esp_eth/libesp_e [... cut ...]
/home/robbe/esp/light/main/app_driver.cpp:59:(.text._Z27app_driver_attribute_updatePvtmmP21esp_matter_attr_val_t+0x4): undefined reference to `light_endpoint_id'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

 *  The terminal process "ninja" terminated with exit code: 1. 
```

I've already tried the following thing and didn't gave me a solusion: "use external log wrapper enabled"
It become a little frustrating that it don't work. I'm already searching 4 days for a solusions and can't find it.

shubhamdp, 2025-03-03:
@ravencarcass did you modify the source code of light example? If yes, you may have removed the definition of `light_endpoint_id` which is causing the problem. As its storage type is `extern`, compilation stage is successful because `declaration` is present in your code but its missing `definition` which assigns the storage for that variable, (C basics). 

For fixing this, I'd suggest you to do this on unmodified light example and change your source code according to them. Please check this I grep the `light_endpoint_id` on examples/light and here are the results, https://github.com/search?q=repo%3Aespressif%2Fesp-matter%20path%3A%2F%5Eexamples%5C%2Flight%5C%2F%2F%20light_endpoint_id&type=code, please check if you have the same in your code.

ravencarcass, 2025-03-03:
Hello, 
After I just paste the failure code in here did I saw the light_endpoint_id end escaped it in the code and it indeed solved the failure.

closed 2025-03-03 by ravencarcass