espressif/esp-matter #1304: Want to control Smart TV with NimBLE used in esp-matter code

Ajay2803, 2025-03-03:
Hi,

I am using esp-matter light example, for my development. I know matter uses Nimble stack, because it uses ble only for commissioning. But in my case i want to use ble for other purpose, like to control Smart TV's using HID Profile after commissioning and I got stuck at this point.

I have tried many different examples from esp-idf (Bluedroid and Nimble) and suggestions from esp-matter issues, but i am unable to solve this problem.

please Help me step by step to achieve this, and also let me know that if i'll do this kind of customisation will it break any matter law's or may issue while matter certification.
Or If you have any alternate way please suggest me, so i can continue with my development.

wqx6, 2025-03-07:
Are you wanting to use BLE after Matter commissioning? If yes, you should disable `CONFIG_USE_BLE_ONLY_FOR_COMMISSIONING` in menuconfig to prevent BLE from deinitialization after matter commissioning. And you should also use the [external platform](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html?highlight=external#advanced-setup) to add your own BLE codes to use HID profile.

Ajay2803, 2025-03-07:
Thankyou for your reply,

yes i want to use BLE after commissioning, I have already disabled CONFIG_USE_BLE_ONLY_FOR_COMMISSIONING. can you clarify what do you mean by External Platforms? I am unable to understand this.

wqx6, 2025-03-07:
> Can you clarify what do you mean by External Platforms? I am unable to understand this.

We will initialize BLE stack and advertise Matter-over-BLE services and handle Matter data in our platform codes https://github.com/project-chip/connectedhomeip/tree/master/src/platform/ESP32/nimble, but you are use BLE with HID profile to control Smart TV, so you must modify the BLE platform codes to support both Matter commissioning and HID Profile. And the external platform is the solution we provide to override the original platform codes. you can refer to https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html?#advanced-setup about how to use it.

Ajay2803, 2025-03-10:
Sorry, But we don't want to use this way. 

I am sharing my sample code, please go through it and give me some suggestions. I am able to turn on BLE and it is visible in TV also but i am not able to connect it. Can you please help me to connect and control the TV using this way?

and also please reply to this question:
(please Help me step by step to achieve this, and also let me know that if i'll do this kind of customisation will it break any matter law's or may issue while matter certification.)

[example_code.zip](https://github.com/user-attachments/files/19160486/example_code.zip)

Ajay2803, 2025-03-12:
Hi @wqx6,

-Thank you for your valuable Reply.
-After you said that we should try to add access callback and try it with nrfconnect app, I have tried and it works with nrfconnect app, but as per our requirement we need to connect it to TV, Device is visible in TV But when we try to connect it, it fails.

-This are the LOG' s can you please help us to solve the problem:
 (142811) app_main: BLE Advertising Started Successfully
 (157671) app_main: connection established; status=0 
 (160821) app_main: disconnect; reason=531 
 (160821) NimBLE: GAP procedure initiated: advertise; 
 (160821) NimBLE: disc_mode=2
 (160831) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
 (160841) NimBLE:

wqx6, 2025-03-12:
> After you said that we should try to add access callback and try it with nrfconnect app, I have tried and it works with nrfconnect app

Did you check the service after connecting the BLE with nrfconnect app? Could you find the gatt service with the right UUID that you added?  Normally there should be access_cb for the characteristics so that the device can handle the message sent by the peer(the TV in your scenario) and send the message to the peer in the access_cb. Did you add the logic in your access_cb? This might be the reason of BLE disconnection.

Ajay2803, 2025-03-12:
[image]

This is my image from NRF Connect apk, I have registered HID service with UUID 1812 that is not visible yet, and yes i have added access_cb for the characteristics.Yes i have also added logic in access_cb. 

Why my service is not visible yet, can you help me?

dhrishi, 2025-03-13:
@Ajay2803 

>Why my service is not visible yet, can you help me?

The questions seem to be specific to BLE. Can you please close this issue and ask the specifics on ESP IDF Github instead?

Ajay2803, 2025-03-17:
Hi @wqx6,

As you said that we need to use external platform feature to add our custom BLE services, I have not any kind of idea related to this feature can you guide me how can i use this feature and add my custom service t BLE.

I have tried https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html?#advanced-setup, but not able to understand what to do with this, if you have any references related to this or if anyone have added custom service using this feature there reference, any patch or something you can provide would be very helpful for us.

wqx6, 2025-03-20:
You can refer to the example [light_wifi_prov](https://github.com/espressif/esp-matter/tree/main/examples/light_wifi_prov) about how to add custom GATT services with external platform, or you can apply the patch https://github.com/project-chip/connectedhomeip/pull/38067 which adds a API so that you can add the custom GATT services.

dhrishi, 2025-05-14:
@Ajay2803 Please check the [light_network_prov example]( https://github.com/espressif/esp-matter/tree/main/examples/light_network_prov) for the latest changes that do not require to use the external platform

Ajay2803, 2025-05-15:
Thank-you @dhrishi this solution, But we have found alternate way to do this, but for this solution thank-you, i'll try this solution.

closed 2025-05-15 by Ajay2803

dhrishi, 2025-05-15:
@Ajay2803 Thanks for the update. Can you please close any other related issues around the same?