espressif/esp-matter #1305: Harness test MoveToLevel TC-LVL-3.1 fail

crivi30, 2025-03-03:
**Describe the bug**  
The test case **TC-LVL-3.1 MoveToLevel** fails in three steps (5g, 5i, 6d).
Here the log of the complete harness test: [failed_MoveToLevel_2025_03_03_11_10_17.log](https://github.com/user-attachments/files/19054535/failed_MoveToLevel_2025_03_03_11_10_17.log)

Failure logs:  
```
WARNING    | 2025-03-03 11:11:31.899197 | Test Failure: The test expectation "value (120) == 100" is false  
WARNING    | 2025-03-03 11:11:33.123481 | Test Failure: The test expectation "value (140) == 100" is false  
WARNING    | 2025-03-03 11:11:41.100635 | Test Failure: The test expectation "value (140) == 120" is false  
```
The issue occurs because the **MoveToLevel** command in the previous step was sent with **OptionsOverride = 0x00**. According to the test, the corresponding attribute should **not** be updated.  
However, the ESP-Matter SDK **does update the attribute**, which causes the following **read steps (5g, 5i, 6d) to fail** .

I would like to understand:  
- Why does the ESP-Matter SDK update the attribute even when **OptionsOverride = 0x00**?  
- Is this behavior controlled by the Matter SDK, or do we need to handle this differently in our implementation?  

**Environment**  
- **ESP-Matter Version**: 1.4
- **SoC**: ESP32  
- **Device Logs**: Disabled

Thank you in advance!

Jerry-ESP, 2025-03-04:
[TC-LVL-3.1.log](https://github.com/user-attachments/files/19068492/TC-LVL-3.1.log) 
I test on my side with the light example, it passed, please refer to the attachment.

Which example you use for testing?

crivi30, 2025-03-04:
I'm not using a specific example. I'm creating an extended color light this way:

```
extended_color_light::config_t light_config;
light_config.on_off.on_off = false;
light_config.on_off.lighting.start_up_on_off = nullptr;
light_config.level_control.current_level = nullptr;
light_config.level_control.on_level = nullptr;
light_config.level_control.lighting.start_up_current_level = nullptr;
light_config.color_control.color_mode = (uint8_t)ColorControl::ColorMode::kCurrentHueAndCurrentSaturation;
light_config.color_control.enhanced_color_mode = (uint8_t)ColorControl::ColorMode::kCurrentHueAndCurrentSaturation;
light_config.color_control.color_temperature.startup_color_temperature_mireds = nullptr;

app_light_color->endpoint = extended_color_light::create(node, &light_config, ENDPOINT_FLAG_NONE, NULL);
```

Maybe I initialized something incorrectly?

jadhavrohit924, 2025-03-04:
@crivi30 From your logs, `5f` was not executed so remaining tests failed.
```
Executing Test Step: Step 5f: TH sends Off command to DUT
Test Step Not Applicable: Step 5f: TH sends Off command to DUT - Test step skipped: Step 5f: TH sends Off command to DUT. OO.S.C00.Rsp == False
```
Can you please share your PICS and the version you are using in the test harness? We are suspecting `OO.S.C00.Rsp` is not set in your PICS file.

crivi30, 2025-03-04:
I'm testing the PICS:

[Level_Control_Cluster_Test_Plan.zip](https://github.com/user-attachments/files/19070100/Level_Control_Cluster_Test_Plan.zip)

with the version v2.11+fall2024. I confirm that in this PICS list i don't have `OO.S.C00.Rsp`

jadhavrohit924, 2025-03-04:
`OO.S.C00.Rsp` will be in the PICS of On/Off cluster. Please share the whole bunch of PICS files you have uploaded to test TC-LVL-3.1.

VaishaliAvhale, 2025-03-04:
Hi @crivi30,

I have tested the same test case on Extended Color Light, and it passed on my end. The failure on your end is due to the On/Off Cluster's PICS being missing. Level Control and On/Off Cluster's PICS are mandatory for this test case. Please upload On/Off Cluster PICS and retest the test case.

crivi30, 2025-03-04:
ok, I uploaded the On/Off Cluster PICS and retested. It passed successfully now

Thank you all for your support! 
Best

closed 2025-03-04 by crivi30