espressif/esp-matter #1306: Issue: Zigbee Bridge Example – Only One Device Connects; Unable to Remove Device in Home Assistant

iamiha, 2025-03-04:
### Description  

I am working with the Zigbee bridge example and encountering two issues:  

#### 1. Only One Device Connects  
- I have flashed two ESP devices with the `HA_on_off_light` example program. However, only one device successfully connects; the other does not.  
- I added a log statement to print `MAX_BRIDGED_DEVICE_COUNT`, which shows a value of `14`, indicating that the bridge should support multiple devices.  

#### 2. Unable to Remove Connected Device in Home Assistant  
- When attempting to remove the connected device from Home Assistant, I receive the error:  
  ```
  ERROR Deleting device: failed to remove device entry, rejected by integration.
  ```
- I have attached an image file illustrating this error message.  

### Steps to Reproduce  
1. Flash two ESP devices with the `HA_on_off_light` example program.  
2. Observe that only one device connects to the Zigbee bridge.  
3. Attempt to remove the connected device from Home Assistant and encounter the error message.  

### Expected Behavior  
- Both ESP devices should connect to the Zigbee bridge without issues.  
- Devices should be removable from Home Assistant without encountering errors.  

### Relevant Logs  
- Log output showing `MAX_BRIDGED_DEVICE_COUNT: 14`.  
- Screenshot of the Home Assistant error message when attempting to delete the device.  

### Environment  
- **ESP-IDF Version:** v5.2.3  
- **ESP-Matter SDK Version:** 2.0  
- **Home Assistant Version:** 2024.12.1  
- **Hardware Used:**  
  - **Zigbee Bridge:** Zigbee Gateway V1.2 board  
  - **Zigbee Light:** ESP32-H2  

### Troubleshooting Steps Taken  
- Verified that both ESP devices are flashed correctly with the `HA_on_off_light` example.  
- Checked network connectivity and ensured that there are no conflicts.  
- Restarted Home Assistant and attempted the removal process multiple times.  

### Attachments  
- 
[image]

### Request for Assistance  
I would appreciate any guidance on:  
- Potential reasons why only one of the two ESP devices is connecting to the Zigbee bridge.  
- Steps to successfully remove or delete a connected device (using chiptool or home assistant) .  

Thank you for your support!

wqx6, 2025-03-07:
> Potential reasons why only one of the two ESP devices is connecting to the Zigbee bridge.

Did you see the following logs in the second HA_onoff_light?
```
I (571) ESP_ZB_ON_OFF_LIGHT: Start network steering
I (3661) ESP_ZB_ON_OFF_LIGHT: Joined network successfully (Extended PAN ID: 74:4d:bd:ff:fe:62:62:6d, PAN ID: 0x2b4a, Channel:13, Short Address: 0x526b)
```
If you didn't get the logs, the second light didn't join the zigbee network so the bridge cannot create Matter endpoint for it. If you saw the above logs, could you send us the bridge's log so that we can analyze why the bridge didn't create Matter endpoint for the zigbee device.

> Steps to successfully remove or delete a connected device (using chiptool or home assistant)

In fact, Matter controller/commissioner(chip-tool/home-assistant controller) has no way to delete a bridged device from a bridge. The bridged-devices management is only maintained by the bridge itself. Not sure why you can see the delete device button on your HA app, In my test, the `DELETE` button is unusable on iOS HA app.

iamiha, 2025-03-10:
solved !

closed 2025-03-10 by iamiha