espressif/esp-matter #1307: Harness test Reporting Requirements TC-LVL-2.3 fail

crivi30, 2025-03-04:
The problem occurs during the first step of the test TC-LVL-2.3, **"Start python test,"** which fails due to a missing `--endpoint` flag in the test command.  The test harness sends the following command:  
```sh
python3 /root/python_testing/scripts/sdk/test_harness_client.py sdk/TC_LVL_2_3 TC_LVL_2_3 --tests test_TC_LVL_2_3 --trace-to json:log --discriminator 1824 --passcode 81606343 --PICS /var/tmp/pics
```
However, the required `--endpoint` flag is missing, leading to test failure.

#### **Error Message:**
```
******************************************************************
*
* Test test_TC_LVL_2_3 failed for the following reason:
* Details=
The --endpoint flag is required for this test.
, Extras=None
*
* File "/usr/local/lib/python3.12/dist-packages/mobly/asserts.py", line 441, in fail
* , Extras=None
*
*******************************************************************
```

### **Environment**
- **Test Harness version**: v2.11+fall2024
- **ESP-Matter version**: 1.4  
- **SoC**: ESP32  
- **Device Logs**: Disabled

Jerry-ESP, 2025-03-04:
You can try to add the --endpoint param in the TH project configuration as below:
"test_parameters": {
  "endpoint": 1 
},

crivi30, 2025-03-04:
It works, thank you.

closed 2025-03-04 by crivi30