espressif/esp-matter #1308: Change the "setup pin code" for test project?

dzungpv, 2025-03-04:
I want to test same matter in the same network, but the problem all setup code is the same for test devices. I want it simple to test so not using chip-tool or mfg-tool to generate partition.

It is: 

- Discriminator: **3840**
- Setup-pin-code: **20202021**

But there is no config to change that. Seem in examples there is no option to change that.
And the code set by the parent chip code in: CHIP_DEVICE_CONFIG_USE_TEST_SETUP_PIN_CODE https://github.com/lucicop/connectedhomeip/blob/df2c8e9072714273caad5c60ed9ce1b4911cf662/config/standalone/CHIPProjectConfig.h#L42

I see that it include in the https://github.com/espressif/esp-matter/blob/a9602810135f2ebee80907b9075603d64d7d68e0/examples/controller/sdkconfig.defaults#L12 but seem not using anywhere.

Jerry-ESP, 2025-03-04:
You can change the discriminator here: https://github.com/project-chip/connectedhomeip/blob/13da787aeb078fb79587139aef0f266e9d2b7c63/src/include/platform/CHIPDeviceConfig.h#L1233, but for the setup pin code, you shall not change it with the  spake2p unchanged.

dzungpv, 2025-03-04:
> You can change the discriminator here: https://github.com/project-chip/connectedhomeip/blob/13da787aeb078fb79587139aef0f266e9d2b7c63/src/include/platform/CHIPDeviceConfig.h#L1233, but for the setup pin code, you shall not change it with the spake2p unchanged.

I have try that but when discriminator change from 3840 to 3850 code setup code gen using https://github.com/espressif/connectedhomeip/tree/9b8fffe0bb4e7ba7aac319f5905070a3476db7cb/src/setup_payload/python output the same:

[image]

So I know by reading the code, when changing setup pin code the  spake2p must be change. I see in the #760 seem the proper way to change that with code. 

But the is simple way to change CHIP_DEVICE_CONFIG_USE_TEST_SETUP_PIN_CODE and re calculate https://github.com/project-chip/connectedhomeip/blob/13da787aeb078fb79587139aef0f266e9d2b7c63/src/include/platform/CHIPDeviceConfig.h#L1282. I try change it but not working, may cause by build cache. You may add if in menu config and gen code for that, it is simpler to change the code when testing, rather simple than using mfg-tool to built the partition like in production.

dzungpv, 2025-03-05:
@Jerry-ESP I try the example light_switch project and enable:
CONFIG_CUSTOM_COMMISSIONABLE_DATA_PROVIDER=y
CONFIG_DYNAMIC_PASSCODE_COMMISSIONABLE_DATA_PROVIDER=y
To generate random setup code each time, but it is not working. Seem this example outdated.
I plan to use this and use 8 char off MAC address , but original code not working. Log:

```
I (761) NimBLE: Failed to restore IRKs from store; status=8

I (761) CHIP[DL]: BLE host-controller synced
I (1271) chip[DL]: Starting ESP WiFi layer
I (1271) wifi:mode : sta (84:fc:e6:00:f9:44)
I (1271) wifi:enable tsf
I (1271) chip[DL]: Posting ESPSystemEvent: Wifi Event with eventId : 43
W (1271) wifi:Haven't to connect to a suitable AP now!
I (1281) chip[DL]: Done driving station state, nothing else to do...
W (1291) wifi:Haven't to connect to a suitable AP now!
I (1291) chip[DL]: Posting ESPSystemEvent: Wifi Event with eventId : 2
I (1301) chip[DL]: Done driving station state, nothing else to do...
I (1301) chip[SVR]: Initializing subscription resumption storage...
I (1311) chip[SVR]: Server initializing...
I (1321) chip[TS]: Last Known Good Time: 2023-10-14T01:16:48
I (1321) chip[DMG]: AccessControl: initializing
I (1331) chip[DMG]: Examples::AccessControlDelegate::Init
I (1331) chip[DMG]: AccessControl: setting
I (1341) chip[DMG]: DefaultAclStorage: initializing
I (1341) chip[DMG]: DefaultAclStorage: 0 entries loaded
I (1351) chip[SVR]: WARNING: mTestEventTriggerDelegate is null
I (1371) chip[ZCL]: Using ZAP configuration...
I (1371) esp_matter_cluster: Cluster plugin init common callback
I (1371) chip[DMG]: AccessControlCluster: initializing
I (1381) chip[ZCL]: WRITE ERR: ep 0 clus 0x0000_0030 attr 0x0000_0000 not supported
I (1391) chip[ZCL]: Initiating Admin Commissioning cluster.
I (1391) chip[DIS]: Updating services using commissioning mode 0
I (1401) chip[DIS]: CHIP minimal mDNS started advertising.
E (1411) chip[SVR]: ERROR setting up transport: 2f
I (1411) esp_matter_core: Dynamic endpoint 0 added
I (1421) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000003's Attribute 0x00000001 is 2 **********
I (1431) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000004's Attribute 0x00000000 is 128 **********
I (1441) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000004's Attribute 0x0000FFFC is 1 **********
I (1451) esp_matter_core: Dynamic endpoint 1 added
I (1461) chip[DL]: WIFI_EVENT_STA_START
W (1461) wifi:Haven't to connect to a suitable AP now!
I (1471) chip[DL]: Done driving station state, nothing else to do...
I (1471) app_main: Commissioning window closed
I (1531) main_task: Returned from app_main()
```

Commissioning window closed close immediately after boot 1.5s.

Jerry-ESP, 2025-03-06:
Please apply this patch and have a try.

dzungpv, 2025-03-06:
> Please apply this patch and have a try.
> 
> [0001-update-for-dynamic-passcode.zip](https://github.com/user-attachments/files/19103798/0001-update-for-dynamic-passcode.zip)

Yes, it work now. I can pair using the code and chiptool. 
How to generate qr install code the fly? I mean generate QR install code from the generate setup code and show the the log console like esp-rainmaker or to show on a  LCD scsreen

dhrishi, 2025-04-04:
@dzungpv Please close the issue if resolved

closed 2025-04-04 by dzungpv