espressif/esp-matter #1309: Matter connection problems

ravencarcass, 2025-03-04:
Hello makers and engineers,
I am having a problem connecting Matter and phone. I would like to make a project with Matter, in this I would like to build on the light example. The problem here is that my ESP microcontroller does not want to connect to the wifi. I have shown the result of the terminal below.

setup:
- virtual machine (due to problems with WSL2)
- OS: Ubuntu 22.04 server
- ESP IDF: 5.4

The steps I followed:
- build and flashed my code to my ESP32C6
- open the terminal
- open my google home app
- add a Matter device
- scanned the QR code from the Espressif SDK site
- I get an can't find product on my phone and get the next one in the terminal.

What strikes me especially is the error code 0x0500300F, I did find a few who experienced this. Here they just never give a solution with it, can someone help me to solve this. The endpoints are not a problem, I add them for my application. But normally this does not give the problem. This worked once but after that worked it never again.

Terminal:
```
I (1593) chip[DMG]: AccessControlCluster: initializing
I (1593) chip[ZCL]: WRITE ERR: ep 0 clus 0x0000_0030 attr 0x0000_0000 not supported
I (1603) chip[ZCL]: Initiating Admin Commissioning cluster.
I (1603) chip[SVR]: Fabric already commissioned. Disabling BLE advertisement
I (1613) chip[IN]: CASE Server enabling CASE session setups
I (1613) chip[SVR]: Joining Multicast groups
I (1623) chip[SVR]: Server Listening...
I (1623) esp_matter_core: Dynamic endpoint 0 added
I (1633) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000003's Attribute 0x00000001 is 1 **********
I (1643) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000004's Attribute 0x00000000 is 128 **********
I (1653) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000004's Attribute 0x0000FFFC is 1 **********
I (1663) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000006's Attribute 0x0000FFFC is 1 **********
I (1673) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000006's Attribute 0x00004003 is null **********
I (1683) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000006's Attribute 0x00000000 is 0 **********
I (1693) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000006's Attribute 0x00000000 is 0 **********
I (1703) chip[ZCL]: Endpoint 1 On/off already set to new value
I (1713) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000008's Attribute 0x00000002 is 1 **********
I (1723) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000008's Attribute 0x00000003 is 254 **********
I (1733) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000008's Attribute 0x0000FFFC is 3 **********
I (1743) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000008's Attribute 0x00000000 is 64 **********
I (1753) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000008's Attribute 0x00004000 is 64 **********
I (1763) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000008's Attribute 0x00000000 is 64 **********
I (1773) esp_matter_core: Dynamic endpoint 1 added
I (1783) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000003's Attribute 0x00000001 is 1 **********
I (1793) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000004's Attribute 0x00000000 is 128 **********
I (1803) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000004's Attribute 0x0000FFFC is 1 **********
I (1813) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000006's Attribute 0x0000FFFC is 1 **********
I (1823) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000006's Attribute 0x00004003 is 0 **********
I (1833) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000006's Attribute 0x00000000 is 0 **********
I (1843) chip[ZCL]: Boot reason: 1
I (1843) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000006's Attribute 0x00000000 is 0 **********
I (1853) chip[ZCL]: Endpoint 2 On/off already set to new value
I (1863) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000008's Attribute 0x00000002 is 1 **********
I (1873) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000008's Attribute 0x00000003 is 254 **********
I (1883) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000008's Attribute 0x0000FFFC is 3 **********
I (1893) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000008's Attribute 0x00000000 is 64 **********
I (1903) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000008's Attribute 0x00004000 is 64 **********
I (1913) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000008's Attribute 0x00000000 is 64 **********
I (1923) esp_matter_core: Dynamic endpoint 2 added
I (1933) esp_matter_attribute: ********** R : Endpoint 0x0003's Cluster 0x00000003's Attribute 0x00000001 is 1 **********
I (1943) esp_matter_attribute: ********** R : Endpoint 0x0003's Cluster 0x00000004's Attribute 0x00000000 is 128 **********
I (1953) esp_matter_attribute: ********** R : Endpoint 0x0003's Cluster 0x00000004's Attribute 0x0000FFFC is 1 **********
I (1963) esp_matter_attribute: ********** R : Endpoint 0x0003's Cluster 0x00000006's Attribute 0x0000FFFC is 1 **********
I (1973) esp_matter_attribute: ********** R : Endpoint 0x0003's Cluster 0x00000006's Attribute 0x00004003 is 0 **********
I (1983) esp_matter_attribute: ********** R : Endpoint 0x0003's Cluster 0x00000006's Attribute 0x00000000 is 0 **********
I (1993) chip[ZCL]: Boot reason: 1
I (2003) esp_matter_attribute: ********** R : Endpoint 0x0003's Cluster 0x00000006's Attribute 0x00000000 is 0 **********
I (2013) chip[ZCL]: Endpoint 3 On/off already set to new value
I (2013) esp_matter_attribute: ********** R : Endpoint 0x0003's Cluster 0x00000008's Attribute 0x00000002 is 1 **********
I (2023) esp_matter_attribute: ********** R : Endpoint 0x0003's Cluster 0x00000008's Attribute 0x00000003 is 254 **********
[... 166 lines cut ...]
W (3913) wifi:Haven't to connect to a suitable AP now!
I (3913) chip[DL]: WiFi station state change: Connecting_Failed -> NotConnected
W (3923) wifi:Haven't to connect to a suitable AP now!
E (3923) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
I (3933) chip[DL]: Next WiFi station reconnect in 100 ms
I (3943) chip[DL]: Done driving station state, nothing else to do...
W (4043) wifi:Haven't to connect to a suitable AP now!
I (4043) chip[DL]: Attempting to connect WiFi station interface
I (4043) chip[DL]: WiFi station state change: NotConnected -> Connecting
W (4063) wifi:Haven't to connect to a suitable AP now!
E (4063) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
I (4073) chip[DL]: Done driving station state, nothing else to do...
I (5493) chip[DL]: BLE deinit successful and memory reclaimed
I (5493) app_main: BLE deinitialized and memory reclaimed
I (6453) chip[DL]: Posting ESPSystemEvent: Wifi Event with eventId : 5
I (6453) chip[DL]: WIFI_EVENT_STA_DISCONNECTED
I (6453) chip[DL]: WiFi station state change: Connecting -> Connecting_Failed
W (6463) wifi:Haven't to connect to a suitable AP now!
E (6463) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
W (6473) wifi:Haven't to connect to a suitable AP now!
I (6473) chip[DL]: WiFi station state change: Connecting_Failed -> NotConnected
W (6493) wifi:Haven't to connect to a suitable AP now!
E (6493) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
I (6503) chip[DL]: Next WiFi station reconnect in 100 ms
```

dhrishi, 2025-03-04:
@ravencarcass Looks like your device was previously connected to a Wi-Fi network that isn't currently available. Can you try `matter esp factoryreset` on the console and retry commissioning with the ecosystem app.

ravencarcass, 2025-03-04:
thanks, it was indeed that. After the command "matter esp factoryreset" could I connect him.

closed 2025-03-04 by ravencarcass