espressif/esp-matter #1312: Failed to modify the Bluetooth name of matter

MaplestoryAlen, 2025-03-05:
I want to change the Bluetooth name, but I have tried many ways and cannot change the Bluetooth name,
 May I ask how to allow users to customize BLE names? Additionally, I noticed that there is a 16 byte limit. Does this 16 byte limit require subtracting 4 bytes from the description code?

Method 1：
  
  Change and set in menuconfig
   
[image]

Method 2：

void SetBLEDeviceName(void) 
{
    const char *deviceName = "Xz_NC";
    chip::DeviceLayer::ConnectivityMgr().SetBLEDeviceName(deviceName);
}

[image]

Method 3：

#define CHIP_DEVICE_CONFIG_BLE_DEVICE_NAME_PREFIX   “81MtCn01”

Jerry-ESP, 2025-03-05:
Do you want to change the BLE advertisement name, you can add as below in the [file](https://github.com/project-chip/connectedhomeip/blob/5becd8151e01eaf7577c47ebc86cce116680626a/src/platform/ESP32/nimble/BLEManagerImpl.cpp#L1084) :
```
    memset(advData, 0, sizeof(advData));
    advData[index++] = 0x02;                                                                // length
    advData[index++] = CHIP_ADV_DATA_TYPE_FLAGS;                                            // AD type : flags
    advData[index++] = CHIP_ADV_DATA_FLAGS;                                                 // AD value
    advData[index++] = 0x04;                                                                // length
    advData[index++] = 0x9;                                            // AD type : flags
    advData[index++] = 'T';                                                 // AD value
    advData[index++] = 'S';                                                 // AD value
    advData[index++] = 'T';                                                 // AD value
    advData[index++] = kServiceDataTypeSize + sizeof(ESP32ChipServiceData);                 // length
    advData[index++] = CHIP_ADV_DATA_TYPE_SERVICE_DATA;                                     // AD type: (Service Data - 16-bit UUID)
    advData[index++] = static_cast<uint8_t>(ShortUUID_CHIPoBLEService.value & 0xFF);        // AD value
    advData[index++] = static_cast<uint8_t>((ShortUUID_CHIPoBLEService.value >> 8) & 0xFF); // AD value
```

[image]

MaplestoryAlen, 2025-03-13:
Thank you, it has been resolved

closed 2025-03-13 by MaplestoryAlen