espressif/esp-matter #1313: matter value handling (IDFGH-14779)

ravencarcass, 2025-03-05:
Hello creator and engineers,
I want to test some things with Matter. I made a custom program using the example of Matter. But I have problems with the data "esp_matter_attr_val_t *val". I thought that the OnOff value passes a True or a False and I also thought that the LevelControl passes a value between 0 and 100. You can see this in the code as well.

result:
After connecting the app to the microcontroller, I can see that it receives data from my phone, but no output is changed. I think I mishandled the value *fall, but I'm not sure how to fix it.

main code:
```
/*
   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

#include <esp_err.h>
#include <esp_log.h>
#include <nvs_flash.h>

#include <esp_matter.h>
#include <esp_matter_console.h>
#include <esp_matter_ota.h>

#include <common_macros.h>
#include <app_priv.h>
#include <app_reset.h>
#if CHIP_DEVICE_CONFIG_ENABLE_THREAD
#include <platform/ESP32/OpenthreadLauncher.h>
#endif

#include <app/server/CommissioningWindowManager.h>
#include <app/server/Server.h>

#ifdef CONFIG_ENABLE_SET_CERT_DECLARATION_API
#include <esp_matter_providers.h>
#include <lib/support/Span.h>
#ifdef CONFIG_SEC_CERT_DAC_PROVIDER
#include <platform/ESP32/ESP32SecureCertDACProvider.h>
#elif defined(CONFIG_FACTORY_PARTITION_DAC_PROVIDER)
#include <platform/ESP32/ESP32FactoryDataProvider.h>
#endif
using namespace chip::DeviceLayer;
#endif

static const char *TAG = "app_main";
uint16_t light_endpoint_id = 0;
uint16_t light_endpoint_id_1 = 0;
uint16_t light_endpoint_id_2 = 0;
uint16_t light_endpoint_id_3 = 0;
uint16_t light_endpoint_id_4 = 0;
uint16_t light_endpoint_id_5 = 0;
uint16_t light_endpoint_id_6 = 0;
uint16_t light_endpoint_id_7 = 0;
uint16_t light_endpoint_id_8 = 0;
uint16_t light_endpoint_id_9 = 0;
uint16_t light_endpoint_id_10 = 0;
uint16_t light_endpoint_id_11 = 0;

using namespace esp_matter;
using namespace esp_matter::attribute;
using namespace esp_matter::endpoint;
using namespace chip::app::Clusters;

constexpr auto k_timeout_seconds = 300;

#ifdef CONFIG_ENABLE_SET_CERT_DECLARATION_API
extern const uint8_t cd_start[] asm("_binary_certification_declaration_der_start");
extern const uint8_t cd_end[] asm("_binary_certification_declaration_der_end");

[... 797 lines cut ...]
    gpio_set_direction(GPIO_NUM_13, GPIO_MODE_OUTPUT);
    gpio_set_direction(GPIO_NUM_12, GPIO_MODE_OUTPUT);

#if CONFIG_BSP_LEDS_NUM > 0
    /* Initialize led */
    led_indicator_handle_t leds[CONFIG_BSP_LEDS_NUM];
    ESP_ERROR_CHECK(bsp_led_indicator_create(leds, NULL, CONFIG_BSP_LEDS_NUM));
    led_indicator_set_hsv(leds[0], SET_HSV(DEFAULT_HUE, DEFAULT_SATURATION, DEFAULT_BRIGHTNESS));
    
    return (app_driver_handle_t)leds[0];
#else
    return NULL;
#endif
}

app_driver_handle_t app_driver_button_init()
{
    /* Initialize button */
    button_handle_t btns[BSP_BUTTON_NUM];
    ESP_ERROR_CHECK(bsp_iot_button_create(btns, NULL, BSP_BUTTON_NUM));
    ESP_ERROR_CHECK(iot_button_register_cb(btns[0], BUTTON_PRESS_DOWN, app_driver_button_toggle_cb, NULL));
    
    return (app_driver_handle_t)btns[0];
}
```

Jerry-ESP, 2025-03-05:
You can add some print info in the app_attribute_update_cb(), include enpoint id, cluster id and attribute id.

ravencarcass, 2025-03-05:
Big thanks, to tip me the printf. 
I forgot to place the val->val.u8; (facepalmed myself).

closed 2025-03-07 by ravencarcass