espressif/esp-matter #1315: failure while building my custom light code (IDFGH-14761)

ravencarcass, 2025-03-06:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

After the build should he show a table where he dispaly the memory usage.

### What is the actual behavior?

Well if I build my customized matter example. He reports some fault if their are in the code. If all the errors from my code are away, do he give me a failure in the build terminal.

### Steps to reproduce.

1. opened VSCode via the Ubuntu terminal 
2. opened the matter example
3. build this code (no problem)
4. changed a few things in the example for my application
5. build this code and it gave the next failure in the build terminal

### Build or installation Logs.

```plain
*  Executing task: ninja  

[1/13] Performing build step for 'bootloader'
[1/1] cd /home/robbe/esp/light/build/bootloader/esp-idf/esptool_py && /home/robbe/.espressif/python_env/idf5.4_py3.10_env/bin/python /home/robbe/esp/esp-idf/components/partition_table/check_sizes.py --offset 0xc000 bootloader 0x0 /home/robbe/esp/light/build/bootloader/bootloader.bin
Bootloader binary size 0x55e0 bytes. 0x6a20 bytes (55%) free.
[2/11] Performing build step for 'chip_gn'
ninja: no work to do.
[3/9] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/app_main.cpp.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/app_main.cpp.obj 
ccache /home/robbe/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/riscv32-esp-elf-g++ -DESP_PLATFORM -DIDF_VER=\"v5.4\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DMD5_ENABLED=1 -DSERIAL_FLASHER_BOOT_HOLD_TIME_MS=50 -DSERIAL_FLASHER_RESET_HOLD_TIME_MS=100 -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -DUNITY_INCLUDE_CONFIG_H -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/home/robbe/esp/light/build/config -I/home/robbe/esp/light/main -I/home/robbe/esp/esp-matter/examples/common/utils -I/home/robbe/esp/esp-idf/components/newlib/platform_include -I/home/robbe/esp/esp-idf/components/freertos/config/include -I/home/robbe/esp/esp-idf/components/freertos/config/include/freertos -I/home/robbe/esp/esp-idf/components/freertos/config/riscv/include -I/home/robbe/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/robbe/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include -I/home/robbe/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include/freertos -I/home/robbe/esp/esp-idf/components/freertos/esp_additions/include -I/home/robbe/esp/esp-idf/components/esp_hw_support/include -I/home/robbe/esp/esp-idf/components/esp_hw_support/include/soc -I/home/robbe/esp/esp-idf/components/esp_hw_support/include/soc/esp32c6 -I/home/robbe/esp/esp-idf/components/esp_hw_support/dma/include -I/home/robbe/esp/esp-idf/components/esp_hw_support/ldo/include -I/home/robbe/esp/esp-idf/components/esp_hw_support/debug_probe/include -I/home/robbe/esp/esp-idf/components/esp_hw_support/port/esp32c6/. -I/home/robbe/esp/esp-idf/components/esp_hw_support/port/esp32c6/include -I/home/robbe/esp/esp-idf/components/esp_hw_support/port/esp32c6/private_include -I/home/robbe/esp/esp-idf/components/heap/include -I/home/robbe/esp/esp-idf/components/heap/tlsf -I/home/robbe/esp/esp-idf/components/log/include -I/home/robbe/esp/esp-idf/components/soc/include -I/home/robbe/esp/esp-idf/components/soc/esp32c6 -I/home/robbe/esp/esp-idf/components/soc/esp32c6/include -I/home/robbe/esp/esp-idf/components/soc/esp32c6/register -I/home/robbe/esp/esp-idf/components/hal/platform_port/include -I/home/robbe/esp/esp-idf/components/hal/esp32c6/include -I/home/robbe/esp/esp-idf/components/hal/include -I/home/robbe/esp/esp-idf/components/esp_rom/include -I/home/robbe/esp/esp-idf/components/esp_rom/esp32c6/include -I/home/robbe/esp/esp-idf/components/esp_rom/esp32c6/include/esp32c6 -I/home/robbe/esp/esp-idf/components/esp_rom/esp32c6 -I/home/robbe/esp/esp-idf/components/esp_common/include -I/home/robbe/esp/esp-idf/components/esp_system/include -I/home/robbe/esp/esp-idf/components/esp_system/port/soc -I/home/robbe/esp/esp-idf/components/esp_system/port/include/riscv -I/home/robbe/esp/esp-idf/components/esp_system/port/include/private -I/home/robbe/esp/esp-idf/components/riscv/include -I/home/robbe/esp/esp-idf/components/lwip/include -I/home/robbe/esp/esp-idf/components/lwip/include/apps -I/home/robbe/esp/esp-idf/components/lwip/include/apps/sntp -I/home/robbe/esp/esp-idf/components/lwip/lwip/src/include -I/home/robbe/esp/esp-idf/components/lwip/port/include -I/home/robbe/esp/esp-idf/components/lwip/port/freertos/include -I/home/robbe/esp/esp-idf/components/lwip/port/esp32xx/include -I/home/robbe/esp/esp-idf/components/lwip/port/esp32xx/include/arch -I/home/robbe/esp/esp-idf/components/lwip/port/esp32xx/include/sys -I/home/robbe/esp/light/managed_components/espressif__esp_bsp_devkit/include -I/home/robbe/esp/esp-idf/components/driver/deprecated -I/home/robbe/esp/esp-idf/components/driver/i2c/include -I/home/robbe/esp/esp-idf/components/driver/touch_sensor/include -I/home/robbe/esp/esp-idf/components/driver/twai/include -I/home/robbe/esp/esp-idf/components/esp_pm/include -I/home/robbe/esp/esp-idf/components/esp_ringbuf/include -I/home/robbe/esp/esp-idf/components/esp_driver_gpio/include -I/home/robbe/esp/esp-idf/components/esp_driver_pcnt/include -I/home/robbe/esp/esp-idf/components/esp_driver_gptimer/include -I/home/robbe/esp/esp-idf/components/esp_driver_spi/include -I/home/robbe/esp/esp-idf/components/esp_driver_mcpwm/include -I/home/robbe/esp/esp-idf/components/esp_driver_ana_cmpr/include -I/home/robbe/esp/esp-idf/components/esp_driver_i2s/include -I/home/robbe/esp/esp-idf/components/esp_driver_sdmmc/include -I/home/robbe/esp/esp-idf/components/sdmmc/include -I/home/robbe/esp/esp-idf/components/esp_driver_sdspi/include -I/home/robbe/esp/esp-idf/components/esp_driver_sdio/include -I/home/robbe/esp/esp-idf/components/esp_driver_dac/include -I/home/robbe/esp/esp-idf/components/esp_driver_rmt/include -I/home/robbe/esp/esp-idf/components/esp_driver_tsens/include -I/home/robbe/esp/esp-idf/components/esp_driver_sdm/include -I/home/robbe/esp/esp-idf/components/esp_driver_i2c/include -I/home/robbe/esp/esp-idf/components/esp_driver_uart/include -I/home/robbe/esp/esp-idf/components/vfs/include -I/home/robbe/esp/esp-idf/components/esp_driver_ledc/include -I/home/robbe/esp/esp-idf/components/esp_driver_parlio/include -I/home/robbe/esp/esp-idf/components/esp_driver_usb_serial_jtag/include -I/home/robbe/esp/esp-idf/components/spiffs/include -I/home/robbe/esp/esp-idf/components/esp_partition/include -I/home/robbe/esp/esp-idf/components/fatfs/diskio -I/home/robbe/esp/esp-idf/components/fatfs/src -I/home/robbe/esp/esp-idf/components/fatfs/vfs -I/home/robbe/esp/esp-idf/components/wear_levelling/include -I/home/robbe/esp/light/managed_components/espressif__button/include -I/home/robbe/esp/esp-idf/components/esp_adc/include -I/home/robbe/esp/esp-idf/components/esp_adc/interface -I/home/robbe/esp/esp-idf/components/esp_adc/esp32c6/include -I/home/robbe/esp/esp-idf/components/esp_adc/deprecated/include -I/home/robbe/esp/light/managed_components/espressif__led_indicator/include -I/home/robbe/esp/light/managed_components/espressif__led_strip/include -I/home/robbe/esp/light/managed_components/espressif__led_strip/interface -I/home/robbe/esp/esp-idf/components/mbedtls/port/include -I/home/robbe/esp/esp-idf/components/mbedtls/mbedtls/include -I/home/robbe/esp/esp-idf/components/mbedtls/mbedtls/library -I/home/robbe/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -I/home/robbe/esp/esp-idf/components/mbedtls/mbedtls/3rdparty/everest/include -I/home/robbe/esp/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m -I/home/robbe/esp/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m/p256-m -I/home/robbe/esp/esp-idf/components/esp_app_format/include -I/home/robbe/esp/esp-idf/components/esp_bootloader_format/include -I/home/robbe/esp/esp-idf/components/app_update/include -I/home/r [... cut ...]
    err = esp_matter_ota_requestor_encrypted_init(s_decryption_key, s_decryption_key_len);
    ABORT_APP_ON_FAILURE(err == ESP_OK, ESP_LOGE(TAG, "Failed to initialized the encrypted OTA, err: %d", err));
#endif // CONFIG_ENABLE_ENCRYPTED_OTA

#if CONFIG_ENABLE_CHIP_SHELL
    esp_matter::console::diagnostics_register_commands();
    esp_matter::console::wifi_register_commands();
    esp_matter::console::factoryreset_register_commands();
#if CONFIG_OPENTHREAD_CLI
    esp_matter::console::otcli_register_commands();
#endif
    esp_matter::console::init();
#endif
}
```

shubhamdp, 2025-03-06:
@ravencarcass if you check, the error it is failing to compile app_main.cpp file, as you have said that you modified the file, they may be wrong. So I'd suggest you to share the diff of your changes on top of original code.

Also, what could help is the actual compilation errors.

ravencarcass, 2025-03-07:
solved it

closed 2025-03-07 by ravencarcass