espressif/esp-matter #1317: out-of-order initializers are nonstandard in C++

ravencarcass, 2025-03-07:
**Describe the bug**
I have a problem though I am going to build my program. I started with the matter example light. Then I added a few things. This all worked up to a certain point. I added I2C. This all worked very well. Once I wanted to add a ledc VSCode gives me a strange error.

“out-of-order initializers are nonstandard in C++”.

**Environment**
- ESP-IDF Commit Id: V5.4
- SoC ESP32C6:

- Host Machine OS: linux Ubuntu 22.04 
- Host Machine Python version: 3.10.12

**Any additional details**
I first did some with the I2C protocol in the matter enviroment and it gave me no error. After these test would I like to add a ledc and then began the problems. This is the code where he runs in trouble:
```
void periferal_init(void){
    i2c_config_t confi2c = {
        .mode = I2C_MODE_MASTER, 
        .sda_io_num = I2C_MASTER_SDA_IO,
        .scl_io_num = I2C_MASTER_SCL_IO,
        .sda_pullup_en = GPIO_PULLUP_ENABLE,
        .scl_pullup_en = GPIO_PULLUP_ENABLE,
        .master = { .clk_speed = I2C_MASTER_FREQ_HZ } // Correcte initialisatie
    };

    i2c_param_config(I2C_NUM_0, &confi2c);
    i2c_driver_install(I2C_NUM_0,confi2c.mode,0,0,0);

    ledc_timer_config_t timer = {
        .speed_mode         = LEDC_LOW_SPEED_MODE,
        .timer_num          = LEDC_TIMER_1,
        .duty_resolution    = LEDC_TIMER_13_BIT,
        .freq_hz            = 500,
        .clk_cfg            = LEDC_AUTO_CLK
    };
    ledc_timer_config(&timer);

    ledc_channel_config_t channel = {
        .gpio_num       = 10,
        .speed_mode     = LEDC_LOW_SPEED_MODE,
        .channel        = LEDC_CHANNEL_1,
        .intr_type      = LEDC_INTR_DISABLE,
        .timer_sel      = LEDC_TIMER_1,
        .duty           = DUTY,
        .hpoint         = 0
    };
    ledc_channel_config(&channel);
}
```

shubhamdp, 2025-03-07:
Can you point me to the place from where you cut-pasted the code snippet?

in C++, designated initializers must follow the same order as the member variable declarations in the struct.

You can fix the order of initialization above by moving around the members as per the declaration of those structs, or simply change them to something like
```
i2c_config_t confi2c;
confi2c.mode = I2C_MODE_MASTER;
.... and remaning ones...
```

ravencarcass, 2025-03-07:
well, I've first made this apart an I copied that code so I could see that it works. You can see that code below here.
This code works seamless for the DRV8825.
```

#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"
#include "esp_log.h"
#include "driver/ledc.h"

 #define STEP 5
 #define DIR 6
 #define RESET 7
 #define SLEEP 0

#define DUTY 4096 

 static const char *TAG = "stepper";

/* Use project configuration menu (idf.py menuconfig) to choose the GPIO to blink,
   or you can edit the following line and set a number here.
*/
void app_main(void)
{
    ledc_timer_config_t timer = {
        .speed_mode     = LEDC_LOW_SPEED_MODE,
        .timer_num      = LEDC_TIMER_0,
        .freq_hz        = 500,
        .duty_resolution = LEDC_TIMER_13_BIT,
        .clk_cfg        = LEDC_AUTO_CLK
    };
    ledc_timer_config(&timer);

    ledc_channel_config_t channel = {

        .speed_mode     = LEDC_LOW_SPEED_MODE,
        .channel        = LEDC_CHANNEL_1,
        .timer_sel      = LEDC_TIMER_0,
        .intr_type      = LEDC_INTR_DISABLE,
        .gpio_num       = 10,
        .duty           = DUTY,
        .hpoint         = 0
    };
    ledc_channel_config(&channel);

    ESP_LOGI(TAG, "voorbeeld stepper");
    gpio_reset_pin(STEP);
    gpio_set_direction(STEP, GPIO_MODE_OUTPUT);
    gpio_reset_pin(DIR);
    gpio_set_direction(DIR, GPIO_MODE_OUTPUT);
    gpio_reset_pin(RESET);
    gpio_set_direction(RESET, GPIO_MODE_OUTPUT);
    gpio_reset_pin(SLEEP);
    gpio_set_direction(SLEEP, GPIO_MODE_OUTPUT);

    gpio_set_level(RESET, 1);
    gpio_set_level(SLEEP, 1);
    gpio_set_level(DIR, 0);

    ledc_set_duty(LEDC_LOW_SPEED_MODE, LEDC_CHANNEL_1, DUTY);
    ledc_update_duty(LEDC_LOW_SPEED_MODE, LEDC_CHANNEL_1);
    vTaskDelay(200/portTICK_PERIOD_MS);

    ledc_set_duty(LEDC_LOW_SPEED_MODE, LEDC_CHANNEL_1, 0);
    ledc_update_duty(LEDC_LOW_SPEED_MODE, LEDC_CHANNEL_1);      

}
```

shubhamdp, 2025-03-07:
how is this code saved as? Is it a .cpp or .c file? If it's .cpp the error you are seeing is expected.

PS: Most of compilation errors can be fixed by a Google search which would point to the stackoverflow link.

shubhamdp, 2025-03-08:
@ravencarcass thanks for confirmation, can you please close if it working.

ravencarcass, 2025-03-08:
@shubhamdp , I've only a problem because I get no PWM out of my pins so their could be somethings wrong. 
I configured my I2C and my LEDC in the same function in the main code. WhereI had no problems with I2C and with the LEDC do he report the next: 

```
open driver -> endpoint: 12 -> cluster: 6 -> attribute: 0 -> 1 ->motor turning
E (29265) ledc: ledc_set_duty(1057): LEDC is not initialized
E (29265) ledc: ledc_update_duty(983): LEDC is not initialized
```

The reported code lines where from the driver file: 
```
esp_err_t app_driver_attribute_update(app_driver_handle_t driver_handle, uint16_t endpoint_id, uint32_t cluster_id,
    uint32_t attribute_id, esp_matter_attr_val_t *val)
{   
    printf("open driver -> ");

    esp_err_t err = ESP_OK;
    uint8_t tempVal = val->val.u8;

    if (endpoint_id == light_endpoint_id) 
    {
        printf("endpoint: %d -> ", endpoint_id);
        if(cluster_id == LevelControl::Id)
        {
            printf("cluster: %ld -> ", cluster_id);
            if(attribute_id == LevelControl::Attributes::CurrentLevel::Id)
            {
                printf("movement to %d\n", tempVal);
            }
        }
        if(cluster_id == OnOff::Id)
        {
            printf("cluster: %ld -> ", cluster_id);
            if(attribute_id == OnOff::Attributes::OnOff::Id)
            {
                printf("attribute: %ld -> %ld ->", attribute_id, (uint32_t)tempVal);

                if(tempVal >= 1)     //0 of 0<
                {
                    printf("motor turning\n");
                    ledc_set_duty(LEDC_LOW_SPEED_MODE, LEDC_CHANNEL_1, 4096);
                    ledc_update_duty(LEDC_LOW_SPEED_MODE, LEDC_CHANNEL_1);
                } 
                else 
                {
                    printf("motor stoped turning\n");
                    ledc_set_duty(LEDC_LOW_SPEED_MODE, LEDC_CHANNEL_1, 0);
                    ledc_update_duty(LEDC_LOW_SPEED_MODE, LEDC_CHANNEL_1);
                }
            }

        } 
    }
}
```

shubhamdp, 2025-03-09:
@ravencarcass Is your original issue "out-of-order initializers are nonstandard in C++" fixed? If yes, please close this, for other problems please open a new one. I don't want users to get confused where we are solving different problem under different heading. 

Anyways, [ledc_channel_config()](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/ledc.html#_CPPv419ledc_channel_configPK21ledc_channel_config_t) returns an error of type `esp_err_t`, please check the return value and do not ignore it to get more idea, it could happen that the config is wrong, or any other thing. Here's the doc to led control https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/ledc.html

closed 2025-03-09 by ravencarcass