espressif/esp-matter #1318: Update esp_encrypted_img Component Version

leandropg, 2025-03-07:
Currently, I am developing a product using the **ESP Matter SDK**. For the characteristic of the project, I am using **HTTPS OTA** and specifically the [OTA Upgrades with Pre-Encrypted Firmware](https://docs.espressif.com/projects/esp-idf/en/v5.2.5/esp32c6/api-reference/system/esp_https_ota.html#ota-upgrades-with-pre-encrypted-firmware).

In the example  [pre_encrypted_ota](https://github.com/espressif/esp-idf/tree/v5.2.5/examples/system/ota/pre_encrypted_ota) it uses the the external component [espressif/esp_encrypted_img](https://components.espressif.com/components/espressif/esp_encrypted_img/) in the latest version 2.3.0.

But this configuration in the **ESP Matter SDK** restrict to use in the project only the version **2.1.0**:

https://github.com/espressif/esp-matter/blob/51e5f957809dc0311a6bb7f58994460974842821/idf_component.yml#L78-L82

Hence when I configured in my project to use the latest version 2.3.0.

```
espressif/esp_encrypted_img: ^2.3.0
```
I obtained this error in compilation time:

```
CMake Error at /home/user/esp-idf/tools/cmake/build.cmake:552 (message):
  ERROR: Because project depends on both espressif/esp_encrypted_img (^2.3.0)
  and espressif/esp_encrypted_img (2.1.0), version solving failed.
```

Could you update this configuration to set the latest version available 2.3.0?

Thanks in advance.

shubhamdp, 2025-03-08:
> Could you update this configuration to set the latest version available 2.3.0?

I do not see any feature addition in the [changelog](https://github.com/espressif/idf-extra-components/blob/master/esp_encrypted_img/CHANGELOG.md) of `esp_encrypted_img`, apart from a minor cmake fix + a get header size API.

Do you have any specific need for using `v2.3.0`?

Purpose of pinning the exact version was to make sure that connectedhomeip do not break when `esp_encrypted_img` keeps moving ahead.

For a quick fix, please change the version in this file and give it a go: esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/idf_component.yml

Please let me know it this works for you, so that I can put up the changes in connectedhomeip and then roll the submodule in esp-matter

leandropg, 2025-03-10:
Hi @shubhamdp The reason of the requirement is that the example use this definition in the configuration and it uses the function **esp_encrypted_img_get_header_size()** only available from the version **2.2.0**

https://github.com/espressif/esp-idf/blob/9b74d1aca5f2154c4ea609d6e36d513dcf91148b/examples/system/ota/pre_encrypted_ota/main/pre_encrypted_ota.c#L147

I will make the change and I will confirm you if it worked.

shubhamdp, 2025-03-10:
@leandropg appreciate the efforts for pointing out the breakage. Please let us know if it works, and I'll roll up the version, do some tests, and update the submodule here.

Also, which esp-matter branch/commit are you using? It would help decide releases for back porting this change.

leandropg, 2025-03-10:
Hi @shubhamdp Changing the version there to `2.3.0`, now the system can download the right component in the version `2.3.0` and now I can compile the project. The system recognize now the function `esp_encrypted_img_get_header_size()`.

Currently my Matter environment is pointing to `main` branch and specifically the **commit 05831a380c536ab12767b0872fdf81faee31195e** (Tue Jan 7 14:45:36 2025).

shubhamdp, 2025-03-10:
I have raised the PR here: https://github.com/project-chip/connectedhomeip/pull/37950, PTAL!

dhrishi, 2025-05-14:
@leandropg The fix is now a part of esp-matter main branch. Can you please confirm and close this issue if it works as expected

dhrishi, 2025-06-06:
@leandropg This is fixed. Please check. Feel free to re-open if you still see the issue

closed 2025-06-06 by dhrishi