espressif/esp-matter #1323: RCP not working over USB

MiguelCan13, 2025-03-12:
IDF version.
v5.4

Operating System used.
Windows

How did you build your project?
ESP-IDF CLI

If you are using Windows, please specify command line type.
Ubuntu

What is the expected behavior?
I have a pi 3 acting as a thread border router with a nordic RCP functioning successfully. However, when I use the RCP example provided by esp-idf, it is unable to function with the rpi3. On the other hand, the RPI has issues writing to the RCP as the USB JTAG is preventing us from writing to the esp from the host PC.

Are there any available configurations to enable spinel over the USB lines? or even the USB-to-UART? I am having issues with that as well. 

When I try to enable USB-CDC it does not let me, I presume this is because "minimal" build is on. However, the project does not successfully build when minimal build is off.

I have tried various other configurations in menu config to try and enable this however it is still to no avail. Any solutions would be greatly appreciated!

gytxxsy, 2025-03-12:
Currently, ESP32-H2 and ESP32-C6 support USB communication at the hardware level. However, our software has not yet implemented USB communication for ESP OpenThread RCP. For now, you can use UART as an alternative.

nichonaugle, 2025-03-12:
@gytxxsy Are there any available configurations or documentation you can offer that shows how to use the built in usb-to-uart convertor present on the esp32c6 dev kit? I have tried setting tx and rx pins on the devkit in menu config to the same ones used by the convertor, however that did not work. Thanks in advance

chshu, 2025-03-13:
@MiguelCan13 @nichonaugle Any particular reason you have to use USB instead of UART?

The ESP [ot_rcp](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_rcp) could work with the OTBR as is. Pleaase refer to this doc: [using-ot-rcp-with-otbr](https://docs.espressif.com/projects/esp-thread-br/en/latest/qa.html#using-ot-rcp-with-otbr-ot-br-posix).

nichonaugle, 2025-03-13:
@chshu Thank you for the documentation, it is very helpful. I am looking to integrate the esp32h2 into a consumer smart product that is currently under development. If we were able to use USB instead of UART it would be easier to use existing esptool utilities to reflash the chip from a host processor. If we used uart, we would have to setup additional pins to handling setting the esp32h2 into boot mode in order to reflash (and the software associated with that). Assuming the RCP will need updates in the future, using USB lines would only take up two pins from our host processor and make updates easier for our team.

MiguelCan13, 2025-03-19:
@chshu Thank you for your response. I followed the doc and it almost worked, however I am getting a failed to process request, no such process error on otbr-startup. Though this is just a warning, and I'm not sure if its going to affect me in later development stages.

When I access the status of the otbr-agent it returns active (running) with relatively normal readings. However, this is the log:
[I] P-Netif----------: Deleting route for NAT64
[I] P-McastRtMgr--: Disable: OK
[INFO]-BA----------: Publish meshcop service openthread boarderrouter #D927._mechcop._udp.local.
[I] RouterTable-----: Route table
[I] P-Netif----------: Host netif Is down 
[INFO]-MDNS------: Successfully registered service openthread boarderrouter #D927._mechcop._udp
[INFO]-BA----------: Result of publish meshcop service OpenThread BoarderRouter #D927._mechcop._udp.local: OK
[INFO]-BA----------: Result of publish meshcop service OpenThread BoarderRouter #D927._mechcop._udp.local: OK
[I] P-Netif----------: Secceeded to process request#1
[I] P-netif----------: Failed to process request#2: No such process
[I] InfraIF-----------: state changes: RUNNING -> NOT RUNNING
[I] InfraIF-----------: state changes: NOT RUNNING -> RUNNING

additionally, when I look at my network interfaces, wpan0 is present however it doesn't seem to have an address. Any advice would be helpful!

nichonaugle, 2025-04-21:
I managed to get it working using the built in usb-uart convertor found on the c6 devkit. There is warning spinel message however the ESP RCP still properly sets up and commissions devices onto the network

bhaskarkishore, 2025-05-08:
Hi @nichonaugle, that's awesome. Please if you have the time can you summarize the steps you took to get it working?

dhrishi, 2025-05-14:
@MiguelCan13 Please see [this](https://github.com/espressif/esp-matter/issues/1323#issuecomment-2814334244) and close the issue if it works.