espressif/esp-matter #1324: User defined mDNS broadcast cannot be turned off

MaplestoryAlen, 2025-03-14:
The SDK I am currently using is matte V1.4, which requires user-defined mDNS broadcasting. After setting “CONFIG_USE_MINIMAL_MDNS”, it can also be scanned after enabling broadcasting. However, I am currently unable to turn off the mDNS broadcasting that I have enabled myself

code:

void mDNS_broadcast_task(void *pvParameters)
{   
#define  AdvTxTLen     5
    char host_name[32] = "";
	char vid_str[8];  
    sprintf(vid_str, "%u", GetIotModuleInfoStruct()->iot_vendor_id);
	char pid_str[8];
    sprintf(pid_str, "%u", GetIotModuleInfoStruct()->iot_product_id);
	char discriminator_str[8]; 
    sprintf(discriminator_str, "%u", GetIotModuleInfoStruct()->iot_discriminator);	
	mdns_txt_item_t serviceTxtData[AdvTxTLen] = {
                                           {"did",GetIotModuleInfoStruct()->iot_did},
                                           {"vid",vid_str},
                                           {"pid",pid_str},
                                           {"discriminator",discriminator_str},
                                           {"port","81"}
                                        };		

    USER_LOGI(TAG, "User mDNS broadcast start ....");
    snprintf(host_name, sizeof(host_name), "xz-%s", GetIotModuleInfoStruct()->iot_sn);
    esp_netif_t *intf = esp_netif_get_default_netif();
    mdns_ip_addr_t addr4, addr6;
    addr4.addr.type = ESP_IPADDR_TYPE_V4;
    esp_netif_ip_info_t info;
    esp_netif_get_ip_info(intf, &info);
    addr4.addr.u_addr.ip4 = info.ip;
    addr6.addr.type = ESP_IPADDR_TYPE_V6;
    esp_netif_get_ip6_linklocal(intf, &addr6.addr.u_addr.ip6);
    addr4.next = &addr6;
    addr6.next = NULL;
	  
      if(mdns_service_exists("_sock", "_tcp", NULL))
      {
          USER_LOGI(TAG, "Service exists, proceeding to remove.");
         esp_err_t err = mdns_service_remove("_sock", "_tcp");
         if (err != ESP_OK) 
         {
            USER_LOGE(TAG, "Failed to remove mDNS service: %s", esp_err_to_name(err));
         }
        USER_LOGI(TAG, "mDNS service removed successfully.");
    }
    // create MDNS record for the web server at 'lowpan-xxx.local'
    ESP_ERROR_CHECK(mdns_delegate_hostname_add(host_name, &addr4));
    ESP_ERROR_CHECK(mdns_service_add_for_host(InstanceName, ServiceName, Protocol, host_name, mDnsPort, serviceTxtData, AdvTxTLen));	 

    while (1) 
    {

	 vTaskDelay(pdMS_TO_TICKS(2000));
        if(mdns_service_exists("_sock", "_tcp", NULL))
	{
             USER_LOGI(TAG, "Service exists, proceeding to remove.");
	     esp_err_t err = mdns_service_remove("_sock", "_tcp");
             if (err == ESP_OK) 
	     {      
			USER_LOGI(TAG, "mDNS service removed successfully.");
             }
	    else
	    {
		    USER_LOGE(TAG, "Failed to remove mDNS service: %s", esp_err_to_name(err));
	    }        
        }
    }
   vTaskDelete(NULL);

}

log:

[image]

[image]

wqx6, 2025-03-14:
`mdns_service_remove()` will only remove the local services, If you want to remove services on delegate host, please use `mdns_service_remove_for_host()`.

MaplestoryAlen, 2025-03-17:
> `mdns_service_remove()` will only remove the local services, If you want to remove services on delegate host, please use `mdns_service_remove_for_host()`.

The problem has been resolved, thank you very much

closed 2025-03-17 by MaplestoryAlen