espressif/esp-matter #1325: Matter over Thread, commissioning fail in chip tool

FMistrorigo, 2025-03-14:
**Description**
I have a Thread Network that seems to work perfectly. The Border Router (BR) is an ESP32S3 that uses an RCP made using an ESP32H2. I have a Minimal Thread Device (MTD) that uses another ESP32H2 with the Matter Light example (https://github.com/espressif/esp-matter/tree/release/v1.2/examples/light). 
The BR creates a new dataset with the name "GATEWAY_000C", creating randomly the other parameters in case of a factoryreset. I'm able to add other devices in the Thread Network and using the CLI command "neighbor table" i can see that they are connected. I'm also able to ping IPv4 servers from the child (MTD), so I think everything about Openthread configuration is right.

```
| Role | RLOC16 | Age | Avg RSSI | Last RSSI |R|D|N|      Extended MAC       | Vers |
+--------+------------------+-------------+-----------+---------+----------+-------+
|    C | 0x5801 |   7 |      -31 |       -29 |1|0|1| 4a:f9:13:ca:38:06:8e:e7 |    4 |
```

The Thread Border Router is not a Matter device, but the Child should be able to be commissioned. I noticed that the BR, once I connect the Child, creates a _matterc._udp service that I suppose it's used to comunicate with the Child from outside the IPv6 network.

```
I (12286135) OPENTHREAD: Host delete
I (12286136) OPENTHREAD: delete host 4AF913CA38068EE7
I (12290120) OPENTHREAD: Host update
I (12290121) OPENTHREAD: update host 4AF913CA38068EE7
I (12290122) OPENTHREAD: Add a new host, addr FDEC:6E9D:C61F:1:C340:F9EF:64B9:43E0
I (12290123) OPENTHREAD: Service FCEB75E1736CBA51._matterc._udp
I (12290124) OPENTHREAD: Add new service
I (12290126) OPENTHREAD: Service subtype _V65521
I (12290126) OPENTHREAD: Service subtype _S15
I (12290127) OPENTHREAD: Service subtype _L3840
I (12290128) OPENTHREAD: Service subtype _CM
```

I tried to ping FCEB75E1736CBA51._matterc._udp from my computer and it works, making me think that the NAT64 in the BR is being used correctly, but pinging the IPv6 address FDEC:6E9D:C61F:1:C340:F9EF:64B9:43E0 is not possible.

At this point I thougt I would be able to add the Child device in my Google Home App (version 3.30.1.5), so I added a Google Home Mini (that supports Matter but not Thread) in the same Wi-Fi network as the BR, checked again that the Child was in the Thread Network of my BR and started the commissioning step in the Google Home App scanning the QR Code. The procedure was going fine, but after the "Generating Credentials" step the App told me it was not able to connet to the Thread Network "myHomexxxxxxx" that's the Thread Network of my other device, a Google Nest that's offline and not in my account at this moment.
It looks like the Thread Network of my Google Nest was saved in the Google Home App or somewere else as the default one.

I decided to use the chip-tool to have more logs and understand if the problem was in the Google Home App or in my devices (BR or Child).
I used the command matter device factoryreset on che Child to reset everything about matetr and also the Openthread dataset.

Here's the LOGS when I try to use the chip-tool:

[LOG THREAD.txt](https://github.com/user-attachments/files/19250032/LOG.THREAD.txt)

I think the error is here:

_Received NetworkConfig response, networkingStatus=1
CHIP:CTL: Error on commissioning step 'ThreadNetworkSetup': 'src/controller/CHIPDeviceController.cpp:2285: CHIP Error 0x000000AC: Internal error'_

After this result I added some dbg messages in the code to see the active dataset, "matter device factoryreset" removed the dataset as expected, after using  the chip-tool nothing changes and it's still empty, and with Google Home is setting the wrong one.

Can you help me understand where the problem is?
I think that Google Home app saved the wrong Thread network as default and I'm not able to change it, this might explain the problems with the wrong Dataset, but the fact that it doesn't work with the chip-tool make me think that the problem might be in the DNS resolver of the BR or maybe in some wrong configuration of the BR.

**Environment**
- ESP-Matter Commit Id: 066791d61fe1802be7f9b6fd925efdcf4cdd516a (HEAD -> release/v1.2)
- ESP-IDF Commit Id: c9763f62dd00c887a1a8fafe388db868a7e44069 (HEAD, tag: v5.2.3)
- BR SoC: ESP32S3 & ESP32H2 (as RCP)
- CHILD SoC: ESP32H2 (ESP32-h2-mini-1 D2N4)
- Chip-tool used in a Raspberrypi with Ubuntu 22.04.5

Jerry-ESP, 2025-03-17:
Please don't let your matter over thread device connect to the thread network before matter pairing, it will connect to the specified thread network with dataset include in your pairing command. After factory reset the device(matter light), you can use the chip-tool to pairing the device directly, no need to connect it to thread BR before.  
Please have a try.

FMistrorigo, 2025-03-17:
Hi Jerry,
As you asked me to do i did a "matter device factoryreset" in the Child device and the dataset was removed.
I proceeded to repeat the tests with the same configuration:
- Google Home Mini in my Wi-Fi network;
- Border Router in the same Wi-Fi network with this dataset: `0e080000000000010000000300001835060004001fffe002081c71f1a38952a1eb0708fd88fa82f97347a80510e280655b55cc33bf5a44064e00e36e3c030c43414d455f47575f303030430102195c0410bbe6e5f547463872a499cc5135b81e7a0c0402a0f7f8`;
- Child without dataset ready to be commissioned with matter.

These are the logs of the child and the commissioner with the chip-tool using the command
`./chip-tool pairing ble-thread 1 0e080000000000010000000300001835060004001fffe002081c71f1a38952a1eb0708fd88fa82f97347a80510e280655b55cc33bf5a44064e00e36e3c030c43414d455f47575f303030430102195c0410bbe6e5f547463872a499cc5135b81e7a0c0402a0f7f8 20202021 3840 --paa-trust-store-path /var/paa-root-certs/`
[LOG child not in OT Network.txt](https://github.com/user-attachments/files/19281871/LOG.child.not.in.OT.Network.txt)

The problem is this:
```
Received NetworkConfig response, networkingStatus=1
[1742199233.477640][2383:2385] CHIP:CTL: Error on commissioning step 'ThreadNetworkSetup': 'src/controller/CHIPDeviceController.cpp:2285: CHIP Error 0x000000AC: Internal error'
```
----------------------------------------

> Please don't let your matter over thread device connect to the thread network before matter pairing

I was wandering why i need to remove the device from the Thread Network, I use my app to add the device in the Thread Network and later on I want the user to be able to add the device with Matter.
I tried to add the Child device in my Thred Network and then use the "onnetwork" command from the chip tool but this is the result:

`./chip-tool pairing onnetwork 1 20202021 --paa-trust-store-path /var/paa-root-certs/`
[LOGS with onnetwork command.txt](https://github.com/user-attachments/files/19282122/LOGS.with.onnetwork.command.txt)

Jerry-ESP, 2025-03-18:
Hi @FMistrorigo ,

What your understand is correct, the matter over thread device can pairing through ble-thread and 
onnetwork ways. It's supposed to be normal in this way. Can you share your esp-matter commit id and the sdkconfig, I will try on my side.

FMistrorigo, 2025-03-18:
Hi @Jerry-ESP,

I'm using this esp-matter commit: 066791d61fe1802be7f9b6fd925efdcf4cdd516a (HEAD -> release/v1.2)
and this esp-idf commit: c9763f62dd00c887a1a8fafe388db868a7e44069 (HEAD, tag: v5.2.3)

This is the sdkconfig of the child device that is running the Matter light example:
[sdkconfig_light.txt](https://github.com/user-attachments/files/19310051/sdkconfig.txt)

And this is the sdkconfig that I'm using in the BR:
[sdkconfig_BR.txt](https://github.com/user-attachments/files/19310068/sdkconfig.txt)

To test the BR example you can use the basic_threa_border_router example in the esp-thread-br directory
https://github.com/espressif/esp-thread-br/tree/main/examples/basic_thread_border_router

Can you tell me more about this log in the chip-tool?
`Received NetworkConfig response, networkingStatus=1`

I'm surprised that with the commissioning using Google Home the dataset is being set (wrong one btw), but with the chip-tool I get the "Error on commissioning step 'ThreadNetworkSetup'" before the dataset is being added to the device, it doesn't make any sense.

Jerry-ESP, 2025-03-18:
It seems the incorrect pairing command cause this issue, in the pairing command, you lose the hex: before the dataset, the correct one should be: 
```
./chip-tool pairing ble-thread 1 hex:0e080000000000010000000300001835060004001fffe002081c71f1a38952a1eb0708fd88fa82f97347a80510e280655b55cc33bf5a44064e00e36e3c030c43414d455f47575f303030430102195c0410bbe6e5f547463872a499cc5135b81e7a0c0402a0f7f8 20202021 3840 --paa-trust-store-path /var/paa-root-certs/
```
please have a try.

FMistrorigo, 2025-03-18:
Hi @Jerry-ESP,

I was able to commission the Thread Network in the device using the command with the correct dataset (with hex:) but there is a new error.

In the step **_'FindOperational'_** it tries 3 times to do this:

```
CHIP:DIS: Retrying operational DNS-SD discovery. Attempts remaining: 1

CHIP:DIS: Resolving 742E6403F6C9D638:0000000000000001 ...

CHIP:CTL: Session establishment failed for <0000000000000001, 1>, error:src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:119: CHIP Error 0x00000032: Timeout.  Next retry expected to get a response to Sigma1 or fail within 60 seconds
```
Once it finishes the 3 attempts it fails.

LOGS suing "hex:" before the dataset: [LOGS right command.txt](https://github.com/user-attachments/files/19313899/LOGS.right.command.txt)

I'm now sure that the problem is in the DNS-SD of the BR or in some firewall of the Wi-Fi.
Do you have some suggestions on how to check if the DNS in the BR is working?

Thanks, 
Fabio.

Jerry-ESP, 2025-03-19:
Please share all the logs of border router, and the thread br's commit id.

Did you change the code of the thread BR? If so, please try the origin br example in the repo.https://github.com/espressif/esp-thread-br/tree/main/examples/basic_thread_border_router

FMistrorigo, 2025-03-20:
Hi @Jerry-ESP,

The code of the BR is not the original one, I made some changes to be able to connect to the Wi-Fi without using the example_wifi_start. The original code works, so I guess the problem is in my code.

After other changes i was able to initialize correctly the netif for the Wi-Fi, and now the BR creates all the services that Matter needs to work (is able to resolve using DNS-SD), but still fails for another problem.

In the chip-tool logs i recieve this error: **_The Resource is busy and cannot process the request_**

Here's more context: 
```
CHIP:SC: Received error (protocol code 4) during pairing process: src/protocols/secure_channel/CASESession.cpp:1981: CHIP Error 0x000000DB: The Resource is busy and cannot process the request
CHIP:IN: SecureSession[0xffff88016710]: Released - Type:2 LSID:52152
CHIP:CTL: Device connection failed. Error src/protocols/secure_channel/CASESession.cpp:1981: CHIP Error 0x000000DB: The Resource is busy and cannot process the request
CHIP:CTL: Error on commissioning step 'FindOperational': 'src/protocols/secure_channel/CASESession.cpp:1981: CHIP Error 0x000000DB: The Resource is busy and cannot process the request'
CHIP:CTL: Failed to perform commissioning step 26
CHIP:CTL: Going from commissioning step 'FindOperational' with lastErr = 'src/protocols/secure_channel/CASESession.cpp:1981: CHIP Error 0x000000DB: The Resource is busy and cannot process the request' -> 'Cleanup'
```

Do you have idea of what the problem might be?

All the LOGS: [LOG fail during FindOperational.txt](https://github.com/user-attachments/files/19367365/LOG.fail.during.FindOperational.txt)

Jerry-ESP, 2025-03-24:
Hi @FMistrorigo ,

The question `In the chip-tool logs i recieve this error: The Resource is busy and cannot process the request` means that the device is in a case session creating process, you should not send another case session build request.

When you met this issue, you can wait for the previous process timeout, or simply reboot the device.

dhrishi, 2025-04-17:
@FMistrorigo Based on the discussion above, it doesn't seem to be a device side issue. Please close the issue if it works