espressif/esp-matter #1326: Unexpected Hue Value in TC-CC-3.2 and TC-CC-8.1 MoveHue Tests

crivi30, 2025-03-14:
This test case verifies the **Hue Move** functionality of the **Color Control Cluster server**.

### Observed Behavior for TC-CC-3.2:

#### Step 2c:
- After moving hue upwards at **5 units/s** starting from **200**, the expected hue after **5s** should be **225**, but the read value is **229**.
- **RPI Test Failure message:** The response value (**229**) is greater than the expected constraint (**227**).

**Question:** Where does the constraint value **227** come from?

#### Step 2d:
- After **15s** more, the expected hue (**300 mod 255**) should be **45**, but the read value is **53**.
- **RPI Test Pass:** The value falls within the **0-227(?)** range.

**Question:** Why is the hue increasing **faster than expected**?

### Additional Observation:
This pattern can also be seen in **TC-CC-8.1** for the **MoveToHue and StopMoveStep test**. In this test:
- Starting from **200** and moving up at **5 units/s** for **10s** should result in **250**, but instead, the read value is **254**.

Is this an implementation issue, or are these values flexible by design?

Jerry-ESP, 2025-03-17:
This is a known TestHarness issue: https://github.com/project-chip/certification-tool/issues/21, should had been fixed in the latest TH version, which TH version are you using? And what is your esp-matter version?

crivi30, 2025-03-17:
Ok, thanks. I'm using: 
- Test Harness version v2.11+fall2024;
- ESP-Matter version 1.4;
- SoC: ESP32

crivi30, 2025-03-18:
If I need to certify a light color device with SDK ESP-Matter 1.4, which is the latest version without this issue that I can use?

Jerry-ESP, 2025-03-18:
The  v2.11+fall2024 should work fine, if this issue still exist, maybe you can provide this issue link to the test Lab.

crivi30, 2025-04-04:
Hi. Yes, the issue still exist. How can i contact the test lab? 
Thanks

Jerry-ESP, 2025-04-07:
When the lab which run the matter certify test cases for you reports this issue, you can simply send them this link. Your company should have the contact information.

dhrishi, 2025-04-17:
@crivi30 Is this resolved?

crivi30, 2025-04-22:
This issue still exists in the v2.11+fall2024 version, but I guess this github issue can be closed