espressif/esp-matter #1327: Zigbee Bridge: Cluster State Updates Even When zigbee Device is Powered Off

iamiha, 2025-03-15:
**Description:**  
I am testing the `zigbee_bridge` example and have successfully commissioned it through Home Assistant. Two Zigbee devices (On/Off Light) have joined the network.  

However, I noticed an issue:  

1. I powered off the Zigbee devices.  
2. In Home Assistant, when I toggle the light through the UI, it shows as "ON," and the clusters update accordingly.  
'PRE_UPDATE' and 'POST_UPDATE' callback are triggered.
3. However, the physical Zigbee devices remain powered off, meaning the state shown in Home Assistant is incorrect.  

**Expected Behavior:**  
Home Assistant should recognize that the Zigbee device is not responding and should not update the cluster state when the device is powered off.  

**Questions:**  
- How can I ensure that the Zigbee device state is accurately reflected when the device is offline?  
- Is there a way to detect and handle device disconnection properly ? 
-  how can i inform the device is not reachable?

iamiha, 2025-03-19:
I'm trying to detect when a ZigBee device becomes unreachable by checking the **Reachable** attribute in the **Bridged Device Basic Information Cluster**. However, even when the device is powered off, the attribute still returns `true`.  

Here's my current approach:  
1. **Reading the Reachable Attribute:**
   ```cpp

    static esp_err_t app_attribute_update_cb(callback_type_t type, uint16_t endpoint_id, uint32_t cluster_id, uint32_t attribute_id, esp_matter_attr_val_t *val, void *priv_data) {
      esp_err_t err = ESP_OK;
    
      ESP_LOGI(TAG, "Type: %d, Endpoint: %" PRId16 ", Cluster: %" PRId32 ", Attribute: %" PRId32,
        type, endpoint_id, cluster_id, attribute_id);
    
      if (type == PRE_UPDATE) {
        bool reachable = false;
        if(Status::Success != BridgedDeviceBasicInformation::Attributes::Reachable::Get(endpoint_id, &reachable)) {
          ESP_LOGE(TAG, "failed to read reachable attribute");
        }
        else 
        {
          if(reachable) {
            ESP_LOGI(TAG, "Device is reachable (%s) , updating attribute", reachable ? "true" : "false");
            err = zigbee_bridge_attribute_update(endpoint_id, cluster_id, attribute_id, val, (app_bridged_device_t *)priv_data);
          }
          else {
            ESP_LOGE(TAG, "Device is not reachable (%s), skip updating", reachable ? "true" : "false");
          }
        }
      }
      return err;
    }
   ```
2.  if i updated the cluster without checking reachable , Observing `ESP_ZB_ZDO_DEVICE_UNAVAILABLE`  
   - After **one minute**, `esp_zb_app_signal_handler()` reports `ESP_ZB_ZDO_DEVICE_UNAVAILABLE`.  
   
Is there a recommended way to confirm that a device is truly offline before updating the cluster?

wqx6, 2025-04-18:
> However, even when the device is powered off, the attribute still returns true.

You should update the status of bridged devices on the bridge :
- The Bridge subscribes the zigbee devices so that it can know whether the zigbee devices are online/offline.
- When the Bridge finds one of the zigbee devices is offline, it will set the reachable attribute of the corresponding bridge endpoint to false, so that the controller can know the bridged device is not controllable. (I am not sure whether the Home Assistant will detect the reachable attribute for the bridged node. If it doesn't, you can use `emberAfEndpointEnableDisable()` to disable the endpoint when it gets offline)
- When the offline device comes to online, and the bridge will set the reachable attribute to true(or enable the endpoint by emberAfEndpointEnableDisable()``).

dhrishi, 2025-05-14:
@iamiha Please close this issue if your questions are answered

closed 2025-09-13 by dhrishi