espressif/esp-matter #1328: TC-CC-6.5 Test Step Completed [FAILED]: Step 4b: TH reads ColorTemperatureMireds attribute from DUT.

bilalahmaddev, 2025-03-15:
**Describe the bug**
WARNING    | 2025-03-15 08:07:27.903972 | Test Failure: The test expectation "value (0) == None" is false
INFO       | 2025-03-15 08:07:27.905753 | Test Step Completed [FAILED]: Step 4b: TH reads ColorTemperatureMireds attribute from DUT.

**Environment**
- ESP-Matter Commit Id:  release 1.3
- ESP-IDF Commit Id: 5.2.1
- SoC (eg: ESP32 or ESP32-C3): ESP32-S3
- Device Logs (Please attach the log file):
[

[tc-cc-6-5_2025_03_15_08_06_39.log](https://github.com/user-attachments/files/19259993/tc-cc-6-5_2025_03_15_08_06_39.log)

](url)
- Host Machine OS:

shubhamdp, 2025-03-17:
@bilalahmaddev can you please share the device logs. I see you have only shared the test harness logs.

If you are using the ZAP based data model then you can check these issues -- https://github.com/project-chip/connectedhomeip/issues/27119 and https://github.com/project-chip/connectedhomeip/issues/26793

shubhamdp, 2025-03-17:
@bilalahmaddev I ran below commands and could get the correct value of the start-up-color-temperature-mireds after reboot.

```
chip-tool pairing ble-wifi 1 (SSID) (PSK) 20202021 3840

chip-tool colorcontrol read color-temperature-mireds  1 1
# --> Data=null

chip-tool colorcontrol write start-up-color-temperature-mireds 32639 1 1

chip-tool colorcontrol read start-up-color-temperature-mireds 1 1
# --> Data=32639

# reboot-the-DUT

chip-tool colorcontrol read start-up-color-temperature-mireds 1 1
# --> Data=32639

chip-tool colorcontrol read color-temperature-mireds 1 1
# --> Data=32639
```

esp-matter: release/v1.3 (427b40d16b532fb55c00201e0dc39e8182ce13d2)
app: examples/light (without a change)

bilalahmaddev, 2025-03-17:
This is resolved thanks

closed 2025-03-17 by bilalahmaddev