espressif/esp-matter #1329: TC-IDM-10.4: An element found on the device, but the corresponding PICS OTAR.S was not found in pics list

bilalahmaddev, 2025-03-15:
[MatterTest] 03-15 13:41:06.449 INFO 
Problem: ProblemSeverity.ERROR
    test_name: PICS check
    location: 
       Endpoint: 0,
       Cluster:  42 (0x2a) OtaSoftwareUpdateRequestor
    problem: An element found on the device, but the corresponding PICS OTAR.S was not found in pics list
    spec_location: 

[MatterTest] 03-15 13:41:06.449 INFO 
Problem: ProblemSeverity.ERROR
    test_name: PICS check
    location: 
        Endpoint: 0,
        Cluster:  42 (0x2a) OtaSoftwareUpdateRequestor,
        Attribute:0 (0x00)
    problem: An element found on the device, but the corresponding PICS OTAR.S.A0000 was not found in pics list
    spec_location: 

[MatterTest] 03-15 13:41:06.450 INFO 
Problem: ProblemSeverity.ERROR
    test_name: PICS check
    location: 
        Endpoint: 0,
        Cluster:  42 (0x2a) OtaSoftwareUpdateRequestor,
        Attribute:1 (0x01)
    problem: An element found on the device, but the corresponding PICS OTAR.S.A0001 was not found in pics list
    spec_location: 

[MatterTest] 03-15 13:41:06.450 INFO 
Problem: ProblemSeverity.ERROR
    test_name: PICS check
    location: 
        Endpoint: 0,
        Cluster:  42 (0x2a) OtaSoftwareUpdateRequestor,
        Attribute:2 (0x02)
    problem: An element found on the device, but the corresponding PICS OTAR.S.A0002 was not found in pics list
    spec_location: 

[MatterTest] 03-15 13:41:06.450 INFO 
Problem: ProblemSeverity.ERROR
    test_name: PICS check
    location: 
        Endpoint: 0,
        Cluster:  42 (0x2a) OtaSoftwareUpdateRequestor,
        Attribute:3 (0x03)
    problem: An element found on the device, but the corresponding PICS OTAR.S.A0003 was not found in pics list
    spec_location: 

Environment

ESP-Matter Commit Id: release 1.3
ESP-IDF Commit Id: 5.2.1
SoC (eg: ESP32 or ESP32-C3): ESP32-S3

[TC-IDM-10-4_2025_03_15_13_39_38.log](https://github.com/user-attachments/files/19262723/TC-IDM-10-4_2025_03_15_13_39_38.log)

bilalahmaddev, 2025-03-16:
this test passes if run from command line instead of web interface.

closed 2025-03-16 by bilalahmaddev

ashish9984, 2025-10-28:
TC-IDM-10.4 failing due to cluster not found on the device but found on PICS ?
what could be the RCA for this issue ??,