espressif/esp-matter #1330: Custom Cluster for Writing Non-Writable Attributes and Triggering Reboot or Factory Reset

jzwyssig, 2025-03-16:
#### MinLevel and MaxLevel Attribute in LevelControl Cluster
I am using esp-matter 1.2 and I want to write the **MinLevel** and **MaxLevel** attributes in the LevelControl cluster. These **attributes are read-only**, so while I can set them in firmware, there’s no option to modify them over Matter.

The best approach seems to be creating a **custom cluster** with **writable** MinLevel and MaxLevel attributes. When these values are updated, we would adjust the LevelControl attributes accordingly. A manufacturer app is used to write the custom cluster's attributes over Matter.

Alternatively, modifying esp-matter to make these attributes writable is possible but would break the Matter specification. Therefore, this is no option.

Is there a recommended way to handle this without using a custom cluster?

#### Factory Reset and Reboot Options
Additionally, I could not find a specification to trigger a device **reboot or factory reset using Matter**.

Is there an existing method to trigger a factory reset and/or a reboot in Matter, or would we need to implement these functions within a custom cluster as well?

Thank you for your guidance!

shubhamdp, 2025-03-17:
@jzwyssig, You are right, these are read-only attributes. You can use any manufacturer specific method to update these.

As suggested by you, you can develop the custom cluster which treats these values as writable and make the needed adjustments in the cluster to update the ones in the level-control or vice-a-versa.

You can find check the [rainmaker cluster](https://github.com/espressif/esp-matter/tree/release/v1.2/components/esp_matter_rainmaker)'s implementation for reference.

Same goes for the factory-reset, there is no external trigger for this. In most cases (that I have came across) we interact with physical device to perform the factory reset.

> Is there an option to change the flags of a cluster after its creation?

Yes, you can, but there is only one operation that you can perform, If it has `CLUSTER_FLAG_SERVER` you can add `CLUSTER_FLAG_CLIENT`, and if it has `CLUSTER_FLAG_CLIENT` you can add `CLUSTER_FLAG_SERVER`. You can do this by creating the cluster again with different flag.
Please check: https://github.com/espressif/esp-matter/blob/release/v1.2/components/esp_matter/esp_matter_core.cpp#L1642

jzwyssig, 2025-03-18:
Thank you @shubhamdp! That answers my questions.