espressif/esp-matter #1331: Issue with OnOffTransitionTime Behavior

jzwyssig, 2025-03-16:
### 1. Wrong Behavior of On/Off Commands
The `OnOffTransitionTime` does not apply to on/off commands, although the specification suggests it should:

> **1.6.6.10. OnOffTransitionTime Attribute**
*The OnOffTransitionTime attribute represents the time taken to move to or from the target level
when On or Off commands are received by an On/Off cluster on the same endpoint. It is specified in
1/10ths of a second.
The actual time taken SHOULD be as close to OnOffTransitionTime as the device is able. Please note
that if the device is not able to move at a variable rate, the OnOffTransitionTime attribute SHOULD
NOT be implemented.*

Instead, the *on* command directly switches to the target brightness without gradually increasing. The same applies to the *off* command. The lighting feature is enabled by default. I am using the official esp-matter 1.2 Docker container with ESP-IDF 5.2.1.

### 2. Transition Time
I noticed that the `OnOffTransitionTime` remains constant, regardless of the relative change in brightness. This seems to be the expected behavior, but I'm curious if there is a way to modify it?

**Example**:
 With a 10s transition time, e.g., `OnOffTransitionTime` attribute ist set to 100:
- Dimming from 50% → 51% takes 10s
- Dimming from 50% → 100% also takes 10s

I want the transition time to apply to the full brightness range.  The transition time should scale based on the brightness change.  For example, a transition time of 10s means that dimming from 50% to 80% takes 3s.

**Problems During Fading**
For small values of `OnOffTransitionTime`, the dimming changes rapidly. As a result, many CHIP log messages are printed, and the new brightness value is written to flash storage every time.
- Of course, for production, I can disable UART, but I also want to do that for testing. How can I suppress CHIP log messages?
- I also realized that from Matter 1.3 onwards, there is an option to suppress the constant flash writes using `attribute::set_deferred_persistence(current_level_attribute);`, which is not possible in Matter 1.2. How can I handle this issue in Matter 1.2?

Jerry-ESP, 2025-03-17:
1. The OnOffTransitiontime is an optional attribute in the LevelControl cluster, so we don't add it default, if you need to use it, you can add it manually.  https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#attributes-and-commands
2. The function you want has not need implemented in the sdk, you can implement it through change the attribute manually.
3. If you develop new product, we advice you use the Matter 1.3 or 1.4, this two versions are more stable. If you still want to matter 1.2, you can try to cherry pick this commit: https://github.com/espressif/esp-matter/commit/80a0868271965e1959bf9ffc7743a2380f0f86aa  , and the option to change the log level in menuconfig :  CONFIG_LOG_DEFAULT_LEVEL_ERROR=y

jzwyssig, 2025-03-18:
Thank you @Jerry-ESP for the fast response!

1. I already implemented the `OnOffTransitionTime` cluster and tested it. As I already mentioned, for brightness levels it works perfectly fine but it doesn't for on/off commands. **Maybe, this is a bug**. At least it is strange that the fading works for brightness levels but not for on/off commands. Therefore, I will not use it for now and may consider it if we switch to Matter 1.4.
*Can you confirm that this is a bug, e.g. not working if you for example dimm from 0% (off) to 70% (on) using the `on` command?*

2. In order to achieve the desired behaviour of a constant fade time, e.g. 10s across the range from 0% to 100%, I would need to store the value of the full brightness, e.g. 10s, in a custom cluster. I would need to adjust the `OnOffTransitionTime` before the target level is set. E.g. if the device is at 40%, we receive 70%, I would need to set `OnOffTransitionTime` to 3s.
*Is this what you mean by implement it manually?*

3. Okay, but the option `CONFIG_LOG_DEFAULT_LEVEL_ERROR=y` disables all info and warning log messages across the system. Is there a better way? I tried `chip::Logging::SetLogFilter(chip::Logging::kLogCategory_None);` as suggested in #999 but it did not work.

Jerry-ESP, 2025-03-18:
Hi @jzwyssig ,

1. You need to add a #define EMBER_AF_PLUGIN_LEVEL_CONTROL in the https://github.com/espressif/esp-matter/blob/39c87bd9b646533c79d128a598997ce1522f6c2c/components/esp_matter/zap_common/zap-generated/gen_config.h#L155
2. This method seems OK, but I think it is not easy to implement in current SDK, it will execute immediately when receive the on/off command.
3.   There are several log levels you can choose:
CONFIG_LOG_DEFAULT_LEVEL_NONE
CONFIG_LOG_DEFAULT_LEVEL_ERROR
CONFIG_LOG_DEFAULT_LEVEL_WARN
CONFIG_LOG_DEFAULT_LEVEL_INFO
CONFIG_LOG_DEFAULT_LEVEL_DEBUG
CONFIG_LOG_DEFAULT_LEVEL_VERBOSE

closed 2025-03-18 by jzwyssig