espressif/esp-matter #1334: light brightness error with turn off light

eiSouthBoy, 2025-03-19:
**Describe the bug**
A clear and concise description of what the bug is.

**Environment**
- ESP-Matter Commit Id: f7646bf8bb1730d51212bd49711650d7bf4613b8
- ESP-IDF Commit Id: 3b8741b172dc951e18509698dee938304bcf1523
- SoC (eg: ESP32 or ESP32-C3): ESP32-C3
- Device Logs (Please attach the log file): light.log

- Host Machine OS: Ubuntu 22.04
- Host Machine Python version: 3.10.12

- Commissioner app and versions if present:
- Home hub app and versions if present:
- Commissioner's logs if present:

**Any additional details**

I'm using Light example, which can compile to the ESP32-C3-DevKitM-1 development board normally, and can also be bound to the matter-hub-app. When the light is on, adjust the brightness to 50%, and then when I turn off the light, the brightness on the UI is displayed as 1%.
This result is not what I need. When I turn off the light, the brightness on the UI should not change, and when I turn on the light, the brightness should also remain the same as before it goes off.

I need your help~
Thank you!

[light.log](https://github.com/user-attachments/files/19332243/light.log)

jadhavrohit924, 2025-03-19:
@eiSouthBoy Please take a look at a [duplicate](https://github.com/espressif/esp-matter/issues/1266#issuecomment-2668143933) issue.

eiSouthBoy, 2025-03-20:
I've looked at the issue about brightness, but my issue hasn't been resolved.
When OnOff=false, the brightness should not change to, but should remain the same, and only the light will be turned off.

Thank you!

jadhavrohit924, 2025-03-20:
The above comment has solutions, too. If you remove the `OnOff` feature from the `LevelControl` cluster, the Light will turn off, and the brightnesses will not change.

jonsmirl, 2025-03-20:
I agree, this behavior is a configurable option. You want to select the other option.

eiSouthBoy, 2025-04-02:
> The above comment has solutions, too. If you remove the `OnOff` feature from the `LevelControl` cluster, the Light will turn off, and the brightnesses will not change.

I have read *1.6 Level Control Cluster* in *Matter-1.4-Application-Cluster-Specification.pdf*. The FeatureMap of this Cluster supports OnOff Feature by default. According to what you mean, I should set FeatureMap to cancel the support for OnOff, right?