espressif/esp-matter #1337: Power Source Cluster issue

sdrasti, 2025-03-20:
I have added the power source cluster for a battery operated device. 

The `PowerSource::Attributes::BatPercentRemaining` attribute can't be set, since it is not instantiated when the Power Source Cluster is created. 

``` C
power_source_device::config_t                          powerSourceConfig;
cluster::power_source::feature::battery::config_t      battery_config;
cluster::power_source::feature::rechargeable::config_t rechargeable_config;

endpoint = power_source_device::create(Node, &powerSourceConfig, ENDPOINT_FLAG_NONE, NULL);

esp_matter::cluster_t *Cluster = esp_matter::cluster::get(endpoint, chip::app::Clusters::PowerSource::Id);
battery_config.bat_charge_level = chip::to_underlying(PowerSource::BatChargeLevelEnum::kWarning);
rechargeable_config.bat_charge_state = chip::to_underlying(PowerSource::BatChargeStateEnum::kIsNotCharging);
cluster::power_source::feature::battery::add(Cluster, &battery_config);
cluster::power_source::feature::rechargeable::add(Cluster, &rechargeable_config);
```

The Cluster is basically working, since the BatChargeLevel-Warning is shown in a commissioned device.

Have I missed something in the power_source config/feature setup?

jadhavrohit924, 2025-03-20:
@sdrasti As `BatPercentRemaining` is an optional attribute for the `Battery` feature, have you enabled the attribute in the application? [This](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#attributes-and-commands) is how an optional attribute can be added.

sdrasti, 2025-03-20:
@jadhavrohit924 That did the trick. Thank you!