espressif/esp-matter #1338: ipad default name "Matter Accessory" instead of the "Product Name" specified in our DCL.

ethan-jy, 2025-03-21:
Hello,

We have developed a Matter device using Espressif chips and set up a DCL (Device Code Library). However, when adding the device to an Apple iPad, it shows the default name "Matter Accessory" instead of the "Product Name" specified in our DCL. Our model "XLC MA" is registered in the DCL, but the issue persists. Could you help us identify why the DCL configuration isn’t working as expected and how to resolve this? We’re unsure where to start.

Thank you!

[image]

[image]

wqx6, 2025-03-21:
Do the Vendor ID/Name and the Product ID/Name of your product(the attributes of BasicInfo cluster) match the model that you uploaded to the DCL? I have tested commissioning a device with the Vendor ID 0x1592 and Product ID 0x1 with Apple home but didn't encounter the issue.

ethan-jy, 2025-03-21:
> Do the Vendor ID/Name and the Product ID/Name of your product(the attributes of BasicInfo cluster) match the model that you uploaded to the DCL? I have tested commissioning a device with the Vendor ID 0x1592 and Product ID 0x1 with Apple home but didn't encounter the issue.

Hi，thanks.
Here is the information I read from the device: the product name is XLC MA, but the DCL lists it as XLC MA (CN). Could this discrepancy be the cause of the issue?

ethan-jy, 2025-03-21:
> Hi，thanks. Here is the information I read from the device: the product name is XLC MA, but the DCL lists it as XLC MA (CN). Could this discrepancy be the cause of the issue?
VendorID:5522
vendorName:MEAN WELL
ProductID:1
ProductName: XLC MA

wqx6, 2025-03-21:
In my test, the product name will not affect the device name when commissioning with Apple Home app.

Could you also check the vendor ID, product ID in the DAC of your device? Also please try to update your iOS version and Apple home version?

ethan-jy, 2025-03-24:
> In my test, the product name will not affect the device name when commissioning with Apple Home app.
> 
> Could you also check the vendor ID, product ID in the DAC of your device? Also please try to update your iOS version and Apple home version?

Hi，thanks。

My iPad is already on the latest version. According to the tutorial, I read the Vendor ID (VID) and Product ID (PID) from the DAC, which were provided by Espressif:
app_main: VID in DAC: 4891, PID in DAC: 32830
Our PAA uses Espressif's solution, and the DAC was also programmed by Espressif.
Could there be an error in our DCL configuration? Do we need to use Espressif's Vendor ID and Product ID in the DCL?

wqx6, 2025-03-24:
How do you generate your CD file? Do you use `--dac-origin-product-id` and `--dac-origin-vendor-id` in your CD? And how do you generate your factory binary?

ethan-jy, 2025-03-24:
Hi,

Our product has passed certification and obtained the Certified Device (CD) files. The ESP32H2 modules were pre-flashed with the Device Attestation Certificate (DAC) during incoming material inspection, as we purchased this service. The product is now in mass production. However, we noticed that when pairing the device with an Apple iPad, the default displayed name is "Matter Accessory". We want to customize this name to show our product name instead. Other companies like Eve Systems have achieved this for their Matter-enabled products.

Our factory firmware flashing instructions:

ethan-jy, 2025-03-24:
> Hi,
>
> Our product has passed certification and obtained the Certified Device (CD) files. The ESP32H2 modules were pre-flashed with the Device Attestation Certificate (DAC) during incoming material [...]

[image]

wqx6, 2025-03-24:
Still cannot reproduce the issue, In my test, the Apple home app can always get the Product name from DCL.

- iOS version 18.3.2
- Homepod mini version:18.3

ethan-jy, 2025-03-24:
Thank you. The issue has been resolved after I replaced my tablet with the latest iPhone, which could detect the HomePod's DCL name during pairing. It might be because my tablet, despite having the latest system version, was connecting to an outdated HomePod software version.

closed 2025-03-24 by ethan-jy