espressif/esp-matter #1339: changing names for the app

ravencarcass, 2025-03-23:
I have a question about how to change the display of a product in the google home app or other apps. I have a total of 12 endpoints and would like to change the names from “test_product (X)” to another name. Is there any way to do this? Because it irritates me sometimes that I don't know which button is for which output. 

[image]

Jerry-ESP, 2025-03-24:
I think you can refer to this example: https://github.com/espressif/esp-matter/blob/main/examples/generic_switch/main/app_main.cpp   to use the tag-list attribute in the descriptor cluster to identify every endpoint function.

ravencarcass, 2025-03-24:
After took a look to the code but I don't see it how I can relise it. This is my code at the moment: 
```
static const char *TAG = "app_main";
uint16_t light_endpoint_id = 0;
uint16_t light_endpoint_id_1 = 1;
uint16_t light_endpoint_id_2 = 2;
uint16_t light_endpoint_id_3 = 3;
uint16_t light_endpoint_id_4 = 4;
uint16_t light_endpoint_id_5 = 5;
uint16_t light_endpoint_id_6 = 6;
uint16_t light_endpoint_id_7 = 7;
uint16_t light_endpoint_id_8 = 8;
uint16_t light_endpoint_id_9 = 9;
uint16_t light_endpoint_id_10 = 10;
uint16_t light_endpoint_id_11 = 11;

```
```
node::config_t node_config;

    // node handle can be used to add/modify other endpoints.
    node_t *node = node::create(&node_config, app_attribute_update_cb, app_identification_cb);
    ABORT_APP_ON_FAILURE(node != nullptr, ESP_LOGE(TAG, "Failed to create Matter node"));

    /*extended_color_light::config_t light_config;
    light_config.on_off.on_off = DEFAULT_POWER;
    light_config.on_off.lighting.start_up_on_off = nullptr;
    light_config.level_control.current_level = DEFAULT_BRIGHTNESS;
    light_config.level_control.on_level = DEFAULT_BRIGHTNESS;
    light_config.level_control.lighting.start_up_current_level = DEFAULT_BRIGHTNESS;
    light_config.color_control.color_mode = (uint8_t)ColorControl::ColorMode::kColorTemperature;
    light_config.color_control.enhanced_color_mode = (uint8_t)ColorControl::ColorMode::kColorTemperature;
    light_config.color_control.color_temperature.startup_color_temperature_mireds = nullptr;*/

    dimmable_light::config_t light_config;
    light_config.on_off.on_off = DEFAULT_POWER;
    light_config.level_control.current_level = DEFAULT_BRIGHTNESS;
    light_config.level_control.on_level = DEFAULT_BRIGHTNESS;
    light_config.level_control.lighting.start_up_current_level = DEFAULT_BRIGHTNESS;

    // endpoint handles can be used to add/modify clusters.
    endpoint_t *endpoint = dimmable_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
    ABORT_APP_ON_FAILURE(endpoint != nullptr, ESP_LOGE(TAG, "Failed to create extended color light endpoint"));

    light_endpoint_id = endpoint::get_id(endpoint);
    ESP_LOGI(TAG, "Light created with endpoint_id %d", light_endpoint_id);

///////////////////////////////////////////////////////////////////////////

    endpoint_t *endpoint_1 = dimmable_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
    ABORT_APP_ON_FAILURE(endpoint != nullptr, ESP_LOGE(TAG, "Failed to create extended color light endpoint"));

    light_endpoint_id = endpoint::get_id(endpoint_1);
    ESP_LOGI(TAG, "Light created with endpoint_id %d", light_endpoint_id_1);

//////////////////////////////////////////////////////////////////////////

    endpoint_t *endpoint_2 = dimmable_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
    ABORT_APP_ON_FAILURE(endpoint != nullptr, ESP_LOGE(TAG, "Failed to create extended color light endpoint"));

    light_endpoint_id = endpoint::get_id(endpoint_2);
    ESP_LOGI(TAG, "Light created with endpoint_id %d", light_endpoint_id_2);

//////////////////////////////////////////////////////////////////////////

    endpoint_t *endpoint_3 = dimmable_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
    ABORT_APP_ON_FAILURE(endpoint != nullptr, ESP_LOGE(TAG, "Failed to create extended color light endpoint"));

    light_endpoint_id = endpoint::get_id(endpoint_3);
    ESP_LOGI(TAG, "Light created with endpoint_id %d", light_endpoint_id_3);
[... 43 lines cut ...]
endpoint_t *endpoint_9 = dimmable_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
ABORT_APP_ON_FAILURE(endpoint != nullptr, ESP_LOGE(TAG, "Failed to create extended color light endpoint"));

light_endpoint_id = endpoint::get_id(endpoint_9);
ESP_LOGI(TAG, "Light created with endpoint_id %d", light_endpoint_id_9);

//////////////////////////////////////////////////////////////////////////

endpoint_t *endpoint_10 = dimmable_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
ABORT_APP_ON_FAILURE(endpoint != nullptr, ESP_LOGE(TAG, "Failed to create extended color light endpoint"));

light_endpoint_id = endpoint::get_id(endpoint_10);
ESP_LOGI(TAG, "Light created with endpoint_id %d", light_endpoint_id_10);

//////////////////////////////////////////////////////////////////////////

endpoint_t *endpoint_11 = dimmable_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
ABORT_APP_ON_FAILURE(endpoint != nullptr, ESP_LOGE(TAG, "Failed to create extended color light endpoint"));

light_endpoint_id = endpoint::get_id(endpoint_11);
ESP_LOGI(TAG, "Light created with endpoint_id %d", light_endpoint_id_11);

//////////////////////////////////////////////////////////////////////////

```

Jerry-ESP, 2025-03-25:
You just need to define the [taglist](https://github.com/espressif/esp-matter/blob/9be61fd27afa9378cca9e62b296f6ac961306d97/examples/generic_switch/main/app_main.cpp#L57) and set it to every [endpoint](https://github.com/espressif/esp-matter/blob/9be61fd27afa9378cca9e62b296f6ac961306d97/examples/generic_switch/main/app_main.cpp#L242)

dhrishi, 2025-04-17:
@ravencarcass Please close the issue if resolved

ravencarcass, 2025-04-17:
It's not solved. I still have some issues with it.

Jerry-ESP, 2025-04-18:
So please share your detail implementation of the taglist attribute, please note that you need to add the taglist feature first, it is not included in the descriptor cluster because it is optional.

ravencarcass, 2025-04-18:
how can I add the taglist feature? 

This is the complete main code.

```
#include "stdio.h"
#include "driver/i2c.h"
#include "driver/i2c_master.h"
#include "driver/gpio.h"
#include "driver/ledc.h"
#include "driver/spi_master.h"

#include <esp_err.h>
#include <esp_log.h>
#include <nvs_flash.h>

#include <esp_matter.h>
#include <esp_matter_console.h>
#include <esp_matter_ota.h>
#include <app/util/attribute-storage.h>

#include <common_macros.h>
#include <app_priv.h>
#include <app_reset.h>
#if CHIP_DEVICE_CONFIG_ENABLE_THREAD
#include <platform/ESP32/OpenthreadLauncher.h>
#endif

#include <app/server/CommissioningWindowManager.h>
#include <app/server/Server.h>

#ifdef CONFIG_ENABLE_SET_CERT_DECLARATION_API
#include <esp_matter_providers.h>
#include <lib/support/Span.h>
#ifdef CONFIG_SEC_CERT_DAC_PROVIDER
#include <platform/ESP32/ESP32SecureCertDACProvider.h>
#elif defined(CONFIG_FACTORY_PARTITION_DAC_PROVIDER)
#include <platform/ESP32/ESP32FactoryDataProvider.h>
#endif
using namespace chip::DeviceLayer;
#endif

static const char *TAG = "app_main";
uint16_t light_endpoint_id = 0;
uint16_t light_endpoint_id_1 = 0;
uint16_t light_endpoint_id_2 = 0;
uint16_t light_endpoint_id_3 = 0;
uint16_t light_endpoint_id_4 = 0;
uint16_t light_endpoint_id_5 = 0;
uint16_t light_endpoint_id_6 = 0;
uint16_t light_endpoint_id_7 = 0;
uint16_t light_endpoint_id_8 = 0;
uint16_t light_endpoint_id_9 = 0;
uint16_t light_endpoint_id_10 = 0;
uint16_t light_endpoint_id_11 = 0;

#define I2C_MASTER_SCL_IO   8
#define I2C_MASTER_SDA_IO   10
#define I2C_MASTER_FREQ_HZ  400000
//#define 12C_MASTER_NUM      I2C_NUM_0

#define MISO  20
#define MOSI  21
#define CLK   22
#define CS    23

//#define STEP 5
#define DIR GPIO_NUM_6
#define RESET GPIO_NUM_7
#define SLEEP GPIO_NUM_0
[... 340 lines cut ...]

/* Matter start */
err = esp_matter::start(app_event_cb);
ABORT_APP_ON_FAILURE(err == ESP_OK, ESP_LOGE(TAG, "Failed to start Matter, err:%d", err));

/* Starting driver with default values */
//app_driver_light_set_defaults(light_endpoint_id);

#if CONFIG_ENABLE_ENCRYPTED_OTA
err = esp_matter_ota_requestor_encrypted_init(s_decryption_key, s_decryption_key_len);
ABORT_APP_ON_FAILURE(err == ESP_OK, ESP_LOGE(TAG, "Failed to initialized the encrypted OTA, err: %d", err));
#endif // CONFIG_ENABLE_ENCRYPTED_OTA

#if CONFIG_ENABLE_CHIP_SHELL
esp_matter::console::diagnostics_register_commands();
esp_matter::console::wifi_register_commands();
esp_matter::console::factoryreset_register_commands();
#if CONFIG_OPENTHREAD_CLI
esp_matter::console::otcli_register_commands();
#endif
esp_matter::console::init();
#endif
}

```

Jerry-ESP, 2025-04-21:
Please refer to this code: https://github.com/espressif/esp-matter/blob/211d2761e528d2edeb15a5c215cd2ad5c37e99e3/examples/generic_switch/main/app_main.cpp#L139
to add this feature, and you should call the SetTagList() api after the err = esp_matter::start(app_event_cb);

Jerry-ESP, 2025-04-23:
I haven't found add taglist feature in your code, please check, you'd better test through the chip-tool first, to check if you have successfully add the taglist feature and set the correct value.  https://github.com/espressif/esp-matter/blob/211d2761e528d2edeb15a5c215cd2ad5c37e99e3/examples/generic_switch/main/app_main.cpp#L140

ravencarcass, 2025-04-23:
I'm sorry, but it's not clear to me how it changed the name. Because nothing has changed in the app. Can I ask you for a favor? Are there any examples for the light example where they do this as well? That might give me more clarity. Because it's been confusing me for a few weeks.
```

#include "stdio.h"
#include "driver/i2c.h"
#include "driver/i2c_master.h"
#include "driver/gpio.h"
#include "driver/ledc.h"
#include "driver/spi_master.h"

#include <esp_err.h>
#include <esp_log.h>
#include <nvs_flash.h>

#include <esp_matter.h>
#include <esp_matter_console.h>
#include <esp_matter_ota.h>
#include <app/util/attribute-storage.h>

#include <common_macros.h>
#include <app_priv.h>
#include <app_reset.h>
#if CHIP_DEVICE_CONFIG_ENABLE_THREAD
#include <platform/ESP32/OpenthreadLauncher.h>
#endif

#include <app/server/CommissioningWindowManager.h>
#include <app/server/Server.h>

#ifdef CONFIG_ENABLE_SET_CERT_DECLARATION_API
#include <esp_matter_providers.h>
#include <lib/support/Span.h>
#ifdef CONFIG_SEC_CERT_DAC_PROVIDER
#include <platform/ESP32/ESP32SecureCertDACProvider.h>
#elif defined(CONFIG_FACTORY_PARTITION_DAC_PROVIDER)
#include <platform/ESP32/ESP32FactoryDataProvider.h>
#endif
using namespace chip::DeviceLayer;
#endif

static const char *TAG = "app_main";
uint16_t light_endpoint_id = 0;
uint16_t light_endpoint_id_1 = 0;
uint16_t light_endpoint_id_2 = 0;
uint16_t light_endpoint_id_3 = 0;
uint16_t light_endpoint_id_4 = 0;
uint16_t light_endpoint_id_5 = 0;
uint16_t light_endpoint_id_6 = 0;
uint16_t light_endpoint_id_7 = 0;
uint16_t light_endpoint_id_8 = 0;
uint16_t light_endpoint_id_9 = 0;
uint16_t light_endpoint_id_10 = 0;
uint16_t light_endpoint_id_11 = 0;

#define I2C_MASTER_SCL_IO   8
#define I2C_MASTER_SDA_IO   10
#define I2C_MASTER_FREQ_HZ  400000
//#define 12C_MASTER_NUM      I2C_NUM_0

#define MISO  20
#define MOSI  21
#define CLK   22
#define CS    23

//#define STEP 5
#define DIR GPIO_NUM_6
#define RESET GPIO_NUM_7
#define SLEEP GPIO_NUM_0

#define LEDC_TIMER        LEDC_TIMER_0
[... 375 lines cut ...]
ABORT_APP_ON_FAILURE(err == ESP_OK, ESP_LOGE(TAG, "Failed to start Matter, err:%d", err));

/* Starting driver with default values */
//app_driver_light_set_defaults(light_endpoint_id);

#if CONFIG_ENABLE_ENCRYPTED_OTA
err = esp_matter_ota_requestor_encrypted_init(s_decryption_key, s_decryption_key_len);
ABORT_APP_ON_FAILURE(err == ESP_OK, ESP_LOGE(TAG, "Failed to initialized the encrypted OTA, err: %d", err));
#endif // CONFIG_ENABLE_ENCRYPTED_OTA

SetTagList(1, chip::Span<const Descriptor::Structs::SemanticTagStruct::Type>(horizontalControlTagList));
//SetTagList(2, chip::Span<const Descriptor::Structs::SemanticTagStruct::Type>(gEp2TagList));

#if CONFIG_ENABLE_CHIP_SHELL
esp_matter::console::diagnostics_register_commands();
esp_matter::console::wifi_register_commands();
esp_matter::console::factoryreset_register_commands();
#if CONFIG_OPENTHREAD_CLI
esp_matter::console::otcli_register_commands();
#endif
esp_matter::console::init();
#endif
}

```

Jerry-ESP, 2025-04-24:
I don't think there is a better way to rename the endpoints, the taglist is defined for this function, but has not been supported in this three platforms.

ravencarcass, 2025-04-24:
I4ve tried it in the rain maker app. But they don't show the endpoints.

Jerry-ESP, 2025-04-25:
> I tried it in the rain maker app. But they don't show the endpoints apart. What I understand now is that it is not possible to rename an endpoint for a google home app.

Correct, Google home app doesn't support rename endpoints by taglist attributes now.

ravencarcass, 2025-04-25:
So their is at the moment no way to rename the endpoints for a Google home app or other apps.

ravencarcass, 2025-04-26:
Okay, 
So if I couldn't do it with the namespaces for a google home app. Then Do I need to rename them by hand in the google home app. In this way: longer tap on the tile you would like to change the name > settings > tap on name > give the name you will > save. The name is now temporary changed. Thanks for the support, unfortunately it didn't work out as I would because of the limitations for Google Home app.

closed 2025-04-26 by ravencarcass