espressif/esp-matter #1340: Change light example from WiWi to Thread?

olavt, 2025-03-23:
What changes are required to change the light example from WiFi to Thread?

I tried to change the following in the SDK Configuration editor:
- Check the OPENTHREAD_ENABLED option
- Clear the ENABLE_WIFI_STATION option

But, I get build errors like these:

/Users/olavt/esp/v5.4/esp-idf/components/openthread/openthread/src/core/meshcop/secure_transport.cpp:340:5: error: 'mbedtls_ssl_conf_handshake_timeout' was not declared in this scope; did you mean 'mbedtls_ssl_conf_read_timeout'?
  340 |     mbedtls_ssl_conf_handshake_timeout(&mConf, 8000, 60000);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |     mbedtls_ssl_conf_read_timeout
/Users/olavt/esp/v5.4/esp-idf/components/openthread/openthread/src/core/meshcop/secure_transport.cpp:349:9: error: 'mbedtls_ssl_conf_dtls_cookies' was not declared in this scope; did you mean 'mbedtls_ssl_conf_dn_hints'?
  349 |         mbedtls_ssl_conf_dtls_cookies(&mConf, mbedtls_ssl_cookie_write, mbedtls_ssl_cookie_check, &mCookieCtx);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |         mbedtls_ssl_conf_dn_hints
/Users/olavt/esp/v5.4/esp-idf/components/openthread/openthread/src/core/meshcop/secure_transport.cpp:365:16: error: 'mbedtls_ssl_set_hs_ecjpake_password' was not declared in this scope
  365 |         rval = mbedtls_ssl_set_hs_ecjpake_password(&mSsl, mPsk, mPskLength);
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/olavt/esp/v5.4/esp-idf/components/openthread/openthread/src/core/meshcop/secure_transport.cpp: In member function 'ot::Error ot::MeshCoP::SecureTransport::SetClientId(const uint8_t*, uint8_t)':
/Users/olavt/esp/v5.4/esp-idf/components/openthread/openthread/src/core/meshcop/secure_transport.cpp:746:16: error: 'mbedtls_ssl_set_client_transport_id' was not declared in this scope; did you mean 'mbedtls_ssl_conf_transport'?
  746 |     int rval = mbedtls_ssl_set_client_transport_id(&mSsl, aClientId, aLength);
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                mbedtls_ssl_conf_transport
/Users/olavt/esp/v5.4/esp-idf/components/openthread/openthread/src/core/meshcop/secure_transport.cpp: In member function 'void ot::MeshCoP::SecureTransport::Process()':
/Users/olavt/esp/v5.4/esp-idf/components/openthread/openthread/src/core/meshcop/secure_transport.cpp:1126:17: error: 'mbedtls_ssl_set_hs_ecjpake_password' was not declared in this scope
 1126 |                 mbedtls_ssl_set_hs_ecjpake_password(&mSsl, mPsk, mPskLength);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[264/1679] Performing build step for 'chip_gn'

Jerry-ESP, 2025-03-24:
@olavt You can simply change the target to esp32h2 to switch from WiFi to thread,  you can refer to  this file for thread device configuration https://github.com/espressif/esp-matter/blob/main/examples/light/sdkconfig.defaults.esp32h2

olavt, 2025-03-24:
I set the Espressif Device Target to "espp32h2" and still got these build errors:

/Users/olavt/esp/v5.4/esp-idf/components/openthread/openthread/src/core/meshcop/secure_transport.cpp: In member function 'ot::Error ot::MeshCoP::SecureTransport::Setup(bool)':
/Users/olavt/esp/v5.4/esp-idf/components/openthread/openthread/src/core/meshcop/secure_transport.cpp:340:5: error: 'mbedtls_ssl_conf_handshake_timeout' was not declared in this scope; did you mean 'mbedtls_ssl_conf_read_timeout'?
  340 |     mbedtls_ssl_conf_handshake_timeout(&mConf, 8000, 60000);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |     mbedtls_ssl_conf_read_timeout
/Users/olavt/esp/v5.4/esp-idf/components/openthread/openthread/src/core/meshcop/secure_transport.cpp:349:9: error: 'mbedtls_ssl_conf_dtls_cookies' was not declared in this scope; did you mean 'mbedtls_ssl_conf_dn_hints'?
  349 |         mbedtls_ssl_conf_dtls_cookies(&mConf, mbedtls_ssl_cookie_write, mbedtls_ssl_cookie_check, &mCookieCtx);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |         mbedtls_ssl_conf_dn_hints
/Users/olavt/esp/v5.4/esp-idf/components/openthread/openthread/src/core/meshcop/secure_transport.cpp:365:16: error: 'mbedtls_ssl_set_hs_ecjpake_password' was not declared in this scope
  365 |         rval = mbedtls_ssl_set_hs_ecjpake_password(&mSsl, mPsk, mPskLength);
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/olavt/esp/v5.4/esp-idf/components/openthread/openthread/src/core/meshcop/secure_transport.cpp: In member function 'ot::Error ot::MeshCoP::SecureTransport::SetClientId(const uint8_t*, uint8_t)':
/Users/olavt/esp/v5.4/esp-idf/components/openthread/openthread/src/core/meshcop/secure_transport.cpp:746:16: error: 'mbedtls_ssl_set_client_transport_id' was not declared in this scope; did you mean 'mbedtls_ssl_conf_transport'?
  746 |     int rval = mbedtls_ssl_set_client_transport_id(&mSsl, aClientId, aLength);
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                mbedtls_ssl_conf_transport
/Users/olavt/esp/v5.4/esp-idf/components/openthread/openthread/src/core/meshcop/secure_transport.cpp: In member function 'void ot::MeshCoP::SecureTransport::Process()':
/Users/olavt/esp/v5.4/esp-idf/components/openthread/openthread/src/core/meshcop/secure_transport.cpp:1126:17: error: 'mbedtls_ssl_set_hs_ecjpake_password' was not declared in this scope
 1126 |                 mbedtls_ssl_set_hs_ecjpake_password(&mSsl, mPsk, mPskLength);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[266/1536] Building CXX object esp-idf/openthread/CMakeFiles/__idf_openthread.dir/openthread/src/core/net/srp_client.cpp.obj
ninja: build stopped: subcommand failed.

 *  The terminal process "ninja" terminated with exit code: 1.

Jerry-ESP, 2025-03-25:
@olavt If you want to run a thread device on esp32c6, you can use the sdkconfig.defaults.c6_thread, the command is:
idf.py -D SDKCONFIG_DEFAULTS="sdkconfig.defaults.c6_thread" set-target esp32c6

olavt, 2025-03-25:
@Jerry-ESP I use the Visual Studio ESP-IDF Extension with Matter.

Should I run this command from a terminal window in VS Code?

olavt, 2025-03-25:
I use VS Code with the Esptressif ESP-IDF Extension with Matter support.

I executed the command and got quite a few warnings and when I build the project I still get errors:

olavt@Olavs-MacBook-Pro light % idf.py -D SDKCONFIG_DEFAULTS="sdkconfig.defaults.c6_thread" set-target esp32c6
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Executing action: set-target
Set Target to: esp32c6, new sdkconfig will be created.
Running cmake in directory /Users/olavt/Source/Repos/light/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/Users/olavt/.espressif/python_env/idf5.4_py3.9_env/bin/python -DESP_PLATFORM=1 -DSDKCONFIG_DEFAULTS=sdkconfig.defaults.c6_thread -DIDF_TARGET=esp32c6 -DCCACHE_ENABLE=0 /Users/olavt/Source/Repos/light"...
-- Existing sdkconfig '/Users/olavt/Source/Repos/light/sdkconfig' renamed to '/Users/olavt/Source/Repos/light/sdkconfig.old'.
-- Found Git: /usr/bin/git (found version "2.39.5 (Apple Git-154)")
IDF_VERSION: 5.4.0; idf version number: 5.4.0
Sequence of SDKCONFIG files: sdkconfig.defaults;sdkconfig.defaults.c6_thread
-- Component directory /Users/olavt/.espressif/esp-matter/examples/common/blemesh_platform does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/olavt/.espressif/esp-matter/examples/common/cmake_common does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/olavt/.espressif/esp-matter/examples/common/external_platform does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/olavt/.espressif/esp-matter/examples/common/relinker does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/olavt/.espressif/esp-matter/examples/common/secondary_ble_adv does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/olavt/.espressif/esp-matter/examples/common/utils does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 14.2.0
-- The CXX compiler identification is GNU 14.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/olavt/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/riscv32-esp-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/olavt/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/riscv32-esp-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/olavt/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/riscv32-esp-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32c6
NOTICE: Skipping optional dependency: espressif/cmake_utilities
NOTICE: Skipping optional dependency: espressif/esp_wifi_remote
NOTICE: Skipping optional dependency: espressif/cmake_utilities
NOTICE: Skipping optional dependency: espressif/esp_wifi_remote
NOTICE: Skipping optional dependency: espressif/cmake_utilities
NOTICE: Skipping optional dependency: espressif/esp_wifi_remote
NOTICE: Skipping optional dependency: espressif/cmake_utilities
NOTICE: Skipping optional dependency: espressif/esp_wifi_remote
NOTICE: Processing 18 dependencies:
NOTICE: [1/18] espressif/button (3.5.0)
NOTICE: [2/18] espressif/cbor (0.6.0~1)
NOTICE: [3/18] espressif/cmake_utilities (0.5.3)
NOTICE: [4/18] espressif/esp-serial-flasher (0.0.11)
NOTICE: [5/18] espressif/esp_delta_ota (1.1.1)
NOTICE: [6/18] espressif/esp_diag_data_store (1.0.2)
NOTICE: [7/18] espressif/esp_diagnostics (1.2.1)
NOTICE: [8/18] espressif/esp_encrypted_img (2.1.0)
NOTICE: [9/18] espressif/esp_insights (1.2.2)
NOTICE: [10/18] espressif/esp_rcp_update (1.2.0)
NOTICE: [11/18] espressif/esp_secure_cert_mgr (2.5.0)
NOTICE: [12/18] espressif/jsmn (1.1.0)
NOTICE: [13/18] espressif/json_generator (1.1.2)
NOTICE: [14/18] espressif/json_parser (1.0.3)
NOTICE: [15/18] espressif/led_strip (1.0.0)
NOTICE: [16/18] espressif/mdns (1.8.2)
NOTICE: [17/18] espressif/rmaker_common (1.4.6)
NOTICE: [18/18] idf (5.4.0)
-- Found Python3: /Users/olavt/.espressif/esp-matter/connectedhomeip/connectedhomeip/.environment/cipd/packages/pigweed/bin/python3.11 (found version "3.11.3") found components: Interpreter
NOTICE: Skipping optional dependency: espressif/esp_wifi_remote
NOTICE: Skipping optional dependency: espressif/cmake_utilities
-- Project sdkconfig file /Users/olavt/Source/Repos/light/sdkconfig
warning: default on the choice symbol SEC_CERT_DAC_PROVIDER (defined at /Users/olavt/.espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/Kconfig:815, /Users/olavt/.espressif/esp-matter/components/esp_matter/Kconfig:48) will have no effect, as defaults do not affect choice symbols
warning: the choice symbol SEC_CERT_DAC_PROVIDER (defined at /Users/olavt/.espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/Kconfig:815, /Users/olavt/.espressif/esp-matter/components/esp_matter/Kconfig:48) is defined with a prompt outside the choice
info: INFO: Symbol SEC_CERT_DAC_PROVIDER defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
[... cut ...]
/Users/olavt/esp/v5.4/esp-idf/components/openthread/openthread/src/core/meshcop/secure_transport.cpp:1126:17: error: 'mbedtls_ssl_set_hs_ecjpake_password' was not declared in this scope
 1126 |                 mbedtls_ssl_set_hs_ecjpake_password(&mSsl, mPsk, mPskLength);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[262/1685] Building CXX object esp-idf/openthread/CMakeFiles/__idf_openthread.dir/openthread/src/core/net/netif.cpp.obj
ninja: build stopped: subcommand failed.

 *  The terminal process "ninja" terminated with exit code: 1.

Jerry-ESP, 2025-03-25:
Please share your current esp-matter and esp-idf commit-id.

Jerry-ESP, 2025-03-27:
I have tried on my side, it seems have some issue in this idf version, can you change a idf version, latest release/v5.3,  release/v5.4 , v5.2.3, v5.3.2 ... are OK

olavt, 2025-03-27:
Using "release/v5.4" worked"

What's the difference between "v5.4" and "release/v5.4"? Does "release/v5.4" contain some changes that was committed after "v5.4" was released?

Jerry-ESP, 2025-03-27:
The release/v5.4 branch will keep update, after release v5.4 tag, it should have many commits.  
But for the matter development, you can use the idf version which we suggested in the sdk if you don't have special request, the current version is [v5.2.3](https://github.com/espressif/esp-idf/tree/v5.2.3).

For other newer idf version, as we haven't tested fully, there maybe some unknown issues.

olavt, 2025-03-27:
Where can I see that the current version is v5.2.3?

Normally I would assume that the release with the highest version number (v5.4) would be the current version and that it have been fully tested.

olavt, 2025-03-27:
Another related question:

I want to change the sensors example from WiFi to Thread, but there is no "sdkconfig.defaults.c6_thread" file in the sensors example. Can I copy the file from the light example and use it with the sensor example with no problems?

Jerry-ESP, 2025-03-27:
> Another related question:
> 
> I want to change the sensors example from WiFi to Thread, but there is no "sdkconfig.defaults.c6_thread" file in the sensors example. Can I copy the file from the light example and use it with the sensor example with no problems?
> 
> I tried and it failed with:
> 
> FAILED: partition_table/partition-table.bin /Users/olavt/Source/Repos/sensors/build/partition_table/partition-table.bin cd /Users/olavt/Source/Repos/sensors/build/esp-idf/partition_table && /Users/olavt/.espressif/python_env/idf5.4_py3.9_env/bin/python /Users/olavt/esp/v5.4/esp-idf/components/partition_table/gen_esp32part.py -q --offset 0x8000 --flash-size 2MB -- /Users/olavt/Source/Repos/sensors/partitions.csv /Users/olavt/Source/Repos/sensors/build/partition_table/partition-table.bin && /Users/olavt/.espressif/tools/cmake/3.30.2/CMake.app/Contents/bin/cmake -E echo "Partition table binary generated. Contents:" && /Users/olavt/.espressif/tools/cmake/3.30.2/CMake.app/Contents/bin/cmake -E echo "_**" && /Users/olavt/.espressif/python_env/idf5.4_py3.9_env/bin/python /Users/olavt/esp/v5.4/esp-idf/components/partition_table/gen_esp32part.py -q --offset 0x8000 --flash-size 2MB -- /Users/olavt/Source/Repos/sensors/build/partition_table/partition-table.bin && /Users/olavt/.espressif/tools/cmake/3.30.2/CMake.app/Contents/bin/cmake -E echo "**_" Partitions tables occupies 3.9MB of flash (4087808 bytes) which does not fit in configured flash size 2MB. Change the flash size in menuconfig under the 'Serial Flasher Config' menu. [20/2291] Building CXX object esp-idf/openthread/CMakeFiles/__idf_openthread.dir/src/esp_openthread_platform.cpp.obj ninja: build stopped: subcommand failed.
> 
> * The terminal process "ninja" terminated with exit code: 1.
> 
> I didn't see any settings related to the flash size in "sdkconfig.defaults.c6_thread".

From the log, the issue is that in the partition.csv,  ***Partitions tables occupies 3.9MB of flash (4087808 bytes)***, but in your sdkconfig, the flash size you specified is 2M, you should change the flash size to 4MB in the sdkconfig. 

[image]

If it still not work, you can show me your operation step by step, and share me the logs.

closed 2025-04-18 by olavt