espressif/esp-matter #1342: How can a single device support multiple device types?

ethan-jy, 2025-03-25:
How can a single device support multiple device types?

We are developing an LED driver with four channels, and we need to implement the following device types:

Mode 1: Extended Color Light — For RGBW and TW color tuning.
Mode 2: Dimmable Light — Each of the four channels can be individually controlled for brightness.
Mode 3: Dimmable Light — All four channels output the same duty cycle simultaneously.
As shown in the figure below:

[image]

We have two proposed approaches:

Approach 1:
Use a DIP switch to toggle between modes. Both the Extended Color Light (Mode 1) and Dimmable Light (Modes 2/3) would be implemented on Endpoint 1. Users physically switch the DIP to select Mode 1, 2, or 3.

Approach 2:

Endpoint 1: Implements Extended Color Light (Mode 1).
Endpoints 2–5: Implement Mode 2 (individual channel dimming).
Endpoint 6: Implements Mode 3 (synchronized dimming).
Questions:

Are both approaches valid?
Are there alternative solutions?
Which method is more practical or recommended?
Are there relevant examples for reference?
Thank you for your guidance!

Jerry-ESP, 2025-03-25:
Hi @ethan-jy ,

I understand that:
Mode 1 is an extended color light in one endpoint
Mode 2 is four dimmable light in four endpoints
Mode 3 is a dimmable light in one endpoint

So for the approach 1, one endpoint is not enough, you can try this method:  when device power on, you can read the dip switch state first,  the judge the mode, if it is mode one, just create and extended color light endpoint, if mode two, create four dimmable light endpoints,  if mode 3, create one dimmable light endpoint.

For approach 2, it is available, but when the device pairing into a ecosystem, for example Apple home, it will show 6 devices in the App, no matter what is the current mode.

Jerry-ESP, 2025-03-25:
For extended color light, the color mode was switch by command, there are 3 color modes as below:

[image]

I think you don't need use the dip switch to switch the color mode.

And the color mode will stay after device power cycle, because its value will be store in the flash.

dhrishi, 2025-05-14:
@ethan-jy Please close the issue if resolved

dhrishi, 2025-09-13:
Closing this as the question is answered