espressif/esp-matter #1346: WPS Feature in Light example with matter

Ajay2803, 2025-03-26:
Hi,

I am working on Light example in esp-matter sdk, If i want to do Provisioning using wps and also with matter standard way. When i press wps button from my router it should get connect to my matter device.

Is this possible? And if yes how can i achieve this?

shripad621git, 2025-03-28:
@Ajay2803 , as a starting point,you can try out the [wps example](https://github.com/espressif/esp-idf/blob/a45d713b03fd96d8805d1cc116f02a4415b360c7/examples/wifi/wps/README.md) in esp-idf to get a hold of wps functionality and then try out the code integration as per you requirements.

Ajay2803, 2025-03-28:
Hi, @shripad621git ,

I have already tried this example, and also tried to integrate it in our code, but we have faced lot of errors, so we just want to know that, it it possible to add provisioning using wps alongside matter standard way.

shripad621git, 2025-04-04:
@Ajay2803, can you briefly elaborate the efforts on wps example integration and your specific use case. I relate the use case to achieve wifi connect using wps and then do a onnetwork commisioning using chip-tool or any other ecosystem which is on the same network.

dhrishi, 2025-04-17:
@Ajay2803 Please provide the requested info

closed 2025-04-18 by Ajay2803