espressif/esp-matter #1348: QR Code with custom Passcode not recognized by SmartThings

OHyungM, 2025-03-27:
Hi team,

I'm currently testing Matter onboarding with SmartThings, and I’ve encountered an issue related to QR code generation.

Situation:
When I only modify the discriminator, generate a new QR code, and try onboarding with SmartThings, it works perfectly.

However, when I change the setup passcode (PIN code) and generate a QR code accordingly, SmartThings responds with:
"This QR code is not supported."

What works:
Default passcode with modified discriminator

QR generation and onboarding both succeed

What fails:
Modified discriminator and Custom passcode with valid QR code

SmartThings fails to recognize the QR and does not allow onboarding

Notes:
The QR code is generated using  
./SetupPayload.py generate -d MY-DISCRIMINATOR -p MY-PASSCODE --vendor-id 65521 --product-id 32768
(MY-DISCRIMINATOR, MY-PASSCODE is num)
, and it reflects the correct updated passcode.

Discriminator and Passcode  value remains valid and within the expected range.

No errors occur during QR generation or device boot.

Questions:
1. Does SmartThings restrict onboarding to specific passcodes (e.g., only the ones registered/certified)?

2. Are there additional fields in the QR code payload that must be updated alongside the passcode?

Jerry-ESP, 2025-03-27:
Hi @OHyungM ,
I have tried to generate 3 qrcodes as below, all of them can be scan correctly in SmartThing App. Please have a try.
MT:Y.K90-Q000UBA50.700
MT:Y.K90-Q0002Z.548G00
MT:Y.K90MPP03Z8P-38G00

And Can you show me your qrcodes , I can try it on my side.

[image]

And this is my app version:

[image]

OHyungM, 2025-03-27:
Hi, Thank you for reply !

I tried generating a QR code using your examples like:

MT:Y.K90-Q000UBA50.700  
MT:Y.K90-Q0002Z.548G00  
MT:Y.K90MPP03Z8P-38G00

However, when I scan any of those QR codes in the SmartThings app, I get the following error message (translated from Korean):

“There was a problem connecting to the device. Please check the QR or setup code and try again.
Error Code: 39-106”

So even with your provided QR codes, SmartThings fails to add the device with that same error.

Let me summarize what I’ve tried on my side as well:

I generated the QR code like this:

./SetupPayload.py generate -d 3333 -p 77777777 --vendor-id 65521 --product-id 32768 -cf 0 -dm 2

Result : 

Manual code: 31931347476
QRCode: MT:Y.K906PF15848S4HO10

I implemented and used shell commands to dynamically change the discriminator and setup passcode at runtime via CLI.

When using default passcode (20202021) and changing only the discriminator, everything works fine on SmartThings.

But when I change the passcode to 77777777, the SmartThings app consistently shows the "unsupported QR code" error or error code 39-106.

Could you please help clarify whether SmartThings currently supports custom passcodes during onboarding?

Thanks again!

Jerry-ESP, 2025-03-27:
There are some invalid passcodes as below:

[image]

So you can't use the 77777777 passcode.

For the qrcodes:
MT:Y.K90-Q000UBA50.700
MT:Y.K90-Q0002Z.548G00
MT:Y.K90MPP03Z8P-38G00

They are available, but did you update the Spake2pVerifier in the device which corresponding to the passcode?

Which example you are using?  How did you change your passcode in the device?

Maybe you can refer to this [example](https://github.com/espressif/esp-matter/tree/main/examples/light_switch) to enable the dynamic passcode.

OHyungM, 2025-03-27:
1. Where did you capture that image from?

Could you share the source of the image that shows the invalid passcodes? I'd like to check the details.

2. Issue with your provided QR codes

Even when I use the QR codes you provided, I still get the following error in the SmartThings app:

"There was a problem connecting to the device. Please check the QR or setup code and try again.
Error Code: 39-106"

The device is not registering successfully. Do you have any insights on why this might be happening? 

3. Regarding Spake2pVerifier

I'm not entirely sure what you mean by:

"Did you update the Spake2pVerifier in the device corresponding to the passcode?"

Could you clarify what specific step I need to take for this?

4. Example and Passcode Modification Method

I am using the Multiple On/Off Plugin Units example.

I modified the passcode by adding a custom shell command in idf.py monitor to update the passcode in NVS dynamically.

Let me know if there are any additional details you need! Thanks again for your help.

Jerry-ESP, 2025-03-28:
1. You can apply for this document form here: https://csa-iot.org/developer-resource/specifications-download-request/ , the ***Matter 1.4 Core Specification***
2. I don't know what is this error code mean, maybe you can test using the chip-tool, it will be more clearly.
3. You can read the matter spec in the question 1 to understand the definition.
4. If you want to change the qrcode or passcode, you can refer to this document: https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#factory-data-providers , use the [esp-matter-mfg-tools](https://github.com/espressif/esp-matter-tools/tree/main/mfg_tool) to generate the factory.bin file which include the whole pairing info.

OHyungM, 2025-03-28:
Thank you for your regards..!

I have a couple of additional questions. Could you help me with them?

1. If there are multiple devices with the same Passcode and Discriminator on the same network, what priority is used to determine which one gets onboarded first?

2. In real-world products, what mechanisms are used to uniquely identify each device?

OHyungM, 2025-03-31:
1. So, can we think that the purpose of the QR code is more for commissioning rather than device identification?

2. Is the Unique ID different from the Node ID? Also, how can we set the Serial Number or Unique ID?

Jerry-ESP, 2025-03-31:
> 1. So, can we think that the purpose of the QR code is more for commissioning rather than device identification?
```
Yes, correct
```
> 3. Is the Unique ID different from the Node ID? Also, how can we set the Serial Number or Unique ID?
```
Both of the Unique ID and Seial number are attributes in the basicinformation cluster. For serial number, you can specified it when you generate the factory info by esp-matter-mfg-tools with param "--serial-num"; for the unique id, it is a random and unique value for device after factory reset.

![Image](https://github.com/user-attachments/assets/e5d1e6d5-5978-4527-9a3d-c45880a00c4a)

dhrishi, 2025-04-15:
@OHyungM Please close the issue if your questions are resolved

closed 2025-04-16 by OHyungM