espressif/esp-matter #1350: How to change sensors example from WiFi to Thread?

olavt, 2025-03-27:
I'm using ESP-IDF v5.2.3 with ESP-Matter. The hardware is an ESP32-C6 dev kit.

I would like to change from Matter over WiFi to Matter over Thread.

I created this "sdkconfig.defaults.c6_thread" file:
CONFIG_IDF_TARGET="esp32c6"

# libsodium
CONFIG_LIBSODIUM_USE_MBEDTLS_SHA=y

# NIMBLE
CONFIG_BT_ENABLED=y
CONFIG_BT_NIMBLE_ENABLED=y
CONFIG_BT_NIMBLE_EXT_ADV=n
CONFIG_BT_NIMBLE_HCI_EVT_BUF_SIZE=70
CONFIG_USE_BLE_ONLY_FOR_COMMISSIONING=n

# Enable OpenThread
CONFIG_OPENTHREAD_ENABLED=y
CONFIG_OPENTHREAD_SRP_CLIENT=y
CONFIG_OPENTHREAD_DNS_CLIENT=y
CONFIG_OPENTHREAD_LOG_LEVEL_DYNAMIC=n
CONFIG_OPENTHREAD_LOG_LEVEL_NOTE=y
CONFIG_OPENTHREAD_CLI=n

# Disable lwip ipv6 autoconfig
CONFIG_LWIP_IPV6_AUTOCONFIG=n

# LwIP config for OpenThread
CONFIG_LWIP_IPV6_NUM_ADDRESSES=8
CONFIG_LWIP_MULTICAST_PING=y

# MDNS platform
CONFIG_USE_MINIMAL_MDNS=n
CONFIG_ENABLE_EXTENDED_DISCOVERY=y

# Disable STA for ESP32C6
CONFIG_ENABLE_WIFI_STATION=n

I issued this command:

idf.py -D SDKCONFIG_DEFAULTS="sdkconfig.defaults.c6_thread" set-target esp32c6

The build resulted in an error.

It looks like this is the fatal error:

/Users/olavt/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/chip/lib/libCHIP.a(cryptopal_mbedtls.CHIPCryptoPALmbedTLS.cpp.o): in function `chip::Crypto::HKDF_sha::HKDF_SHA256(unsigned char const*, unsigned int, unsigned char const*, unsigned int, unsigned char const*, unsigned int, unsigned char*, unsigned int)':
/Users/olavt/Source/Repos/sensors/build/esp-idf/chip/../../../../../../.espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/crypto/CHIPCryptoPALmbedTLS.cpp:299:(.text._ZN4chip6Crypto8HKDF_sha11HKDF_SHA256EPKhjS3_jS3_jPhj+0x68): undefined reference to `mbedtls_hkdf'

This is the build log (some uninteresting parts omitted):

 *  Executing task: ninja  

[5/2201] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[1007/2201] Performing configure step for 'chip_gn'
Done. Made 4450 targets from 310 files in 222ms
[1123/2201] Building C object esp-idf/espressif__esp_diagnostics/CMakeFiles/__idf_espressif__esp_diagnostics.dir/src/esp_diagnostics_utils.c.obj
In file included from /Users/olavt/Source/Repos/sensors/managed_components/espressif__esp_diagnostics/src/esp_diagnostics_utils.c:52:
/Users/olavt/esp/v5.2.3/esp-idf/components/freertos/esp_additions/include/freertos/task_snapshot.h:8:2: warning: #warning freertos/task_snapshot.h header is no longer used, and will be removed in future versions. [-Wcpp]
    8 | #warning freertos/task_snapshot.h header is no longer used, and will be removed in future versions.
      |  ^~~~~~~
[... 216 lines cut ...]
In file included from /Users/olavt/.espressif/esp-matter/connectedhomeip/connectedhomeip/src/app/ConcreteAttributePath.h:23,
                 from /Users/olavt/.espressif/esp-matter/connectedhomeip/connectedhomeip/zzz_generated/app-common/app-common/zap-generated/cluster-objects.h:26,
                 from /Users/olavt/.espressif/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/test-cluster-server/test-cluster-server.cpp:23:
In copy constructor 'constexpr chip::Optional<T>::Optional(const chip::Optional<T>&) [with T = chip::app::Clusters::Globals::TestGlobalEnum]',
    inlined from 'constexpr chip::app::Clusters::UnitTesting::Structs::SimpleStruct::Type::Type(const chip::app::Clusters::UnitTesting::Structs::SimpleStruct::Type&)' at /Users/olavt/.espressif/esp-matter/connectedhomeip/connectedhomeip/zzz_generated/app-common/app-common/zap-generated/cluster-objects.h:45819:8,
    inlined from 'bool emberAfUnitTestingClusterTestEmitTestEventRequestCallback(chip::app::CommandHandler*, const chip::app::ConcreteCommandPath&, const chip::app::Clusters::UnitTesting::Commands::TestEmitTestEventRequest::DecodableType&)' at /Users/olavt/.espressif/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/test-cluster-server/test-cluster-server.cpp:906:107:
/Users/olavt/.espressif/esp-matter/connectedhomeip/connectedhomeip/src/lib/core/Optional.h:72:74: warning: 'arg4.chip::app::Clusters::UnitTesting::Structs::SimpleStruct::Type::i.chip::Optional<chip::app::Clusters::Globals::TestGlobalEnum>::mValueHolder.chip::Optional<chip::app::Clusters::Globals::TestGlobalEnum>::ValueHolder::<anonymous>.chip::Optional<chip::app::Clusters::Globals::TestGlobalEnum>::TrivialDestructor::mValue.chip::Optional<chip::app::Clusters::Globals::TestGlobalEnum>::TrivialDestructor::Value::mData' may be used uninitialized [-Wmaybe-uninitialized]
   72 |             new (&mValueHolder.mValue.mData) T(other.mValueHolder.mValue.mData);
      |                                                ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
/Users/olavt/.espressif/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/test-cluster-server/test-cluster-server.cpp: In function 'bool emberAfUnitTestingClusterTestEmitTestEventRequestCallback(chip::app::CommandHandler*, const chip::app::ConcreteCommandPath&, const chip::app::Clusters::UnitTesting::Commands::TestEmitTestEventRequest::DecodableType&)':
/Users/olavt/.espressif/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/test-cluster-server/test-cluster-server.cpp:901:33: note: 'arg4' declared here
  901 |     Structs::SimpleStruct::Type arg4;
      |                                 ^~~~
[2197/2201] Generating ld/sections.ld
warning: default on the choice symbol SEC_CERT_DAC_PROVIDER (defined at /Users/olavt/.espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/Kconfig:815, /Users/olavt/.espressif/esp-matter/components/esp_matter/Kconfig:48) will have no effect, as defaults do not affect choice symbols
warning: the choice symbol SEC_CERT_DAC_PROVIDER (defined at /Users/olavt/.espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/Kconfig:815, /Users/olavt/.espressif/esp-matter/components/esp_matter/Kconfig:48) is defined with a prompt outside the choice
[2199/2201] Linking CXX executable sensors.elf
FAILED: sensors.elf 
: && /Users/olavt/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/riscv32-esp-elf-g++ -march=rv32imac_zicsr_zifencei -nostartfiles -march=rv32imac_zicsr_zifencei --specs=nosys.specs    -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32C6=0 -Wl,--Map=/Users/olavt/Source/Repos/sensors/build/sensors.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32c6.peripherals.ld -T esp32c6.rom.ld -T esp32c6.rom.api.ld -T esp32c6.rom.rvfp.ld -T esp32c6.rom.newlib.ld -T esp32c6.rom.version.ld -T esp32c6.rom.phy.ld -T esp32c6.rom.coexist.ld -T esp32c6.rom.net80211.ld -T esp32c6.rom.pp.ld -T esp32c6.rom.wdt.ld -T esp32c6.rom.systimer.ld -T esp32c6.rom.newlib-normal.ld -T esp32c6.rom.heap.ld -T memory.ld -T sections.ld CMakeFiles/sensors.elf.dir/project_elf_src_esp32c6.c.obj -o sensors.elf -L/Users/olavt/esp/v5.2.3/esp-idf/components/soc/esp32c6/ld   -L/Users/olavt/esp/v5.2.3/esp-idf/components/esp_rom/esp32c6/ld   -L/Users/olavt/Source/Repos/sensors/build/esp-idf/esp_system/ld   -L/Users/olavt/esp/v5.2.3/esp-idf/components/esp_phy/lib/esp32c6   -L/Users/olavt/esp/v5.2.3/esp-idf/components/esp_coex/lib/esp32c6   -L/Users/olavt/esp/v5.2.3/esp-idf/components/esp_wifi/lib/esp32c6 esp-idf/riscv/libriscv.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_bootloader_format/libesp_bootloader_format.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/esp_common/libesp_common.a [... cut ...] ned int, unsigned char*, unsigned int)':
/Users/olavt/Source/Repos/sensors/build/esp-idf/chip/../../../../../../.espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/crypto/CHIPCryptoPALmbedTLS.cpp:299:(.text._ZN4chip6Crypto8HKDF_sha11HKDF_SHA256EPKhjS3_jS3_jPhj+0x68): undefined reference to `mbedtls_hkdf'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

 *  The terminal process "ninja" terminated with exit code: 1.

shripad621git, 2025-03-28:
@olavt , please enable `CONFIG_MBEDTLS_HKDF_C=y` in the `sdkconfig.defaults.c6_thread `to resolve the `linker issue`.

dhrishi, 2025-04-17:
@olavt Please close the issue if resolved. The above suggestion should help you fix the linking error

closed 2025-04-18 by olavt