espressif/esp-matter #1352: Adding some Air Quality cluster features results in linker error

olavt, 2025-03-28:
**Environment**
- ESP-Matter Commit Id: f9f62dd6851d93dd9c770f0ed0066e822f955c35
- ESP-IDF Commit Id: c9763f62dd00c887a1a8fafe388db868a7e44069
- SoC (eg: ESP32 or ESP32-C3): ESP32-C6

This works fine:

    cluster::air_quality::feature::fair::add(cluster);

When I try to add these features, I get a linker errors:

    cluster::air_quality::feature::mod::add(cluster);
    cluster::air_quality::feature::vpoor::add(cluster);
    cluster::air_quality::feature::xpoor::add(cluster);

/Users/olavt/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/main/libmain.a(app_main.cpp.obj): in function `AddAirQualityClusterFeatures(unsigned int*)':
/Users/olavt/Source/Repos/light/main/app_main.cpp:376:(.text._Z28AddAirQualityClusterFeaturesPj+0x16): undefined reference to `esp_matter::cluster::air_quality::feature::mod::add(unsigned int*)'
/Users/olavt/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: /Users/olavt/Source/Repos/light/main/app_main.cpp:377:(.text._Z28AddAirQualityClusterFeaturesPj+0x20): undefined reference to `esp_matter::cluster::air_quality::feature::vpoor::add(unsigned int*)'
/Users/olavt/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: /Users/olavt/Source/Repos/light/main/app_main.cpp:379:(.text._Z28AddAirQualityClusterFeaturesPj+0x30): undefined reference to `esp_matter::cluster::air_quality::feature::xpoor::add(unsigned int*)'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

Jerry-ESP, 2025-03-31:
Please make sure the names of the Api correct: https://github.com/espressif/esp-matter/blob/f9f62dd6851d93dd9c770f0ed0066e822f955c35/components/esp_matter/esp_matter_feature.cpp#L1030

shripad621git, 2025-04-04:
@olavt , the change in available on the latest [main branch ](https://github.com/espressif/esp-matter/commit/2eef0abd141394106686b72445aef4e4500e8168) now.

olavt, 2025-04-05:
Thanks, all these are working now:

    cluster::air_quality::feature::fair::add(cluster);
    cluster::air_quality::feature::moderate::add(cluster);
    cluster::air_quality::feature::very_poor::add(cluster);
    cluster::air_quality::feature::extremely_poor::add(cluster);

closed 2025-04-05 by olavt