espressif/esp-matter #1354: Want to do Matter Firmware update over OTA

Ajay2803, 2025-03-29:
Hi,

I have one question, Let's say i have one product smart plug with-out matter, which have esp-idf firmware and now i want to update firmware over-the-air(OTA) which have matter support, Is it possible? let's say it matter smart plug.

Is this kind of functionality is possible or not? if Yes can you please guide us how can we do so?

olavt, 2025-03-29:
I know vendors doing this today. They have existing firmware with the ability to do firmware updates. The user may choose to upgrade the device from non-Matter to a Matter capable firmware. I think this is just providing a Matter capable firmware in your current firmware update functionality.

Ajay2803, 2025-03-31:
Hi, @olavt 

Thanks for your reply,
As you said that this kind of functionality is possible, but how does user will get matter pairing code or generated qr code for there matter device after firmware update in non-matter firmware?

And If you have any example which does this functionality or which have working similar to this, can you please share that, it would be very helpful for us.

Jerry-ESP, 2025-03-31:
Hi @Ajay2803 ,

One possible solution:
You need to pre-programming the matter certificate and factory info(qrcode, serial number ...) when you produce the non-matter device, then when you can upgrade directly when you want to update to matter. And for this solution, you can paste the matter pairing qrcode on the production line.

Ajay2803, 2025-03-31:
Hi @Jerry-ESP ,
Thanks for suggestion, But suppose My Non-Matter product is already running in Market with OTA Support, Now if i want to Update Matter Firmware into it, How can I do so, It does not have  pre-programming the matter certificate and factory info it.

Is it possible?

olavt, 2025-03-31:
@Ajay2803 I don't have any details, but from my own experience as a user, the vendor sent the QR-Code by email as a part of the process where the device firmware was to be updated to a Matter-based version. This was triggered by the vendors proprietary app that was used as a part of the upgrade firmware process.

Jerry-ESP, 2025-03-31:
> Hi [@Jerry-ESP](https://github.com/Jerry-ESP) , Thanks for suggestion, But suppose My Non-Matter product is already running in Market with OTA Support, Now if i want to Update Matter Firmware into [...]

If you want to generate the qrcode in the device, you can refer to the dynamic passcode example: https://github.com/espressif/esp-matter/tree/main/examples/light_switch

And for the dac pre-provisioning, we ESP have a solution for our own pre-provisioning service, if you want to know more information, you can contact our business colleague: https://www.espressif.com/en/contact-us/sales-questions

closed 2025-04-18 by Ajay2803