espressif/esp-matter #1355: How to add Software Diagnostics Cluster?

olavt, 2025-03-30:
I would like to add the Software Diagnostiocs Cluster to the root endpoint.

I have tried with:

void AddSoftwareDiagnosticsCluster(node_t* node)
{
    endpoint_t* root_endpoint = endpoint::get(node, 0);

    cluster::software_diagnostics::config_t config;
    uint32_t features = cluster::software_diagnostics::feature::watermarks::get_id();
    cluster::software_diagnostics::create(root_endpoint, &config, CLUSTER_FLAG_SERVER, features);
}

But, I get this linker error:

/Users/olavt/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/esp_matter/libesp_matter.a(esp_matter_cluster.cpp.obj): in function `.L0 ':
/Users/olavt/.espressif/esp-matter/components/esp_matter/esp_matter_cluster.cpp:2612:(.text._ZZN10esp_matter7cluster20software_diagnostics6createEPjPNS0_6common6configEhmENUlvE_4_FUNEv+0x10): undefined reference to `MatterSoftwareDiagnosticsPluginServerInitCallback()'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

wqx6, 2025-03-31:
Hi, we deselect all the optional clusters support for light example for flash optimization. If you want to add software diagnostic cluster please add the following config to your sdkconfig file.
```
CONFIG_SUPPORT_SOFTWARE_DIAGNOSTICS_CLUSTER=y
```

olavt, 2025-03-31:
@wqx6 I suppose I should add it to sdkconfig.defaults as sdkconfig is not supposed to be edited?

olavt, 2025-03-31:
@wqx6 I added the setting you suggested to sdkconfig.defaults and now it built with no errors.

But, a lot of relevant data is missing:

[image]

I would have expected something like this:

[image]

wqx6, 2025-04-01:
> But, a lot of relevant data is missing:

The three attributes are optional for the cluster, so they will not be added to the cluster by default. You need to add them manually after creating the cluster on root_node endpoint.Also, you need to enable CONFIG_FREERTOS_USE_TRACE_FACILITY to enable the ThreadMetrics attribute

[image]

olavt, 2025-04-01:
I enabled this in sdkconfig:

CONFIG_FREERTOS_USE_TRACE_FACILITY=y

I still get no data in the ThreadMetrics attribute:

[image]

wqx6, 2025-04-02:
Sorry for that we didn't add the APIs to add other attributes in SoftwareDiagnostics Cluster, Please try the following sample code, and we will provide the APIs of `cluster::software_diagnostics::attribute::create_XXX()` later:
```
    endpoint_t *root_node_ep = endpoint::get(node, 0);
    cluster::software_diagnostics::config_t sd_conf;
    cluster_t *sd_cluster = cluster::software_diagnostics::create(root_node_ep, &sd_conf, CLUSTER_FLAG_SERVER,
                                          cluster::software_diagnostics::feature::watermarks::get_id());
    attribute::create(sd_cluster, SoftwareDiagnostics::Attributes::CurrentHeapFree::Id, ATTRIBUTE_FLAG_MANAGED_INTERNALLY,
                      esp_matter_uint64(0));
    attribute::create(sd_cluster, SoftwareDiagnostics::Attributes::CurrentHeapUsed::Id, ATTRIBUTE_FLAG_MANAGED_INTERNALLY,
                      esp_matter_uint64(0));
    attribute::create(sd_cluster, SoftwareDiagnostics::Attributes::ThreadMetrics::Id, ATTRIBUTE_FLAG_MANAGED_INTERNALLY,
                      esp_matter_array(nullptr, 0, 0));
```

Note that please enable CONFIG_FREERTOS_USE_TRACE_FACILITY so that the ThreadMetrics can be read correctly.

olavt, 2025-04-02:
Thanks! It's working now:

[image]

wqx6, 2025-04-03:
> Why are the StackFreeCurrent and StackSize fields not populated?

FreeRTOS does not provide API to get the two parameters for each task, so we will set the two values NULL.

closed 2025-04-03 by olavt