espressif/esp-matter #1356: How to get debugger to work with Visual Studio Code?

olavt, 2025-04-01:
How can I get the debugger to work properly with Visual Studio Code?

When I run my project using the debugger it stops at a temporary breakpoint in app_main.

But, when I set various breakpoints in my code, the debugger does not stop. Stepping manually from the temporary breakpoint does also not work properly. If I step over this statement "nvs_flash_init();", the debugger does not stop on the next statement.

I have selected this OpenOCD Board Configuration:

[image]

This is my launch.json file:
```json
{
  "version": "0.2.0",
  "configurations": [
    {
      "type": "gdbtarget",
      "request": "attach",
      "name": "Eclipse CDT GDB Adapter"
    },
    {
      "type": "espidf",
      "name": "Launch",
      "request": "launch"
    }
  ]
}
```

What could be the issue?

shripad621git, 2025-04-10:
@olavt, please refer to the documentation to configure debugger with vscode [here](https://docs.espressif.com/projects/vscode-esp-idf-extension/en/latest/debugproject.html).

dhrishi, 2025-04-17:
@olavt Please close the issue if your issue is resolved

olavt, 2025-04-18:
I find the documentation quite confusing.

I would like to verify that I have the basics correct:

I'm using the ESP32-C6 Dev Kit 

1) Which USB-C port should I connect to in order to debug?

I'm currently using the port labeled "ESP32-C6 USB Type-C Port" (not the one labeled "USB Type-C to UART Port") in the documentation. This is the port to the right of the board.

Is this correct?

2) When selecting the port to use, I get these options:

[image]

Is the "/dev/tty.usbmodem1101 ESP32-C6 (QFN40) (revision vO.1)" the correct one to use?

3) When selecting the "OpenOCD Board Configuration", I get this list:

[image]

Is "ESP32-C6 chip (via builtin USB-JTAG) ESP32-C6 debugging via builtin USB-JTAG (board/esp3..." the correct option to select?

shripad621git, 2025-04-28:
@olavt , selection of the port is the port which you are using and specific to your use case. The second and third choices made by you seem to be correct for debugging purposes.

olavt, 2025-04-28:
@shripad621git Ok, I'm still not able to get this to work.

Are the manual steps mentioned under MacOS here required when running this using VS Code and the ESP-IDF extension?