espressif/esp-matter #1357: Matter Device Dimming Anomaly

ethan-jy, 2025-04-02:
**Describe the bug**
esp-idf: V5.3.1
esp-matter: V1.3

We implemented a Dimmable Light device using Espressif’s ESP32H2, which has entered mass production. However, customers have encountered an issue during practical use. When adding the device to an Apple tablet and configuring two scenarios:

Scenario 1: "Away from home – Turn off the light."
Scenario 2: "Return home – Turn on the light at 100% brightness,"
this results in a failure to turn on the light. Refer to the attached log for details.

During debugging with the chip-tool, we reproduced the issue by alternately using the commands:
1. onoff off
2. levelcontrol move-to-level-with-on-off 0xdd 0 0 0
When the level value (0xdd) remains unchanged, dimming fails (i.e., turning the light off and then on again). However, if the level value is modified, the operation succeeds as expected.

Please advise on resolving this issue.

**Environment**
- ESP-Matter Commit Id: V1.3
- ESP-IDF Commit Id: V5.3.1
- SoC (eg: ESP32 or ESP32-C3):ESP32H2
- Device Logs (Please attach the log file):

[Scene 2 turn on the light --- Error-- .txt](https://github.com/user-attachments/files/19559955/Scene.2.turn.on.the.light.---.Error--.txt)

- Host Machine OS:
- Host Machine Python version:

- Commissioner app and versions if present:
- Home hub app and versions if present:
- Commissioner's logs if present:

**Any additional details**
...

ethan-jy, 2025-04-02:
OFF Log:
[Scene 1 --- OFF ---- Log(OFF动作).txt](https://github.com/user-attachments/files/19559967/Scene.1.---.OFF.----.Log.OFF.txt)

When the Level value changes, the behavior works correctly：：LOG

[level 0xdd changed,  works correctly.txt](https://github.com/user-attachments/files/19560001/level.0xdd.changed.works.correctly.txt)

shripad621git, 2025-04-18:
@ethan-jy , I have tested the dimmable light with below commands alternatively on release/v1.3.
```
onoff off
levelcontrol move-to-level-with-on-off 0xdd 0 0 0
```
As per the spec, the move-to-level-with-on-off command is trying to set a level greater than minimum defined level in your use case and hence it will set the onoff attribute to on in this case.

[image]

Did you give a try with esp32h2 light example on release/v1.3 and observe this behaviour as the tests seem to run fine as per spec on my end. Off command sets the light off and move-to-level-with-on-off sets it to the level after setting the onoff attribute to on.

I have also read the value using chip-tool. Off command sets it to 1 and move-to-level-with-on-off again sets it to 221 in this case. Below are the attached logs which depict the above behaviour.

[move_to_level_with_on_off.txt](https://github.com/user-attachments/files/19812095/move_to_level_with_on_off.txt)

shripad621git, 2025-04-21:
@ethan-jy , any updates on the issue?

dhrishi, 2025-05-14:
@ethan-jy Can you please provide an update?