espressif/esp-matter #1358: There was an error initializing “nimble” again

MaplestoryAlen, 2025-04-02:
I want to turn off nimble when Matter is not using BLE, for example, when "Commissioning window opened", I want to actively turn off "Commissioning wind", and then initialize my own BLE service. However, after turning off BLE, there is an error message when initializing again

chip::DeviceLayer::Internal::BLEMgr().Shutdown();

static void app_event_cb(const ChipDeviceEvent *event, intptr_t arg)
{
    switch (event->Type) 
   {
     case chip::DeviceLayer::DeviceEventType::kBLEDeinitialized:
                  ESP_LOGI(TAG, "BLE deinitialized and memory reclaimed");
		  EnableBleDegbugMode();
                  break;	
    }
}

void EnableBleDegbugMode(void)
{ 
    ....
    nimble_port_init();
    ....
}

log:

I (1770) main_task: Returned from app_main()
I (3426) app_main: BLE deinitialized and memory reclaimed

btdm: bss start 0x3fcef180, len 36
btdm: data start 0x3fcef174, data start rom 0x40057350, len 12
MAGIC fadebead VERSION 00010009
I (3439) BLE_INIT: BT controller compile version [c66a703]
MAGIC fadebead VERSION 00010001
HLI Magic mismatch: 00000020, fadebead
RWIP Heap alloc: ENV [0x3fcd3bf4 2088], MSG [0x3fcd00f0 9356], NORET [0x3fcb2738 1868], DB [0x00000000 0]
Uart ENV [0x00000000], VHCI ENV [0x3fcb0bcc]
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4037fa96  PS      : 0x00060033  A0      : 0x8214efd0  A1      : 0x3fcd3380  
--- 0x40057350: _data_start_btdm_rom in ROM

.--- 0x4037fa96: xTaskRemoveFromEventList at /home/alen/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3958

.A2      : 0x3fcb0c04  A3      : 0xffffffff  A4      : 0x8037f188  A5      : 0x3fcc5c80  
A6      : 0x00060723  A7      : 0x00000000  A8      : 0x3fcb0c14  A9      : 0x00013b18  
A10     : 0x00000002  A11     : 0x00000000  A12     : 0x3fcb0c2c  A13     : 0x3fcc5d60  
A14     : 0x40376748  A15     : 0x3c177faa  SAR     : 0x00000000  EXCCAUSE: 0x0000001d  
.........--- 0x40376748: btdm_controller_task at ??:?

EXCVADDR: 0x0000000a  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x4037fa93:0x3fcd3380 0x4214efcd:0x3fcd33a0 0x4208e72e:0x3fcd33e0 0x403768b1:0x3fcd3400
......................--- 0x4037fa93: xTaskRemoveFromEventList at /home/alen/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3958
0x4214efcd: xQueueSemaphoreTake at /home/alen/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1763
0x4208e72e: semphr_take_wrapper at /home/alen/esp-idf/components/bt/controller/esp32c3/bt.c:590

wqx6, 2025-04-02:
`chip::DeviceLayer::Internal::BLEMgr().Shutdown()` will release the static memory in IRAM used by the BLE, which is irreversible, so you cannot re-initialize BLE stack after that.

wqx6, 2025-04-07:
If you want to reuse BLE after Matter commissioning, please disable `CONFIG_USE_BLE_ONLY_FOR_COMMISSIONING` so that the BLE memory will not be claimed, and you should stop the Matter advertisement and clear Matter gatt services so that the BLE will not be used for Matter commissioning again.

MaplestoryAlen, 2025-04-07:
> CONFIG_USE_BLE_ONLY_FOR_COMMISSIONING

I have now configured ' CONFIG_USE_BLE_ONLY_FOR_COMMISSIONING=n', but I cannot find the API interface to turn off the GATT service in the 'BLEManagerial Impl. cpp' file. Can you tell me more specifically how to do it? Thank you

wqx6, 2025-04-07:
1. Call nimble_port_stop() to stop the nimble host task
2. Call ble_gatts_free_svcs() to clear current GATT services.
3. Add your GATT services with `ble_gatts_count_cfg()` and `ble_gatts_add_svcs()`
4. Re-run the nimble host task with `nimble_port_freertos_init()`

MaplestoryAlen, 2025-04-08:
> 2\. ble_gatts_free_svcs()

    My version is matter V1.4 and IDF version is V5.2.3, so there is no ble_gatts_free_stvcs() function. However, I replaced it with the ble_gatts_reset() function. According to steps 1-4 given above, I cannot see my own broadcast. If I do not turn on Bluetooth on matter and directly turn on my Bluetooth, there will be a broadcast and I can still connect.
   Start executing user-defined BLE from the call to "EnableBleDegbugMode();"
void EnableBleDegbugMode]void)
{   
      MatterCloseCommissioningWindow();
      IotBleStart();
   
}

void IotBleStart(void)
{
    esp_ble_conn_config_t config = {0};

    unsigned int mon, year,sn;
    uint32_t temp;
    sscanf(GetIotModuleInfoStruct()->iot_sn, "%02d%04d%06d", &mon, &year, &sn);
    // ESP_LOGI(TAG, "Ble year [%d], mon [%d], sn[%d]", year, mon, sn);
    temp = mon * 1000000 + sn;
    config.device_name[0] = 'G';
    config.device_name[1] = 's';
    config.device_name[2] = encode_chars[((uint16_t)IOT_DEV_CODE >> 8) % 64];
    config.device_name[3] = encode_chars[((uint16_t)IOT_DEV_CODE & 0x00FF) % 64];
    config.device_name[4] = encode_chars[(year - 2025) % 64];

    config.device_name[8] = encode_chars[temp % 64];
    temp /= 64;
    config.device_name[7] = encode_chars[temp % 64];
    temp /= 64;
    config.device_name[6] = encode_chars[temp % 64];
    temp /= 64;
    config.device_name[5] = encode_chars[temp % 64];
    ESP_LOGI(TAG, "Ble name [%s]", config.device_name);

    config.broadcast_data[0] = 'X';
    config.broadcast_data[1] = 'Z';
    uint8_t state = 0;
    if(GsStoreGetBindFlg())
        state |= 0x01;
    if(IotIsWifiprovisioned())
        state |= 0x02;
    config.broadcast_data[2] = state;
    config.broadcast_data[3] = GetIotModuleInfoStruct()->iot_type & 0x0F;
    char payload[7] = {0};
    strncpy(payload, GetIotSetupPayload() + strlen(GetIotSetupPayload()) - 6, 6);
    memcpy(&config.broadcast_data[4], payload, 6);
    
    IotSha256Pbkdf2GenerateKey((unsigned char *)GetIotModuleInfoStruct()->iot_sn, strlen(GetIotModuleInfoStruct()->iot_sn), 
                                (unsigned char *)GetIotModuleInfoStruct()->iot_model, strlen(GetIotModuleInfoStruct()->iot_model), (unsigned char *)iot_db_auth.comm_auth_key, AES_CBC_KEY_LEN);
    iot_db_auth.auth_active = false;
    iot_db_auth.auth_ok = false;
    if(iot_db_auth.encrypt_key)
        free(iot_db_auth.encrypt_key);
    iot_db_auth.encrypt_key = NULL;

    if(iot_ble_notify_mux != NULL)
    {
        vSemaphoreDelete(iot_ble_notify_mux);
        iot_ble_notify_mux = NULL;
    }
    iot_ble_notify_mux = xSemaphoreCreateRecursiveMutex();
    assert(iot_ble_notify_mux);

    esp_event_loop_create_default();
    esp_event_handler_register(BLE_CONN_MGR_EVENTS, ESP_EVENT_ANY_ID, app_ble_conn_event_handler, NULL);

    config.include_service_uuid = true;

    esp_ble_conn_init(&config);
[... 90 lines cut ...]
    }

    /* XXX Need to have template for store */
    ble_store_config_init();
    nimble_port_freertos_init(esp_ble_conn_host_task);

    return ESP_OK;
}

static void esp_ble_conn_host_task(void *param)
{
    /* This function will return only when nimble_port_stop() is executed */
    ESP_LOGD(TAG, "BLE Host Task Started");
    nimble_port_run();

    nimble_port_freertos_deinit();
}

log:

[image]

[iot_nimble.txt](https://github.com/user-attachments/files/19641975/iot_nimble.txt)

[iot_ble_debug.txt](https://github.com/user-attachments/files/19641977/iot_ble_debug.txt)