espressif/esp-matter #1359: Not getting subscription updates to administratorcommissioning subscribe window-status?

olavt, 2025-04-02:
I have two Matter over Thread devices.

One is developed on ESP-Matter SDK using ESP32-C6 hardware.

The other is developed using the Simplicity SDK with Silicon Labs Matter extensions.

I have a controller, which is subscribing to events from the Matter devices to observe any changes in attributes. The attribute I'm interested in particular is the WindowStatus of the Administrator Commissioning Cluster. The issue is that when I open a commissioning window on these devices, the controller only get update events on the WindowStatus from one of the devices (the Silicon Labs based device). The controller does not get any events for the ESP32 based device. From the chip-tool I can observe that the WindowStatus attribute does in fact change for the ESP32 based device and I get a valid QR-Code for pairing.

What could be the reason for the controller not getting the attribute updates from the ESP32-based device?

It does get attribute updates for other attributes.

wqx6, 2025-04-03:
Could you please provide more details about the issue?
- What controller did you use?
- How did you send the subscription request to ESP32-C6 device and what subscription parameters did you use for the request?

olavt, 2025-04-03:
I tested with the chip-tool on a Raspberry Pi 4 as the controller.

./chip-tool interactive start
administratorcommissioning subscribe window-status 5 10 1 0

wqx6, 2025-04-03:
Thanks for reporting! I can reproduce the issue, and I am looking into it.

wqx6, 2025-04-07:
Looks like the weak function [MatterReportingAttributeChangeCallback](https://github.com/project-chip/connectedhomeip/blob/master/src/app/reporting/Engine.cpp#L1222-L1224) was linked instead of the real report function in [reporting.cpp](https://github.com/project-chip/connectedhomeip/blob/master/src/app/reporting/reporting.cpp). 

We are trying to fix in in upstream repo. And you can try the following workaround fix for it:

1. Add a C function calling the `void MatterReportingAttributeChangeCallback(EndpointId endpoint)` in your app_main.cpp
```
#include <app/reporting/reporting.h>

extern "C" void matter_report_link_helper_impl(void)
{
    MatterReportingAttributeChangeCallback(0);
}
```

2. Add `target_link_libraries(${COMPONENT_LIB} INTERFACE "-u matter_report_link_helper_impl")` to the CMakeLists.txt file in your main component to make sure that the link helper function is linked so that the `reporting.cpp` will be linked.

wqx6, 2025-04-09:
Hi, You can also apply the upstream repo fix https://github.com/project-chip/connectedhomeip/pull/38280 for this issue

olavt, 2025-04-12:
When will the fix be available in this repo?

wqx6, 2025-04-14:
> When will the fix be available without a workaround?

It will be available after we update the submodule next time. Now you can cherry-pick this patch https://github.com/project-chip/connectedhomeip/pull/38280 to connectedhomeip repo to fix the issue which is not a workaround.

closed 2025-05-15 by dhrishi