espressif/esp-matter #1360: BLE_INIT Error when PSRAM is enabled

sdrasti, 2025-04-03:
When enabling PSRAM I get the following errors during startup when building with esp-idf v5.2.3:

```
E (1664) BLE_INIT: Malloc failed
E (1664) BLE_INIT: controller init failed
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
```

While a related fix was implemented in esp-idf v5.3-beta:  https://github.com/espressif/esp-idf/commit/0f29b3cef97d406c5887d79da8592308540cdda0, the same issue persists when building with the release version v5.3:

```
I (1684) BLE_INIT: BT controller compile version [ba3b3e3]
E (1694) BLE_INIT: Malloc failed
E (1694) BLE_INIT: controller init failed
```

Setting `CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_EXTERNAL=y` didn't change anything.

Could this be due to a configuration issue? Are there specific Nimble Stack settings required when using PSRAM? Or do I need to use another esp-idf build?

- ESP-Matter Commit Id: #2eef0ab
- ESP-IDF Commit Id:  TAG v5.2.3 / v5.3
- SoC:  ESP32-S3 / N8R8 DevkitC-1

CONFIG_SPIRAM_BOOT_INIT=y
CONFIG_SPIRAM_MODE_OCT=y
CONFIG_SPIRAM_TYPE_AUTO=y
CONFIG_SPIRAM_USE_CAPS_ALLOC=y

shripad621git, 2025-04-04:
@sdrasti , can you please confirm if you have followed the instructions [here](https://github.com/espressif/esp-matter/tree/main/examples/controller#a13-ram-optimization) in the controller example for PSRAM.

sdrasti, 2025-04-04:
@shripad621git That resolved the issue. The project now builds successfully with v5.2.3. Thank you!

closed 2025-04-04 by sdrasti

JsonJiu, 2025-04-26:
@sdrasti 请问你是怎么解决的