espressif/esp-matter #1361: esp-matter installing failed on install.sh

lunDreame, 2025-04-06:
My environment is mac os on 15.5 and python 3.9.6
ESP-IDF: v5.2.1
ESP-MATTER: release/v1.2

During Install CHIP prerequisite packages I gettting error

I run on below command
$ cd ${sdk_path}/esp-matter
$ bash ./install.sh

[log.txt](https://github.com/user-attachments/files/19620784/log.txt)
at the log file I may be "gn_out" dependancy expect error
how can i fix it?

shripad621git, 2025-04-07:
@lunDreame , can you please take a look at this [link](https://github.com/google/protobuf-gradle-plugin/issues/543) and try out the fix suggested as I can see `OSError: [Errno 86] Bad CPU type in executable: PosixPath('protoc')` in you error logs.

lunDreame, 2025-04-12:
> [@lunDreame](https://github.com/lunDreame) , can you please take a look at this [link](https://github.com/google/protobuf-gradle-plugin/issues/543) and try out the fix suggested as I can see `OSError: [Errno 86] Bad CPU type in executable: PosixPath('protoc')` in you error logs.

thank you, It was resolved through the corresponding link solution.

closed 2025-04-12 by lunDreame