espressif/esp-matter #1362: If I use the official code of matter, how to connect to esp_rmaker

wolen666, 2025-04-08:
**Describe the question/query that you have**
A clear and concise description of what you want to ask.

**Additional context**
...

Jerry-ESP, 2025-04-08:
Please refer to this example: https://github.com/espressif/esp-rainmaker/tree/master/examples/matter

wolen666, 2025-04-09:
I use esp32 and matter esp_rmaker. Below is the code of esp_rmaker. It initializes after the matter device is connected to the network. I used Do not use Claiming in the configuration of esp_rmaker. After running the program, the following problem occurs.The following is the main code and basic configuration

[code.txt](https://github.com/user-attachments/files/19659266/code.txt)

[image]

[image]

Jerry-ESP, 2025-04-09:
Did you run the matter example in rainmaker repo?  Please run the example and refer to its setup flow:  https://github.com/espressif/esp-rainmaker/blob/master/examples/matter/README.md, it seems that you haven't downloaded the certificate into the device.

wolen666, 2025-04-10:
I embedded Rainmaker as a component into the official code of matter. I have burned the certificate DAC into the fctry partition. The picture below is my partition description, but I am using a test certificate.

[image]

[image]

[image]

[image]

wolen666, 2025-04-10:
But it seems that the above problem still exists

Jerry-ESP, 2025-04-10:
If you want to commissioning rainmaker, you need to execute claim command as the document describe: https://github.com/espressif/esp-rainmaker/blob/master/examples/matter/README.md#claiming-device-certificates, the dac certificate not only use as the matter dac certificate, but also use as the rainmaker certificate.

closed 2025-04-18 by wolen666