espressif/esp-matter #1364: Power Source Cluster on Endpoint 0

sdrasti, 2025-04-08:
I'm encountering an issue implementing the Power Source Cluster. The specification indicates it should reside on the Root Node, but I can't figure how to instantiate it there. Currently I have created it on Endpoint_1:

```C
if (Node != nullptr)
   {
      endpoint_t *Endpoint1;
      endpoint_t *Endpoint2;

      power_source_device::config_t                          PowerSourceConfig;
      battery_storage::config_t                              BatteryStorageConfig;
      cluster::power_source::feature::battery::config_t      Battery_config;
      cluster::power_source::feature::rechargeable::config_t Rechargeable_config;

      Endpoint1 = power_source_device::create(Node, &PowerSourceConfig, ENDPOINT_FLAG_NONE, NULL);
      esp_matter::cluster_t *Cluster = esp_matter::cluster::get(Endpoint1, chip::app::Clusters::PowerSource::Id);

      /* Set default values */
      Battery_config.bat_charge_level = chip::to_underlying(PowerSource::BatChargeLevelEnum::kCritical);
      Rechargeable_config.bat_charge_state = chip::to_underlying(PowerSource::BatChargeStateEnum::kIsNotCharging);

      /* Add features  */
      cluster::power_source::feature::battery::add(Cluster, &Battery_config);
      cluster::power_source::feature::rechargeable::add(Cluster, &Rechargeable_config);

      /* Add optional 'percent remaining' attribut */
      attribute_t *Attribute = cluster::power_source::attribute::create_bat_percent_remaining(Cluster, 30, 0, 200);
```

It works like so. However, I'm looking for the right approach to get it hooked on endpoint 0.

Jerry-ESP, 2025-04-09:
You can refer to the ota requestor implementation: https://github.com/espressif/esp-matter/blob/210311d1626e4eb54f99733f019a2d209cb35623/components/esp_matter/esp_matter_ota.cpp#L63

sdrasti, 2025-04-09:
@Jerry-ESP Thank you!

I've since learned that with Matter 1.2, the power source configuration cluster is no longer used. Its endpoint list has been migrated to the power source cluster. This means that for devices with only one power source, if no endpoints are explicitly listed in the power source cluster, it applies to the entire Node and can be located on any endpoint.

closed 2025-04-09 by sdrasti