espressif/esp-matter #1365: Update docker hub images

jzwyssig, 2025-04-09:
The Docker image `release-v1.2_idf_v5.2.1` for Matter 1.2 has not been updated for 10 months:

[image]

It would be great to update this version from time to time. You could number the release candidates, e.g., `release-v1.2_idf_v5.2.1-r2`, `release-v1.2_idf_v5.2.1-r3`, etc.

Thank you!

shripad621git, 2025-04-10:
@jzwyssig , thanks for reporting the issue. The docker image for `release/v1.2` is pushed through the `github actions` to the `Dockerhub` once any change is merged on v1.2-branch. As of now, the docker build specifically for v1.2-branch in the actions seems to be broken. We will fix it asap and the image will be modified accordingly.

closed 2025-04-11 by dhrishi

jzwyssig, 2025-04-13:
Awesome, thanks a lot!